ros-navigation/navigation2 #4970: MPPI Controller Biasing Speed to 0.5 m/s Instead of Reaching vx_max

alex-roba, 2025-03-07:
## Bug report

- Operating System:
  - Ubuntu 24.04 (Noble)
- Computer:
  - 13th Gen Intel
- ROS2 Version:
  - Jazzy binaries
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
I am using the default MPPI parameters, with only a few modifications based on the official guide. Specifically, I adjusted:

Pruning distance: Set to be the same or greater than my max projectable distance.
Thresholds (Goal and Path Following Critics): Set to my max projectable distance for a smooth transition.
Local costmap size: Adjusted accordingly.
However, I noticed unexpected behavior in velocity control. The robot accelerates normally up to 0.5 m/s (default value), but then only increases gradually with noise, reaching around 0.6 m/s at most, even though vx_max is set to 1.0 m/s.
here is my paramters
```
controller_server:
  ros__parameters:
    use_sim_time: false
    controller_frequency: 20.0
    odom_topic: "/odom"
    progress_checker_plugin: "progress_checker"
    goal_checker_plugins: ["stopped_goal_checker"]
    controller_plugins: ["FollowPath"]
    # Progress checker parameters
    progress_checker:
      plugin: "nav2_controller::PoseProgressChecker"
      required_movement_radius: 0.2
      required_movement_angle: 0.3
      movement_time_allowance: 3.0
    stopped_goal_checker:
      plugin: "nav2_controller::StoppedGoalChecker"
      trans_stopped_velocity: 0.02
      xy_goal_tolerance: 0.1
      rot_stopped_velocity: 0.1
      yaw_goal_tolerance: 0.05
    FollowPath:
      plugin: "nav2_mppi_controller::MPPIController"
      time_steps: 56
      model_dt: 0.05
      batch_size: 2000
      ax_max: 3.0
      ax_min: -3.0
      az_max: 3.5
      vx_std: 0.2
      wz_std: 0.4
      vx_max: 1.0
      vx_min: -1.0
      wz_max: 1.5
      iteration_count: 1
      prune_distance: 3.2
      transform_tolerance: 0.1
      temperature: 0.3
      gamma: 0.015
      motion_model: "DiffDrive"
      visualize: false
      regenerate_noises: true
      TrajectoryVisualizer:
        trajectory_step: 5
        time_step: 3
      critics: [
        "ConstraintCritic", "GoalCritic",
        "GoalAngleCritic", "PathAlignCritic", "PathFollowCritic",
        "PathAngleCritic"]
      ConstraintCritic:
        enabled: true
        cost_power: 1
        cost_weight: 4.0
      GoalCritic:
        enabled: true
        cost_power: 1
        cost_weight: 5.0
        threshold_to_consider: 2.8
      GoalAngleCritic:
        enabled: true
        cost_power: 1
        cost_weight: 3.0
        threshold_to_consider: 0.5
      PathAlignCritic:
        enabled: true
        cost_power: 1
        cost_weight: 14.0
        max_path_occupancy_ratio: 0.05
        trajectory_point_step: 4
        threshold_to_consider: 0.5
        offset_from_furthest: 20
        use_path_orientations: false
      PathFollowCritic:
        enabled: true
        cost_power: 1
        cost_weight: 5.0
        offset_from_furthest: 5
        threshold_to_consider: 2.8
      PathAngleCritic:
        enabled: true
        cost_power: 1
        cost_weight: 2.0
        offset_from_furthest: 4
        threshold_to_consider: 0.5
        max_angle_to_furthest: 1.0
        mode: 1
```
```
local_costmap:
  local_costmap:
    ros__parameters:
      footprint_padding: 0.0
      update_frequency: 10.0
      publish_frequency: 2.0
      global_frame: odom
      robot_base_frame: base_link
      rolling_window: true
      width: 6
      height: 6
      resolution: 0.03
      use_sim_time: False
      footprint: "[ [0.45, 0.3], [0.45, -0.3], [-0.45, -0.3], [-0.45, 0.3] ]"
      plugins: ["obstacle_layer", "inflation_layer"]
      obstacle_layer:
        plugin: "nav2_costmap_2d::ObstacleLayer"
        enabled: True
        observation_sources: scan
        footprint_clearing_enabled: true
        max_obstacle_height: 2.0
        combination_method: 1
        scan:
          topic: <robot_namespace>/scan_multi_filtered
          obstacle_max_range: 2.5
          obstacle_min_range: 0.0
          raytrace_max_range: 3.0
          raytrace_min_range: 0.0
          max_obstacle_height: 2.0
          min_obstacle_height: 0.0
          clearing: True
          marking: True
          data_type: "LaserScan"
          inf_is_valid: false
      inflation_layer:
        plugin: "nav2_costmap_2d::InflationLayer"
        enabled: True
        cost_scaling_factor: 3.0
        inflation_radius: 0.52
      always_send_full_costmap: True
```

#### Expected behavior
The robot should continue accelerating smoothly up to 1.0 m/s when feasible, without artificial constraints.

#### Actual behavior
The robot seems biased toward the default 0.5 m/s speed in MPPI.
Acceleration appears to slow down significantly after 0.5 m/s, even though there are no explicit constraints preventing higher speeds.
This suggests that something embedded in MPPI might be unintentionally biasing the speed toward the default 0.5 m/s.

[image]

#### Reproduction instructions

#### Additional information

SteveMacenski, 2025-03-07:
what are your velocity smoother parameters? Is this happening in free zero-cost space or confined spaces with potential (please provide a video)? 

What have you tried to tune so far?

alex-roba, 2025-03-11:
@SteveMacenski  I already figured out the reason. To achieve the desired result, we also need to increase std_vx_std. In my opinion, it would be worth mentioning this in the README, as all the MPPI tuning guides in the forum tend to overlook this and focus only on tuning the cost.

[image]

SteveMacenski, 2025-03-11:
Ah this is a really good insight, thank you @alex-roba. Would you mind opening a PR against the README with that note in the tuning section? We should probably mirror it in the docs config guide as well (https://docs.nav2.org/configuration/packages/configuring-mppic.html#notes-to-users)

SteveMacenski, 2025-03-14:
I updated it! https://github.com/ros-navigation/docs.nav2.org/blob/master/configuration/packages/configuring-mppic.rst

phanhoc1902, 2025-06-04:
> [@SteveMacenski](https://github.com/SteveMacenski) I already figured out the reason. To achieve the desired result, we also need to increase std_vx_std. In my opinion, it would be worth mentioning [...]

Can you give me a reference to the yaml file with the parameter set that results in the robot moving with vmax=1.0m/s? Thank you.