ros-navigation/navigation2 #4972: [Question] Handling of unknown space between costmap initialization and isCollision checking

mini-1235, 2025-03-07:
In the following line, if `track unknown` is false then we treat it as free space
https://github.com/ros-navigation/navigation2/blob/443915ca8e6622e2e953ee8ef8168fc7ee0ae716/nav2_costmap_2d/plugins/obstacle_layer.cpp#L119-L123

However, when checking collision
https://github.com/ros-navigation/navigation2/blob/443915ca8e6622e2e953ee8ef8168fc7ee0ae716/nav2_mppi_controller/src/critics/obstacles_critic.cpp#L182-L203

if `is tracking unknown` is false then we consider it in collision

I think these two pieces of code actually have opposite meanings, or did I misunderstand something?

mini-1235, 2025-03-08:
Another point here is that I didn't see how NO_INFORMATION is treated as free space in obstacle critic
https://github.com/ros-navigation/navigation2/blob/443915ca8e6622e2e953ee8ef8168fc7ee0ae716/nav2_mppi_controller/src/critics/obstacles_critic.cpp#L138-L144
shouldn't we add something like 
```cpp
if(is_tracking_unknown_ && pose_cost.cost == 255.0f) {continue;}  // Treat unknown as free space?
// or 
if(!is_tracking_unknown_ && pose_cost.cost == 255.0f) {continue;}  // Treat unknown as free space?
```

SteveMacenski, 2025-03-10:
If we don't track unknown space, then the default in the costmap is `FREE` (0). Thus in the MPPI obstacles critic, if we have `NO_INFORMATION` (255) and track unknown space is off, then it is not a valid state (inCollision = true). 

Complimentary, if we track unkown space so the default background cost is `NO_INFORMATION` (255), then if we find `NO_INFORMATION` in MPPI, it is valid (i.e. inCollision = false).

I believe these are consistent. 

> if is tracking unknown is false then we consider it in collision

I think you're inverting what this function does. The return of `inCollision` is true when its in collision, not when its valid and collision-free -- unless I'm missing something obvious 😄 

---

> Another point here is that I didn't see how NO_INFORMATION is treated as free space in obstacle critic

 Its not treated as freespace, you are correct. Unknown when it is being tracked is not equal to free. It means we don't know anything about it. That's not the same thing as being freespace where we have sensor data that is clearing out freespace with raycasting to know that it is free. Unknown slowly becomes free as the robot explores and sensor data hits more bounds to know what is free. Thus, unknown space actually is still scored as `255` as a high cost since we don't know anything about it and thereby it is much higher risk than space we do know something about.

mini-1235, 2025-03-13:
> If we don't track unknown space, then the default in the costmap is `FREE` (0). Thus in the MPPI obstacles critic, if we have `NO_INFORMATION` (255) and track unknown space is off, then it is not a valid state (inCollision = true).
> 
> Complimentary, if we track unkown space so the default background cost is `NO_INFORMATION` (255), then if we find `NO_INFORMATION` in MPPI, it is valid (i.e. inCollision = false).
> 
> I believe these are consistent.
> 
> > if is tracking unknown is false then we consider it in collision
> 
> I think you're inverting what this function does. The return of `inCollision` is true when its in collision, not when its valid and collision-free -- unless I'm missing something obvious 😄
> 
> > Another point here is that I didn't see how NO_INFORMATION is treated as free space in obstacle critic
> 
> Its not treated as freespace, you are correct. Unknown when it is being tracked is not equal to free. It means we don't know anything about it. That's not the same thing as being freespace where we have sensor data that is clearing out freespace with raycasting to know that it is free. Unknown slowly becomes free as the robot explores and sensor data hits more bounds to know what is free. Thus, unknown space actually is still scored as `255` as a high cost since we don't know anything about it and thereby it is much higher risk than space we do know something about.
> 
> I hope that answers your questions!

Hi @SteveMacenski, thanks for the explanation in this issue(and the review in my PRs). You are right, I have actually misunderstood the function.

I am currently busy with my school work, I will continue my PRs next monday, and complete them as soon as possible.

closed 2025-03-13 by mini-1235