ros-navigation/navigation2 #4975: Discussion / question about `simulate_ahead_time` parameter of rotationShimController

gennartan, 2025-03-10:
**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Computer:
  - Nvidia Jetson Orin / Thinkpad (with gazebo simulation)
- ROS2 Version:
  - Iron
- DDS implementation:
  - Cyclone DDS
- Nav2 Package:
  - RotationShimController (+MPPI)
- Have you asked this on Robotics Stack Exchange?
  - NO

#### Tuning / Configuration Goal

Good performance of robot controller.

#### What I've Tried

I noticed performances issues in the rotation shim controller due to the feature behind the parameter `simulate_ahead_time`. I observed it because I could never reach the expected controller frequency while in rotation shim controller (20Hz), and once the MPPI controller takes back the control, there are no more warnings for missed controller frequency. With problem in control frequency, I also observed failed transform, causing the rotationShimController to give the control directly to the MPPI controller, even if it hadn't properly performed the rotation

So I was wondering what was the exact idea behind this parameter. In my case, I simply set it to 0.0. In term of pure control, it is simply less CPU consuming. In term of safety / obstacle avoidance, it doesn't change anything since I am using the collision monitor to reduce speeds / stop the robot while close to obstacles.

Is there something that I am missing about this parameter ?

Thanks for your responses !

SteveMacenski, 2025-03-10:
What value of `simulate_ahead_time` where you using that caused you issues? When set to something like `1.0` for 1s at the 20hz frequency, that should only be 20 collision checks that should be very doable to 20hz.

I'm noticing that we do full SE2 collision checks there with `footprintCostAtPose`. Are you using a circular robot and/or have your costmap set to use a radius rather than a footprint?

If so, maybe the optimization is the do something like we do in other places ([for example](https://github.com/ros-navigation/navigation2/blob/e3c41beb08cd1939d8eed378768d2744bf47904c/nav2_planner/src/planner_server.cpp#L691-L702)) where we check if using a footprint to do the SE2 checks, otherwise do the point checks that are equivalent. 

Though, if using a circular robot, given the rotation shim controller only rotates the robot, [I think that collision checking should actually just be bypassed here](https://github.com/ros-navigation/navigation2/blob/main/nav2_rotation_shim_controller/src/nav2_rotation_shim_controller.cpp#L324), since the circle rotating is always just itself. 

```
if (!use_radius) {
  isCollisionFree(cmd_vel, angular_distance_to_heading, pose);
}
```

Let me know on those points. I think we should be able to address both.

> Is there something that I am missing about this parameter ?

This makes sure that you don't run into things on the control basis. I don't think its a good idea to rely fully on the collision monitor for that. The collision monitor will stop the robot from rotating too far, but won't actually pass it off to a non-rotating controller to get out of the situation when rotation is not possible. It'll just stall out indefinitely until the control progress checker expires. I think we should work to address your issue so that you can use that feature to be more reliable if rotating to pose near obstacles often.

gennartan, 2025-03-11:
Thanks for your answer and your explanation, it makes things easier to understand.

Unfortunately, we are not using a circular robot. We are using a rectangular footprint, and we are rotating around the front wheels which are not at the center of the robot, so changing the footpring to a circular one is not possible.

We use a `simulate_ahead_time = 1.0` with `controller_frequency = 20.0`. So just as you said, there should be only 20 collisions check performed, which should be doable. This makes me think that there is something when tyring to check for collisions that is buggy (either a bug in the code, or a bug in the configuration). Whenever I have time, I will try to investigate further to better understand what exactly causes the isCollisionFree to slow down the entire control.

> This makes sure that you don't run into things on the control basis. I don't think its a good idea to rely fully on the collision monitor for that. The collision monitor will stop the robot from [...]

The current bug that we are encoutering is a bit the opposite of the one this feature is trying to prevent. Because the robot cannot rotate*, it passes the control the the MPPI controller. The MPPI controller will eventully fail its operation because it did not start in the good orientation** resulting in a navigation failure.

To summarize, on my project I will strongly reduce it for now `simulate_ahead_time = 0.1` (or something like this), and I will keep an eye on this project and to try to find the root cause.

*During rotation, we often encounter a TF error, this TF error is observed only when the controller cannot meet its frequency.
** Thats a constraint of our project, we can have very small distance to navigate, if not in the good orientation, the MPPI will not perform a correct rotation.

SteveMacenski, 2025-03-11:
>  Whenever I have time, I will try to investigate further to better understand what exactly causes the isCollisionFree to slow down the entire control.

Much appreciated!  

> I will strongly reduce it for now simulate_ahead_time = 0.1 (or something like this)

That's not wholesale unreasonable in general. The only constraint I would put on it is given your robot's angular acceleration limits and max rotational speed set in the rotation shim controller, make sure the time is set to greater than the time it would take to stop from full rotational speed by its acceleration limit for safety purposes. I suspect your robot can stop quite quickly, so something like ~0.3 might be well sufficient. 

> *During rotation, we often encounter a TF error, this TF error is observed only when the controller cannot meet its frequency.

Ah, that actually may be your issue itself. If TF is having trouble looking up transforms, it might be using up to the transform timeout duration trying to acquire one. That timeout duration might be what is exceeding the control frequency. That is usually the sign of a system that has problems with its TF tree (i.e. not publishing fused map->odom fast enough, for example).  

Where is the TF error coming from?

gennartan, 2025-03-13:
You seem to have nailed it ! 👍  Thanks! 

We recently switched from the TF publisher map -> odom from the `ros2_control/diff_drive_controller` to the `robot_localization/ekf_node` (because we changed the hardware controller and the diff drive controller is already integrated within the hardware controller).

I will investigate what is the root cause in my project then.
Thanks again !

closed 2025-03-13 by gennartan