ros-navigation/navigation2 #4979: MPPI and SMAC planner not compiling on debian:12-slim for Jazzy

javijurado, 2025-03-11:
## Bug report

**Required Info:**
- Operating System:
  - docker container: debian:12-slim
- Computer:
  - Dell Inc. XPS 15 9530 with Ubuntu 24.04.2 LTS
- ROS2 Version:
  - jazzy
- Version or commit hash:
  - Compiling source on jazzy branch: [last commit](https://github.com/ros-navigation/navigation2/commit/e4ef813e6b5c6edc4307231e75a9cd01ebbd8c5b)

#### Steps to reproduce issue
```
I'm trying to compile navigation2 stack from source for a debian:12-slim but when I'm compiling the following errors appear on `nav2_mppi_controller` and also `nav2_smac_planner`
```

#### Expected behavior

Be able to compile the code from source

#### Actual behavior

nav2_mppi_controller:
```bash
In file included from /usr/include/xtensor/xmath.hpp:30,
                 from /usr/include/xtensor/xcontainer.hpp:25,
                 from /usr/include/xtensor/xtensor.hpp:20,
                 from /home/javier/workspace/KION/kion-ros2-rt/src/navigation2/nav2_mppi_controller/include/nav2_mppi_controller/optimizer.hpp:25,
                 from /home/javier/workspace/KION/kion-ros2-rt/src/navigation2/nav2_mppi_controller/src/optimizer.cpp:15:
In function ‘void xt::detail::shape_computation(RS&, R&, E&, const AX&, std::enable_if_t<(! is_fixed<RS>::value), int>) [with O = xt::reducer_options<float, std::tuple<xt::evaluation_strategy::immediate_type> >; RS = std::array<long unsigned int, 0>; R = xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 0, xt::layout_type::row_major, xt::xtensor_expression_tag>; E = xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 1, xt::layout_type::row_major, xt::xtensor_expression_tag>; AX = std::array<long unsigned int, 1>]’,
    inlined from ‘auto xt::reduce_immediate(F&&, E&&, X&&, O&&) [with F = xreducer_functors<math::minimum<void>, const_value<float>, math::minimum<void> >; E = xtensor_container<uvector<float, xsimd::aligned_allocator<float, 32> >, 1, xt::layout_type::row_major, xtensor_expression_tag>&; X = std::array<long unsigned int, 1>; O = std::tuple<evaluation_strategy::immediate_type>&]’ at /usr/include/xtensor/xreducer.hpp:314:45,
    inlined from ‘auto xt::detail::reduce_impl(F&&, E&&, X&&, xt::evaluation_strategy::immediate_type, O&&) [with F = xt::xreducer_functors<xt::math::minimum<void>, xt::const_value<float>, xt::math::minimum<void> >; E = xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 1, xt::layout_type::row_major, xt::xtensor_expression_tag>&; X = std::array<long unsigned int, 1>; O = std::tuple<xt::evaluation_strategy::immediate_type>&]’ at /usr/include/xtensor/xreducer.hpp:876:13,
    inlined from ‘auto xt::reduce(F&&, E&&, EVS&&) [with F = xreducer_functors<math::minimum<void>, const_value<float>, math::minimum<void> >; E = xtensor_container<uvector<float, xsimd::aligned_allocator<float, 32> >, 1, xt::layout_type::row_major, xtensor_expression_tag>&; EVS = std::tuple<evaluation_strategy::immediate_type>&; typename std::enable_if<xtl_requires<is_reducer_options<X>, detail::is_xreducer_functors_impl<typename std::decay<_Tp>::type> >, int>::type <anonymous> = 0]’ at /usr/include/xtensor/xreducer.hpp:948:9,
    inlined from ‘auto xt::amin(E&&, EVS) [with T = void; E = xtensor_container<uvector<float, xsimd::aligned_allocator<float, 32> >, 1, xt::layout_type::row_major, xtensor_expression_tag>&; EVS = std::tuple<evaluation_strategy::immediate_type>; typename std::enable_if<xtl_requires<is_reducer_options<X> >, int>::type <anonymous> = 0]’ at /usr/include/xtensor/xmath.hpp:784:5,
    inlined from ‘void mppi::Optimizer::updateControlSequence()’ at /home/javier/workspace/KION/kion-ros2-rt/src/navigation2/nav2_mppi_controller/src/optimizer.cpp:413:47:
/usr/include/xtensor/xreducer.hpp:209:43: error: null pointer dereference [-Werror=null-dereference]
  209 |                         result_shape[idx] = expr.shape()[i];
      |                         ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~
In function ‘void xt::detail::shape_computation(RS&, R&, E&, const AX&, std::enable_if_t<(! is_fixed<RS>::value), int>) [with O = xt::reducer_options<float, std::tuple<xt::evaluation_strategy::immediate_type> >; RS = std::array<long unsigned int, 0>; R = xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 0, xt::layout_type::row_major, xt::xtensor_expression_tag>; E = xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 1, xt::layout_type::row_major, xt::xtensor_expression_tag>; AX = std::array<long unsigned int, 1>]’,
    inlined from ‘auto xt::reduce_immediate(F&&, E&&, X&&, O&&) [with F = xreducer_functors<detail::plus, const_value<float>, detail::plus>; E = xtensor_container<uvector<float, xsimd::aligned_allocator<float, 32> >, 1, xt::layout_type::row_major, xtensor_expression_tag>&; X = std::array<long unsigned int, 1>; O = std::tuple<evaluation_strategy::immediate_type>&]’ at /usr/include/xtensor/xreducer.hpp:314:45,
    inlined from ‘auto xt::detail::reduce_impl(F&&, E&&, X&&, xt::evaluation_strategy::immediate_type, O&&) [with F = xt::xreducer_functors<plus, xt::const_value<float>, plus>; E = xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 1, xt::layout_type::row_major, xt::xtensor_expression_tag>&; X = std::array<long unsigned int, 1>; O = std::tuple<xt::evaluation_strategy::immediate_type>&]’ at /usr/include/xtensor/xreducer.hpp:876:13,
    inlined from ‘auto xt::reduce(F&&, E&&, EVS&&) [with F = xreducer_functors<detail::plus, const_value<float>, detail::plus>; E = xtensor_container<uvector<float, xsimd::aligned_allocator<float, 32> >, 1, xt::layout_type::row_major, xtensor_expression_tag>&; EVS = std::tuple<evaluation_strategy::immediate_type>&; typename std::enable_if<xtl_requires<is_reducer_options<X>, detail::is_xreducer_functors_impl<typename std::decay<_Tp>::type> >, int>::type <anonymous> = 0]’ at /usr/include/xtensor/xreducer.hpp:948:9,
    inlined from ‘auto xt::sum(E&&, EVS) [with T = void; E = xtensor_container<uvector<float, xsimd::aligned_allocator<float, 32> >, 1, xt::layout_type::row_major, xtensor_expression_tag>&; EVS = std::tuple<evaluation_strategy::immediate_type>; typename std::enable_if<xtl_requires<is_reducer_options<X> >, int>::type <anonymous> = 0]’ at /usr/include/xtensor/xmath.hpp:1848:5,
    inlined from ‘void mppi::Optimizer::updateControlSequence()’ at /home/javier/workspace/KION/kion-ros2-rt/src/navigation2/nav2_mppi_controller/src/optimizer.cpp:415:51:
/usr/include/xtensor/xreducer.hpp:209:43: error: null pointer dereference [-Werror=null-dereference]
  209 |                         result_shape[idx] = expr.shape()[i];
```

nav2_smac_planner
```bash
--- stderr: nav2_smac_planner
/usr/bin/ld: ../libnav2_smac_planner_2d.so: undefined reference to `vtable for ompl::base::RealVectorStateSpace'
/usr/bin/ld: ../libnav2_smac_planner_2d.so: undefined reference to `ompl::base::CompoundStateSpace::CompoundStateSpace()'
/usr/bin/ld: ../libnav2_smac_planner_2d.so: undefined reference to `ompl::base::StateSpace::StateSpace()'
/usr/bin/ld: ../libnav2_smac_planner_2d.so: undefined reference to `vtable for ompl::base::SE2StateSpace'
/usr/bin/ld: ../libnav2_smac_planner_2d.so: undefined reference to `vtable for ompl::base::SO2StateSpace'
/usr/bin/ld: ../libnav2_smac_planner_2d.so: undefined reference to `ompl::base::CompoundStateSpace::addSubspace(std::shared_ptr<ompl::base::StateSpace> const&, double)'
/usr/bin/ld: ../libnav2_smac_planner_2d.so: undefined reference to `ompl::base::CompoundStateSpace::lock()'
/usr/bin/ld: ../libnav2_smac_planner_2d.so: undefined reference to `ompl::base::StateSpace::getName[abi:cxx11]() const'
/usr/bin/ld: ../libnav2_smac_planner_2d.so: undefined reference to `vtable for ompl::base::ReedsSheppStateSpace'
/usr/bin/ld: ../libnav2_smac_planner_2d.so: undefined reference to `vtable for ompl::base::DubinsStateSpace'
/usr/bin/ld: ../libnav2_smac_planner_2d.so: undefined reference to `vtable for ompl::base::CompoundStateSpace'
/usr/bin/ld: ../libnav2_smac_planner_2d.so: undefined reference to `ompl::base::StateSpace::setName(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: ../libnav2_smac_planner_2d.so: undefined reference to `ompl::base::StateSpace::~StateSpace()'
/usr/bin/ld: ../libnav2_smac_planner_2d.so: undefined reference to `ompl::base::RealVectorBounds::resize(unsigned long)'
collect2: error: ld returned 1 exit status
[... cut ...]
/usr/bin/ld: ../libnav2_smac_planner_lattice.so: undefined reference to `ompl::base::RealVectorBounds::resize(unsigned long)'
collect2: error: ld returned 1 exit status
gmake[2]: *** [test/CMakeFiles/test_smac_lattice.dir/build.make:546: test/test_smac_lattice] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:530: test/CMakeFiles/test_smac_lattice.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
```

#### Reproduction instructions

#### Additional information
To fix smac_planner I have added `set(OMPL_LIBRARIES "ompl::ompl")` on the CMakeList but I don't know if this is the proper way of fixing this

SteveMacenski, 2025-03-11:
It seems to me like in this documentation that you're using a newer version of OMPL than Nav2 is supporting: https://github.com/ompl/ompl/blob/main/doc/markdown/buildSystem.md?plain=1

That mentions the "previous" behavior as setting up the cmake variables, whereas the newer system has `ompl::ompl`. We use the version of OMPL that is shipped with rosdistro for a given distribution, [which is set here](https://github.com/ros/rosdistro/blob/master/jazzy/distribution.yaml#L5188-L5202) as  1.6.0. What are you using? 

I'm going to venture to say that the same issue occurs with xtensor - are you building these from source and using the correct version matched to the OS binaries?

javijurado, 2025-03-12:
First of all, 
Thanks @SteveMacenski for your feedback on this. I have spent many hours trying to find out the culprit.

> We use the version of OMPL that is shipped with rosdistro for a given distribution, [which is set here](https://github.com/ros/rosdistro/blob/master/jazzy/distribution.yaml#L5188-L5202) as 1.6.0. What are you using?

We are building almost everything from source and what I noticed after your suggestion is that we are using ompl pointing to main branch. I have changed and I will let you know if that solves the issue

```
ompl/ompl:
    type: git
    url: https://github.com/ompl/ompl.git
    version: main
```

> xtensor - are you building these from source and using the correct version matched to the OS binaries?

xtensor is not build from binaries `DEBIAN_PKG_DEV_DEPENDS+=("libxtensor-dev")`, we just use the package provided by Debian. How do I know which version should I use?

```bash
dpkg -l | grep xtensor
ii  libxtensor-dev:amd64                     0.24.3-1                         amd64        multi-dimensional arrays with broadcasting and lazy computing
ii  xtl-dev                                  0.7.2-2.1                        all          basic tools (containers, algorithms) used for xtensor and xeus
```

javijurado, 2025-03-12:
I have tried also to match the version installed on ros ubuntu jazzy image:
What I did:

1. Git clone repositories xtl, xtensor and xsimd on my src folder:
              - xtl: 0.7.7
              - xtensor: 0.24.7
              - xsimd: 11.2.0
2. Compile: colcon build --packages-up-to nav2_mppi_controller

but still same error
```
In function ‘void xt::detail::shape_computation(RS&, R&, E&, const AX&, std::enable_if_t<(! is_fixed<RS>::value), int>) [with O = xt::reducer_options<float, std::tuple<xt::evaluation_strategy::immediate_type> >; RS = std::array<long unsigned int, 0>; R = xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 0, xt::layout_type::row_major, xt::xtensor_expression_tag>; E = xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 1, xt::layout_type::row_major, xt::xtensor_expression_tag>; AX = std::array<long unsigned int, 1>]’,
    inlined from ‘auto xt::reduce_immediate(F&&, E&&, X&&, O&&) [with F = xreducer_functors<detail::plus, const_value<float>, detail::plus>; E = xtensor_container<uvector<float, xsimd::aligned_allocator<float, 32> >, 1, xt::layout_type::row_major, xtensor_expression_tag>&; X = std::array<long unsigned int, 1>; O = std::tuple<evaluation_strategy::immediate_type>&]’ at /opt/ros/jazzy/include/xtensor/xreducer.hpp:361:45,
    inlined from ‘auto xt::detail::reduce_impl(F&&, E&&, X&&, xt::evaluation_strategy::immediate_type, O&&) [with F = xt::xreducer_functors<plus, xt::const_value<float>, plus>; E = xt::xtensor_container<xt::uvector<float, xsimd::aligned_allocator<float, 32> >, 1, xt::layout_type::row_major, xt::xtensor_expression_tag>&; X = std::array<long unsigned int, 1>; O = std::tuple<xt::evaluation_strategy::immediate_type>&]’ at /opt/ros/jazzy/include/xtensor/xreducer.hpp:961:13,
    inlined from ‘auto xt::reduce(F&&, E&&, EVS&&) [with F = xreducer_functors<detail::plus, const_value<float>, detail::plus>; E = xtensor_container<uvector<float, xsimd::aligned_allocator<float, 32> >, 1, xt::layout_type::row_major, xtensor_expression_tag>&; EVS = std::tuple<evaluation_strategy::immediate_type>&; typename std::enable_if<xtl_requires<is_reducer_options<X>, detail::is_xreducer_functors_impl<typename std::decay<_Tp>::type> >, int>::type <anonymous> = 0]’ at /opt/ros/jazzy/include/xtensor/xreducer.hpp:1045:9,
    inlined from ‘auto xt::sum(E&&, EVS) [with T = void; E = xtensor_container<uvector<float, xsimd::aligned_allocator<float, 32> >, 1, xt::layout_type::row_major, xtensor_expression_tag>&; EVS = std::tuple<evaluation_strategy::immediate_type>; typename std::enable_if<xtl_requires<is_reducer_options<X> >, int>::type <anonymous> = 0]’ at /opt/ros/jazzy/include/xtensor/xmath.hpp:1840:5,
    inlined from ‘void mppi::Optimizer::updateControlSequence()’ at /build/src/ros-navigation/navigation2/nav2_mppi_controller/src/optimizer.cpp:415:51:
/opt/ros/jazzy/include/xtensor/xreducer.hpp:234:43: error: null pointer dereference [-Werror=null-dereference]
  234 |                         result_shape[idx] = expr.shape()[i];
      |                         ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
```
What I saw is this line **cc1plus: all warnings being treated as errors**.
I hope this clarify more the issue

SteveMacenski, 2025-03-12:
For xtensor, we could silence the warnings around the include header that is offending (for example):

```
#pragma GCC diagnostic push
#pragma GCC diagnostic ignored "-Wall"  // Adjust the warning type as needed
#include "problematic_header.h"
#pragma GCC diagnostic pop
```

It might be that the flags in your OS are different than in Ubuntu 20/22/24 so we aren't picking them up in our CI. Since they're in xtensor itself, I don't think we can do much here other than silence them. 

I assume getting the compatible version of OMPL did the trick? 😄

javijurado, 2025-03-13:
> I assume getting the compatible version of OMPL did the trick? 😄
Yes

> It might be that the flags in your OS are different than in Ubuntu 20/22/24 so we aren't picking them up in our CI. Since they're in xtensor itself, I don't think we can do much here other than silence them.

The main difference between ubuntu and debian was gcc/g++. On Ubuntu by default gcc-13 but on debian gcc-12. After installing gcc-13 on my debian container, I was able to compile without any error but system still behave weird. See video attached

https://github.com/user-attachments/assets/7535806a-df6a-4a7c-b123-3f9580659308

javijurado, 2025-03-13:
Another question that I have, I read that you recently replaced xtensor with Eigen. First of all congrats for that big achievement. Second, are those changes portable to `jazzy`??

SteveMacenski, 2025-03-13:
Those changes are not backportable to Jazzy since they are a complete underlying library rewrite and all the ABI/API policies are broken. 

I opened a Jazzy docker image and installed Nav2 from binaries and I don't see this behavior, so I think something is possibly wrong on your end in that development environment. I'm starting to compiling from source before I head to a meeting and will update once that compiles and I'm out of my meeting to check on the source.

SteveMacenski, 2025-03-25:
@javijurado any update?

javijurado, 2025-03-25:
Hi @SteveMacenski , I did some progress on this. Sorry for the delay. First I will answer your questions and second my findings on how to solve the issue

> Maybe you had some cached builds in your install/ and build/ space from before changing to the new compiler version? Did you try to compile cleanly after deleting install/build/log once updating the compiler version?

I tried the same as you did, build from scratch using a jazzy docker image and works perfectly but unfortunately not on my Debian 12 image with ROS. I cleaned the build/install/log folder but same issue :(

The main discrepancy between jazzy and debian docker image was the compiler (gcc) used. On jazzy is gcc13 but on debian gcc12. For some reason gcc12 triggers this error `cc1plus: all warnings being treated as errors` . Already mentioned [here](https://github.com/ros-navigation/navigation2/issues/4979#issuecomment-2720335427). So to be able to compile using gcc12, I have removed from [here](https://github.com/ros-navigation/navigation2/blob/main/nav2_common/cmake/nav2_package.cmake#L40) ` -Wnull-dereference`

Then I decided to install the same versions used by jazzy docker image:
- ompl: 1.6.0
- xtensor: 0.25.0
- xtl: 0.7.5

And that solves the issue

https://github.com/user-attachments/assets/f72a364c-8f14-4b9f-95c1-22f5e8b0e154

Maybe is not the best solution but we can close the issue

Thanks for the support on this @SteveMacenski. I really appreciate your feedback

SteveMacenski, 2025-03-25:
Great, happy you got it all working!

closed 2025-03-25 by SteveMacenski