ros-navigation/navigation2 #4981: SMAC hybrid not following the minimum turning radius set

w-kok, 2025-03-12:
## Bug report
The turning radius is not as expected, I think it is a bug related to the minimum_turning_radius in the SMAC hybrid planner for ROS:Iron.
I looked into the problem, it seems not related to my current code, although I might have missed something that will affect the turning radius.
Thank you in advance for any insight into the problem or the bug.

**Required Info:**
- Operating System:
Docker, Ubuntu 22 
- ROS2 Version:
ROS2 Iron
- Version or commit hash:
sudo apgt install for docker >>   ros-${ROS_DISTRO}-navigation2 

#### Steps to reproduce issue
Load a simulation, e.g. turtlebot, or preferably an Ackermann robot sim. 
Set the planner smac hybrid and inspect the turning radius corresponding to the actual distances on a map.
#### Expected behavior
Turning radius modifications make the planner take bigger turners, or only as small as the minimum turning radius.
The turning radius when set to a radius in meters corresponds to that radius on the map.

#### Actual behavior
Parameter seems to affect the turning, but not by the set radius.

#### Reproduction instructions
Change the minimum_turning_radius parameter from the planner: SMAC hybrid.

#### Additional information
Ackerman cart, fixing the in proper paths through the controller, is not fixing the issue as of yet.

SteveMacenski, 2025-03-12:
Iron's been EOL for some time now, so it wouldn't surprise me if there were bugs that have been resolved in Jazzy, main that are not reflected in Iron since that distribution is end of life. Have you checked against Jazzy and/or main if you see this issue?

Please provide some specific metrics you found, reproduction instructions, parameters, your expected behavior, and so forth. So far, this report isn't enough to do any action / debugging based off of 😄

w-kok, 2025-03-14:
Updated to Jazzy I get the same issue, that the SMAC hybrid planner is not using the set minimum turning radius.

Additionally, to make sure I understand the parameter correctly, do you set the radius of the curve that is part of a corner in a path in meters? Or do you take the length of the vehicle, let's say 2m and set it to 4.0 to get a 2.0m radius for the mimum curve the planner is allowed to make? (Neither do match with the map nor does it match with what the controller takes a minimum turn.)

<name>.minimum_turning_radius, double: 0.4, Description:
Minimum turning radius in meters of vehicle. Also used in the smoother to compute maximum curvature. Must be greater than 0.

I don't see a general simulation to benchmark it on, turtle bot does seem to work, but hard to validate. I need Ackerman settings for a car driving. Keeping the other settings in the yaml file the same and directly from the nav2 doc webpages and just varying the turning radius does showcase the issue. If you want to replicate it.

SteveMacenski, 2025-03-14:
The minimum turning radius is a constraint of an ackermann vehicle based on what it can do due to its drive kinematics. A car's steering wheel can only turn the wheels so much, for example. Visually: when moving the vehicle at that maximum steering angle an driving forward, you get the minimum turning radius of the vehicle in the circle it creates. 

It is not a function of the vehicle's length (well, I mean, it _kind of_ is for a car-like vehicle, but not in the way you mean). Its a function of the kinematic constraint on the vehicle. This may be where you're misunderstanding the planner's behavior. 

For a non-car-like vehicle, you can set that turning radius to be something else based on the behavior you'd like your robot to have and the radius that is performed with doing turns. For a differential drive vehicle for example you have a broad range of values you can use from very small (which turns essentially into grid search at the lowest values) to very large based on the smooth behavior you'd like to set.

SteveMacenski, 2025-03-17:
I don't know what you refer to as the standard definition of turning radius, but you may want to read the Hybrid-A* paper if you need more details: https://ai.stanford.edu/~ddolgov/papers/dolgov_gpp_stair08.pdf

>  I know my cars turning radius, but setting it e.g. to 1.5 the planners takes turns smaller than 1.5 meter when measured from the map.

SteveMacenski, 2025-03-21:
>       minimum_turning_radius: 2.5  # For Hybrid/Lattice nodes: minimum turning radius in m of path / vehicle

Do you see changes in the planner when you change this value? I'm wondering if for some reason these parameters aren't being read.

What are your global costmap settings?

The images are admittedly not diagnostically helpful without scale. What does this topic look like? https://github.com/ros-navigation/navigation2/blob/9f09b064ef1145bbd82aa1e2eb211314ae20347b/nav2_smac_planner/src/smac_planner_hybrid.cpp#L254 maybe the smoother is causing you issues with your large turning radius. I would be _shocked_ if the planner itself is actually at fault since this has been tested and deployed numerous times for a wide range of vehicles - unless you have some particularly odd settings. 

Please provide me a link to your planner playground fork with your map / costmap / planner settings and starting/goal pose that demonstrates the issue

w-kok, 2025-03-23:
>>Do you see changes in the planner when you change this value? 
I'm wondering if for some reason these parameters aren't being read.

Yes, but issues remain. RQT and the param list shows values are set.

>>What are your global costmap settings?

The playground default.

>>The images are admittedly not diagnostically helpful without scale. What does this topic look like?

>"With the rviz "measure" tool, the diameter of the circle created by the turn can be measured." 

The images showcase the paths to replicate the issue and do not uphold the minimum turning radius as mentioned and measured in a couple ways. 

>>What does this topic look like?

Smoother and smooth server are default. 
ros2 topic info /unsmoothed_plan 
Type: nav_msgs/msg/Path
Publisher count: 1
Subscription count: 0

[image]

[The link](https://github.com/w-kok/SMAC_that_planner-playground/tree/SMAC_testing) for the default settings with the aforementioned adjustments in a fork.

w-kok, 2025-03-26:
@SteveMacenski I would say this issue is a bug not a question.

SteveMacenski, 2025-04-11:
Sorry, I've not had the bandwidth to dig into this deeper yet (or for some time coming, unfortunately). Do you see an improvement when using `allow_primitive_interpolation: True` for newer ROS distributions that support it (not Humble - Jazzy or main)? I notice that your param file is rather sparse and not cinluding a large number of the parameter options https://docs.nav2.org/configuration/packages/smac/configuring-smac-hybrid.html. 

Do you see the issue with your curvature in the unsmoothed path topic? That was the question I was asking before 😉 I'm wondering if the default smoother params are breaking for that small of distance between via-points vs turning radius and the smoother would need to be retuned.

SteveMacenski, 2026-01-13:
https://github.com/mini-1235/planner_playground a tool for visualizing the constraint violation (to be verified)

mini-1235, 2026-09-17:
@w-kok @DAG00DAG33 FYI, I have a fix in #6512 that should address this issue. I believe the main cause is a mismatch between the ROS parameter "resolution" and the map resolution (static layer).

If you have a chance, please give it a try and feel free to leave any feedback directly on the PR. Thank you!

closed 2026-09-18 by SteveMacenski