ros-navigation/navigation2 #4984: controller_server crash

federica-ciani, 2025-03-14:
## Bug report

**Required Info:**

- Operating System: 
  - Ubuntu 22.04.5 LTS
- Computer:
  - HP 255 G8 Notebook PC
- ROS2 Version:
  - ROS2 Humble
- Version or commit hash:
  - 507b3659e6c622d9339b61b53810320b600526a2
- DDS implementation:
  - rmw_cyclonedds_cpp

#### Steps to reproduce issue

Sending a lot of laserscan type data the controller.

#### Expected behavior

No crashes

#### Actual behavior

Sending a lot of laserscan type data the controller server crashes.
It happens several times a day

#### Backtrace

[backtrace.txt](https://github.com/user-attachments/files/19246099/backtrace.txt)

#### Possible solution

We solved by adding a mutex nav2_costmap_2d/plugins/obstacle_layer.cpp before the function `projector_.transformLaserScanToPointCloud`.
Could there be any possible side effects?

SteveMacenski, 2025-03-14:
I see it [happening here](https://github.com/ros-navigation/navigation2/blob/085d235db0ef0d189ca19315d7c69a359778fc93/nav2_costmap_2d/plugins/obstacle_layer.cpp#L368-L408). Can you compile with debug flags and try again so we can get a full trace? This tutorial will guide you through it: https://docs.nav2.org/tutorials/docs/get_backtrace.html. 

How do you have this layer configured? Unless you're using a multithreaded executor and either (a) both `laserScanValidInfCallback` and `laserScanCallback` are being called or (b) modified so that the subscriptions are in different callback groups, I don't see how there could be a parallel issue. 

> We solved by adding a mutex in nav2_costmap_2d/plugins/obstacle_layer.cpp before the function projector_.transformLaserScanToPointCloud.

Where else is the mutex locking? Its important for us to understand what parallel resources that protects.

federica-ciani, 2025-03-25:
Thank you for your reply, I will provide the backtrace following your guide.
In the meantime, I will answer you on the other points.
Here the configuration:
```
local_costmap:
    local_costmap:
      ros__parameters:
        always_send_full_costmap: True
        update_frequency: 25.0
        publish_frequency: 25.0
        global_frame: map # TODO by default this is odom
        robot_base_frame: base_link
        rolling_window: True
        width: 10
        height: 10
        resolution: 0.025
        plugins: ["obstacle_layer", "inflation_layer"]
        footprint: '[[-0.36, -0.34], [-0.36, 0.36], [1.0, 0.36], [1.0, -0.34]]'
        lethal_cost_threshold: 100  # new in this file
        map_topic: /map # new in this file
        inflation_layer:
          enabled: True
          plugin: "nav2_costmap_2d::InflationLayer"
          cost_scaling_factor: 10.0
          inflation_radius: 1.5 #0.4
        obstacle_layer:
          combination_method: 1
          enabled: true
          footprint_clearing_enabled: true
          max_obstacle_height: 2.0
          plugin: nav2_costmap_2d::ObstacleLayer
          observation_sources: camera_right camera_left lidarmulti
          camera_right:
            clearing: true
            data_type: LaserScan
            expected_update_rate: 0.0 # new in this file
            inf_is_valid: true # new in this file
            marking: true
            max_obstacle_height: 2.0
            min_obstacle_height: -1.0
            observation_persistence: 0.0 # new in this file
            obstacle_max_range: 2.8 
            obstacle_min_range: 0.2
            raytrace_max_range: 3.5
            raytrace_min_range: 0.0
            sensor_frame: base_link
            topic: /scan_dx
          camera_left:
            clearing: true
            data_type: LaserScan
            expected_update_rate: 0.0 # new in this file
            inf_is_valid: true # new in this file
            marking: true
            max_obstacle_height: 2.0
            min_obstacle_height: -1.0
            observation_persistence: 0.0 # new in this file
            obstacle_max_range: 2.8 
            obstacle_min_range: 0.2
            raytrace_max_range: 3.5
            raytrace_min_range: 0.0
            sensor_frame: base_link
            topic: /scan_sx
          lidarmulti:
            clearing: true
            data_type: LaserScan
            expected_update_rate: 0.0 # new in this file # TODO
            inf_is_valid: true # new in this file # TODO
            marking: true
            max_obstacle_height: 2.0
            min_obstacle_height: -1.0
            observation_persistence: 0.0 # new in this file # TODO
            obstacle_max_range: 4.5
            obstacle_min_range: 0.05
            raytrace_max_range: 5.0
            raytrace_min_range: 0.0
            sensor_frame: base_link
            topic: /lidar_scan_merged
```
Two cameras work at 6 hz and the lidar at 10 hz.
I can confirm that the executor is single threaded and only the `laserScanValidInfCallback` is called, since the `inf_is_valid` parameter is set to `true` for all sensors. Absolutely I did not change the code in any way.
Furthermore, I can state that with this configuration and data from these sensors the crash occurs every few hours. Whereas by increasing the number of sources and sending laserscans via script the crash occurs more often.

SteveMacenski, 2025-03-25:
I'll prefix this with no crash should happen so its worth understanding why and fixing that. 

However, these are a bit unusual for a local costmap - I'm curious to know the reasons for it. If your lidar and cameras only update at 10, 6 hz respectively, why update the costmap at 25hz? Also, why is the global frame set to map, that is not a smooth or continuous control frame. 

```
        update_frequency: 25.0
        publish_frequency: 25.0
        global_frame: map # TODO by default this is odom
```

>    We solved by adding a mutex in nav2_costmap_2d/plugins/obstacle_layer.cpp before the function projector_.transformLaserScanToPointCloud.

I'd like to see that code to see exactly what parts of it are protected by that mutex to try to debug. The compiling with debug flags (which you may want to include laser geometry with) will tell us a more complete story about where its happening to try to find where the issue lies. ASAN might also be helpful.

I'm curious if you're able to test on Jazzy or `main` and see if you can reproduce this error. It may very well be that we fixed this in mainline that was not backported to Humble. If that's the case, then we can find in the git logs what change fixed it and apply to humble.

federica-ciani, 2025-03-28:
Here the full trace:

[backtrace.txt](https://github.com/user-attachments/files/19501072/backtrace.txt)

Here the lock guard I added to prevent simultaneous access to `transformLaserScanToPointCloud` function.

https://gist.github.com/federica-ciani/066b82bc3fa779c2d0609c2e84786ed7

With this the crash no longer occurred.

SteveMacenski, 2025-04-01:
I don't think you compiled with `-pg`  since we still don't see a line in the library `liblayers.so` that is failing. 

```
#10 0x00007afda87989e4 in nav2_costmap_2d::ObstacleLayer::laserScanValidInfCallback(std::shared_ptr<sensor_msgs::msg::LaserScan_<std::allocator<void> > const>, std::shared_ptr<nav2_costmap_2d::ObservationBuffer> const&) () at /home/fede/sedia_ws/install/nav2_costmap_2d/lib/liblayers.so
```

> [controller_server-1] [INFO] [1743151705.395606829] [ObstacleLayer]: Thread ID: 14887398069642844755 - Call to transformLaserScanToPointCloud
> [controller_server-1] [INFO] [1743151705.404523221] [ObstacleLayer]: Thread ID: 27545833675783143 - Call to transformLaserScanToPointCloud

Who are the 2 threads? I'm at a loss to identify them in https://github.com/ros-navigation/navigation2/blob/main/nav2_costmap_2d/plugins/obstacle_layer.cpp. They should be using the same callback group for all layers (see refs below) and if it were the case that each data source had its own thread due to some internal `message_filter` oddity, we should see 3x thread IDs, not 2x, since you have 3x data sources (all 3 are publishing, right?).

Putting that lock there is problematic because that means there's a bug in how things are processing, these should _not_ be parallel processed! 

[1] https://github.com/ros-navigation/navigation2/blob/main/nav2_costmap_2d/plugins/obstacle_layer.cpp#L131-L132
[2] https://github.com/ros-navigation/navigation2/blob/main/nav2_costmap_2d/plugins/obstacle_layer.cpp#L238-L239
[3] https://github.com/ros-navigation/navigation2/blob/08268f9922f92511f12ad9b951148d1c1d30096b/nav2_costmap_2d/src/costmap_2d_ros.cpp#L181-L182

SteveMacenski, 2025-04-14:
@federica-ciani do you have a sense of where that's coming from yet?

padhupradheep, 2025-05-08:
@federica-ciani: did you find what's causing the issue? if you need some help, let us know. We would love to get this one fixed

SteveMacenski, 2025-09-18:
Closing due to lack of response