ros-navigation/navigation2 #4990: Replace `OpaqueFunction` + `ParseMultiRobotPose` with `launch.actions.ForEach`

christophebedard, 2025-03-15:
## Feature request

#### Feature description

An `OpaqueFunction` is used with [`ParseMultiRobotPose`](https://github.com/ros-navigation/navigation2/blob/abcd8fe6785cc9f7ce0ba2e7de1cb2989475f307/nav2_common/nav2_common/launch/parse_multirobot_pose.py) to repeat entities based on values in a launch argument. For example, see [`cloned_multi_tb3_simulation_launch.py`](https://github.com/ros-navigation/navigation2/blob/abcd8fe6785cc9f7ce0ba2e7de1cb2989475f307/nav2_bringup/launch/cloned_multi_tb3_simulation_launch.py#L240)

As of https://github.com/ros2/launch/pull/802, `launch` has a [`ForEach` action](https://github.com/ros2/launch/blob/151b024a7cfac06f0d56798cbe87ab180b9ea3ab/launch/launch/actions/for_loop.py) to do the same thing. See [this ROS Discourse comment](https://discourse.ros.org/t/requesting-feedback-new-for-loop-action-for-ros-2-launch-files-to-repeat-entities/42021/5).

Update Nav2 to use `ForEach` instead of `OpaqueFunction` + `ParseMultiRobotPose`.

#### Implementation considerations

Steps:

1. In every launch file where `OpaqueFunction` + `ParseMultiRobotPose` are used:
    1. Change the launch file to use a `ForEach` action instead of an `OpaqueFunction`
    2. Make the existing `generate_robot_actions()` function the `ForEach` action's `function`, and turn the values parsed by `ParseMultiRobotPose` into parameters of the function, with default values wherever appropriate
    3. Provide the `LaunchConfiguration()` to the `ForEach` action instead of `ParseMultiRobotPose`
2. Consider deprecating `ParseMultiRobotPose`

christophebedard, 2025-03-15:
I opened this intending for it to be a "good first issue" for a new contributor, but I can also take care of it if no one ends up picking it up.

SteveMacenski, 2025-03-16:
Agreed! I added those labels. Let’s see!

leander-dsouza, 2025-03-17:
I would like to work on this issue, Steve :)

SteveMacenski, 2025-03-17:
Great, thanks!

closed 2025-03-21 by SteveMacenski