ros-navigation/navigation2 #4991: Lattice generator hangs on big turning radius

w-kok, 2025-03-17:
## Bug report

**Required Info:**

- Operating System:
Linux server/Ubuntu/Kubuntu

#### Steps to reproduce issue
1.
config.json:
{
    "motion_model": "ackermann",
    "turning_radius": 4.5,
    "grid_resolution": 0.05,
    "stopping_threshold": 5,
    "num_of_headings": 16
}
2.
python3 generate_motion_primitives.py --config config.json 
3.
Output.json as a lattice file for SMAC lattice planner.

#### Expected behavior
Gives lattice output file.

#### Actual behavior
The script freezes or hangs on something.

#### Reproduction instructions
1 > 2 > 3 

#### Additional information
Default config, with just a big turning radius.

SteveMacenski, 2025-03-17:
How long are you waiting before stopping the script? It may be that it is still working since you're asking it for pretty big maneuvers, the minimum control set may be quite large based on curvature minimization in this case.

w-kok, 2025-03-24:
Hours/a day without resolve it hangs on good systems/machines. 
I have some fix for compute effort resolving up to like 2.5m but after that it doesn't seem to find solutions.
I can use a smaller turning radius, but as an example to see the bug you can try the following config.

motion_model "ackermann"
turning_radius 4.5
grid_resolution 0.01
stopping_threshold 15
num_of_headings 32

SteveMacenski, 2025-03-24:
Your costmap is 1cm? That is very fine resolution for global planning / mapping, is that needed? I think that's more the issue than 4.5m since the computation is proportionate to costmap resolution to radius ratio. That's eq. to a 22.5m request to a standard 5cm resolution costmap. That search space is very large and the stop threshold / trajectory condensation might not be appropriately stopping with that huge of a space. 

>  I have some fix for compute effort resolving up to like 2.5m but after that it doesn't seem to find solutions.

Can you open a PR for that?

w-kok, 2025-04-02:
Thanks for the detailed explanation!

Just to clarify—are you suggesting that if a vehicle has a minimum turning radius of, say, 1 meter, then it shouldn't be able to reach a position within 1 cm, given a lattice file with 1cm resolution, operate on a costmap with a resolution of 1 cm? Or are you saying the costmap resolution should be coarser and more aligned with the resolution of the lattice file (setting it to also a lower resolution) used by the planner?

I'm a bit confused by that idea, since the resolution set in the lattice file defines the arc discretization for the planner itself, not the resolution of the costmap. While the planner uses the global costmap for collision checks and general planning, the arc resolution is more about how finely we sample motion primitives, not directly tied to the costmap resolution.

In my understanding, the sampling distance or resolution for trajectory generation should not be constrained by the costmap resolution. Instead, the lattice resolution is chosen based on the planner's needs and the vehicle's dynamics—not the grid granularity of the map. So tying similarity comparisons directly to costmap resolution (especially at 1 cm) could lead to overly conservative behavior, especially when the minimum turning radius is large.

Let me know if I'm missing something, but that's how I currently see the separation of concerns here.

SteveMacenski, 2025-04-02:
The costmap resolution isn't necessarily related to the accuracy tolerances of goal achievement. Its more related to the coarseness / fineness needed for collision avoidance activities, and the resolution that is 'reasonable' for your underlying mapping technology (if using mapping rather than GPS localization). 

You may have a costmap resolution of say, 5cm for example, but be able to achieve using the Smac Planner a fully accurate path to the goal pose that is exact (using Smac Planner's analytic expansions, this is what it does. It finds the exact solution, if one is possible). The Smac Planners while using the grid are not themselves bound in its accuracy by the grid resolution. 

---

The lattice resolution should be set to the costmap resolution. The costmap resolution is set to be aligned with your SLAM map, in general (but not absolutely; but that's another subject for another time). If not using a SLAM map with GPS localization, you can set it essentially arbitrarily according to your robot and environment. But unless you're in highly confined settings where the robot is right up against obstacles that needs 1cm level understanding of the space, 1cm is very fine. 5cm is a pretty general purpose good starting point. 

But, the resolution doesn't define the accuracy of the planning request. The planning uses the world continuous coordinates. So even if you set a resolution of say 10 meters resolution (very very coarse), you could still generate a plan accurate to exactness with analytic expansions. 

The resolution of the lattice however is also used to define the resolution of the path-points it generates along the curve, so that can impact collision checking coarse/fine-ness. Having that aligned with the costmap resolution is a pretty good bet since that is the resolution of the collision-space that you're checking against.

w-kok, 2025-04-02:
I would like to use the resolution of 1 cm (it can be a bit bigger) with a turning radius of 1m or 1.5m. If I understand correctly that is not possible with the current software?

> "The resolution of the lattice however is also used to define the resolution of the path-points it generates along the curve, so that can impact collision checking coarse/fine-ness."

Assuming that I can use a more coarse costmap (~5cm) which gives a close to optimal path with a tuned planner, can you elaborate on the impact on collision checking? For the continuous map with the static costmap combined with some inflation which both will be the costmap that will have different values only every resolution step of 5cm. Lethal cost is still checked for the final path planning etc. with the precision of the planner not the generated lattice nor the costmap? Or are they a values based on the resolution and with updated values after each step/resolution of 5cm. Visual inspection doesn't quite tell me the answer.

See also the currently more pressing related [other issue,](https://github.com/ros-navigation/navigation2/issues/4981) as the lattice planner was an attempt to force a bigger turning radius to avoid the hybrid issues.  The lattice planner suffers from similar issues. Can the lattice arc be cut short and used as a partial curve? That might explain why also the lattice is able to generate invalid solutions.

SteveMacenski, 2025-04-11:
> If I understand correctly that is not possible with the current software?

I think its possible, but it sounds like from the behavior you're describing, we need to refine a bit the similarity comparison to be proportional to turning radius and costmap resolution so we terminate faster with a more reasonable set of potential trajectory candidates.

> can you elaborate on the impact on collision checking?

You'd check every costmap cell, so now you'd check every 5cm rather than 1cm.