ros-navigation/navigation2 #4998: Create a Humble distribution compatible version of the main (Rolling) branch

SteveMacenski, 2025-03-21:
In https://github.com/ros-navigation/navigation2/commit/9f09b064ef1145bbd82aa1e2eb211314ae20347b I enabled Jazzy distribution to work on the `main` branch by using preprocessor macros and ROS 2 versions to set the API differences of underlying libraries.

However, that is not possible for Humble due to the sweeping changes made in the CMakeLists to modernize it. That would involve a near complete duplication in the CMakeLists that is unrealistic to maintain and support. There are some other API changes to be made w.r.t. service QoS that are pretty simple to resolve with `[get_rmw_qos_profile](https://docs.ros2.org/foxy/api/rclcpp/classrclcpp_1_1QoS.html#abe5cb7d1ee2f6874e03f0c02937a3194)` (or similar).

So, I think it would be useful to have a `humble_main` branch that is the `main` branch with the humble-compatible ROS 2 APIs and old-version of CMake. This can facilitate users on Humble to leverage more modern capabilities of Nav2 and/or test changes for PRs made to `main` and `humble` without using the `rolling` distribution that may not be installed on their development machines. 

This is a good first issue for users:
- Clone `main` and build with ROS 2 Humble
- Address compiler errors until completed
  * Many will relate to CMakeLists that things like `tf2_geometry_msgs::tf2_geometry_msgs` library doesn't exist. See the current `humble` branch method of linking against ROS libraries using `ament_target_dependencies` and the `dependencies` variable
  * Others will relate to service clients requiring in RMW QoS profile rather than an `rclcpp::QoS` object. That can be resolved easily with `.get_rmw_qos_profile()`
- Open a PR for a review!

XueSongTap, 2025-03-21:
Can I work on this？

Dhagash4, 2025-03-21:
There is also PoseStampedArray/PoseStamped[] which has been changed to Goals now. How do you want to handle that? Fix with the humble API or by using macros and preprocessor?

SteveMacenski, 2025-03-21:
@XueSongTap  Please! 

> There is also PoseStampedArray/PoseStamped[] which has been changed to Goals now. How do you want to handle that? Fix with the humble API or by using macros and preprocessor?

I just backported them this week :-) https://github.com/ros2/common_interfaces/pull/272 It sounds be available now if you build that package from source in your workspace (binaries to come in coming weeks)

XueSongTap, 2025-03-23:
Hi @SteveMacenski,

 Should we be creating a new branch (like "rolling_humble" or "humble_main") in the official repository that would be maintained alongside the main branch? 

If a new branch is needed, is there a way I could create it myself, or would you prefer to set it up? I'm happy to follow whatever process works best for the team.

Thank you for your guidance!

Dhagash4, 2025-03-24:
I am also trying to backport, I noticed couple of inconsistencies, which are following

- Expose `subscription_callback` as virtual function was not backported to humble in `tf2_ros`, opened [PR](https://github.com/ros2/geometry2/pull/772)
- Then `spin_all` shortcut in `rclcpp` is not available in humble branch, need to [backport](https://github.com/ros2/rclcpp/commit/a17d26b20ac41cc9d5bf3583de8475bb7c18bb1e) this PR!

I hope, I am understanding this correctly! @SteveMacenski if you can clarify this would be great!

SteveMacenski, 2025-03-24:
I created a new branch for this work to PR against: `humble_main` that is the `main` branch that the backporting for `humble` compilation should be made!

> Expose subscription_callback as virtual function was not backported to humble in tf2_ros, opened https://github.com/ros2/geometry2/pull/772

I think that util should simply have the cmake for compiling it commented out, I knew about that. There's no solution for that other than not to compile it

> Then rclcpp::spin_all shortcut in rclcpp is not available in humble branch, need to [backport](https://github.com/ros2/rclcpp/commit/a17d26b20ac41cc9d5bf3583de8475bb7c18bb1e) this PR!

Spin some should do for that BT node (I think its just one spot)!

MaciejGroszyk, 2025-03-25:
Hello, I made a PR #5017. There are still unsolved issues and require further work, but it is building successfully.

SteveMacenski, 2025-05-14:
Merging imminent

closed 2025-05-14 by SteveMacenski