ros-navigation/navigation2 #4999: I need help using the waypoint_follow function in humble.

DongDong-Juice, 2025-03-21:
**Required Info:**

- Operating System:
  ubuntu 22.04
- Computer:
  desktop
- ROS2 Version:
   humble binaries
- Version or commit hash:
   dpkg-query --show "ros-$ROS_DISTRO-nav2-*" 
   -> 
"""
              ros-humble-nav2-amcl	1.1.18-1jammy.20250207.201743
              ros-humble-nav2-behavior-tree	1.1.18-1jammy.20250207.201742
              ros-humble-nav2-behaviors	1.1.18-1jammy.20250210.223656
              ros-humble-nav2-bringup	1.1.18-1jammy.20250210.230152
              ros-humble-nav2-bt-navigator	1.1.18-1jammy.20250210.223033
              ros-humble-nav2-collision-monitor	1.1.18-1jammy.20250210.222802
              ros-humble-nav2-common	1.1.18-1jammy.20250207.195657
              ros-humble-nav2-constrained-smoother	1.1.18-1jammy.20250210.223441
              ros-humble-nav2-controller	1.1.18-1jammy.20250210.223035
              ros-humble-nav2-core	1.1.18-1jammy.20250210.222759
              ros-humble-nav2-costmap-2d	1.1.18-1jammy.20250210.220811
              ros-humble-nav2-dwb-controller	1.1.18-1jammy.20250210.224143
              ros-humble-nav2-lifecycle-manager	1.1.18-1jammy.20250210.215820
              ros-humble-nav2-map-server	1.1.18-1jammy.20250207.201742
              ros-humble-nav2-mppi-controller	1.1.18-1jammy.20250210.224502
              ros-humble-nav2-msgs	1.1.18-1jammy.20250207.200716
              ros-humble-nav2-navfn-planner	1.1.18-1jammy.20250210.223026
              ros-humble-nav2-planner	1.1.18-1jammy.20250210.223023
              ros-humble-nav2-regulated-pure-pursuit-controller	1.1.18-1jammy.20250210.223025
              ros-humble-nav2-rotation-shim-controller	1.1.18-1jammy.20250210.223616
              ros-humble-nav2-rviz-plugins	1.1.18-1jammy.20250210.220742
              ros-humble-nav2-simple-commander	1.1.18-1jammy.20250207.201213
              ros-humble-nav2-smac-planner	1.1.18-1jammy.20250210.223026
              ros-humble-nav2-smoother	1.1.18-1jammy.20250210.223036
              ros-humble-nav2-theta-star-planner	1.1.18-1jammy.20250210.223028
              ros-humble-nav2-util	1.1.18-1jammy.20250207.201211
              ros-humble-nav2-velocity-smoother	1.1.18-1jammy.20250207.202134
              ros-humble-nav2-voxel-grid	1.1.18-1jammy.20250207.200358
              ros-humble-nav2-waypoint-follower	1.1.18-1jammy.20250210.223038
  """

#### Tuning / Configuration Goal

"Implement the waypoint follow function in ROS Humble using the Nav2 package with a Python source code. Or plans to use" 

#### What I've Tried
"I ran the launch file and used the Nav2 plugin in RViz to test the waypoint follow function, but I kept encountering execution errors with the example_waypoint_follow.py file inside the nav2_sample_commander package while trying to implement it in source code.

To resolve this issue, I searched through the documentation and various resources, but I couldn't find any information on the execution procedure.

I want to know how to use the example and how to start the required services.

I tried running ros2 run nav2_waypoint_follower waypoint_follower, but it didn't work. I don't know why the waypoint_follow action server is not running. Please help me."

Related logs 
waypoint_follow 
[INFO] [1742464485.361931051] [waypoint_follower]: 
	waypoint_follower lifecycle node launched. 
	Waiting on external lifecycle transitions to activate
	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[INFO] [1742464485.362217325] [waypoint_follower]: Creating  
[INFO] [1742465032.037666441] [rclcpp]: signal_handler(signum=2)
[INFO] [1742465032.037806458] [waypoint_follower]: Running Nav2 LifecycleNode rcl preshutdown (waypoint_follower)
[INFO] [1742465032.037903073] [waypoint_follower]: Destroying bond (waypoint_follower) to lifecycle manager.

[INFO] [1742464485.362217325] [waypoint_follower]: Creating  
At this point, I expect the action to come up in the status. However, it doesn't.

[image]

**The executable I run**
# Copyright (c) 2021 Samsung Research America
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.

import os

from ament_index_python.packages import get_package_share_directory

from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument, ExecuteProcess, IncludeLaunchDescription
from launch.conditions import IfCondition
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch.substitutions import LaunchConfiguration, PythonExpression
from launch_ros.actions import Node

def generate_launch_description():
    # warehouse_dir = get_package_share_directory('aws_robomaker_small_warehouse_world')
    nav2_bringup_dir = get_package_share_directory('nav2_bringup')
    python_commander_dir = get_package_share_directory('nav2_simple_commander')

    map_yaml_file = "/home/dong/robot_web_path_plane/robot/turtlebot4/src/path_plan/map/map4.yaml"
    world = os.path.join(python_commander_dir, 'warehouse.world')

    # Launch configuration variables
    use_rviz = LaunchConfiguration('use_rviz')
    headless = LaunchConfiguration('headless')

    # Declare the launch arguments
    declare_use_rviz_cmd = DeclareLaunchArgument(
        'use_rviz',
        default_value='False',
        description='Whether to start RVIZ')

    declare_simulator_cmd = DeclareLaunchArgument(
        'headless',
        default_value='False',
        description='Whether to execute gzclient)')

    # # start the simulation
    # start_gazebo_server_cmd = ExecuteProcess(
    #     cmd=['gzserver', '-s', 'libgazebo_ros_factory.so', world],
    #     cwd=[warehouse_dir], output='screen')
    #
    # start_gazebo_client_cmd = ExecuteProcess(
    #     condition=IfCondition(PythonExpression(['not ', headless])),
    #     cmd=['gzclient'],
    #     cwd=[warehouse_dir], output='screen')

    urdf = os.path.join(nav2_bringup_dir, 'urdf', 'turtlebot3_waffle.urdf')
    start_robot_state_publisher_cmd = Node(
        package='robot_state_publisher',
        executable='robot_state_publisher',
        name='robot_state_publisher',
        output='screen',
        arguments=[urdf])

    # start the visualization
    rviz_cmd = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(
            os.path.join(nav2_bringup_dir, 'launch', 'rviz_launch.py')),
        condition=IfCondition(use_rviz),
        launch_arguments={'namespace': '',
                          'use_namespace': 'False'}.items())

    # start navigation
    bringup_cmd = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(
            os.path.join(nav2_bringup_dir, 'launch', 'bringup_launch.py')),
        launch_arguments={'use_sim_time':"False",'autostart':"True",'map': map_yaml_file}.items())

    # start the demo autonomy task
    demo_cmd = Node(
        package='nav2_simple_commander',
        executable='example_waypoint_follower',
        emulate_tty=True,
        output='screen')

    ld = LaunchDescription()
    # ld.add_action(declare_use_rviz_cmd)
    # ld.add_action(declare_simulator_cmd)
    # ld.add_action(start_gazebo_server_cmd)
    # ld.add_action(start_gazebo_client_cmd)
    ld.add_action(start_robot_state_publisher_cmd)
    # ld.add_action(rviz_cmd)
    ld.add_action(bringup_cmd)
    ld.add_action(demo_cmd)
    return ld

---------
Can you give me a guide or how-to on how to do this? The docs didn't cover what I was looking for.

DongDong-Juice, 2025-03-21:
I am not good at English and I am also lacking in ros2. Any opinions or suggestions are welcome. However, please explain in more detail or kindly.

SteveMacenski, 2025-03-21:
Please ask on Robotics Stack Exchange with your error messages / logs and what you expect to have happen

closed 2025-03-21 by SteveMacenski