ros-navigation/navigation2 #5004: Achieving Precise Goal Orientation with ROS2 Nav2 on a Differential Drive Robot

Hitesh-waykole, 2025-03-22:
- Operating System:
  - UBUNTU 22.04
- Computer:
  - Intel NUC i7
- ROS2 Version:
  - Humble

- DDS implementation:
  - FastDDS
- Nav2 Package:
  - DWB
 
Configuration Goal
Achieve goal orientation at goal.
I am working with a differential drive mobile robot using ROS2 Nav2 in an industrial environment. I have successfully achieved 10 mm positional accuracy at the goal pose. However, I am facing challenges with achieving precise goal orientation.
Behaviour -
Robot successfully navigates to the target X,Y position. 
At the goal location, the robot attempts to achieve the target orientation. 
Instead of smoothly rotating to the desired angle, it rotates completely.
Navigation task eventually aborts.

What I've Tried
1)Parameter Tuning: Adjusting xy_goal_tolerance and yaw_goal_tolerance values. Tuned controller parameters (such as angular velocity limits).

2)I have used DWB controller.

SteveMacenski, 2025-03-24:
> I have successfully achieved 10 mm positional accuracy at the goal pose. 

Hey, that's pretty neat!

> However, I am facing challenges with achieving precise goal orientation.

Have you set the parameter `stateful` to true so that once it meets positional goal tolerances it'll just rotate to heading? I'm not sure what you mean by "it rotates completely". 

Can you share your parameters?

Hitesh-waykole, 2025-03-26:
> Have you set the parameter `stateful` to true so that once it meets positional goal tolerances it'll just rotate to heading?

Yes the stateful is set to True

> I'm not sure what you mean by "it rotates completely".

It used to rotate in counterclockwise direction .
Now the behviour is that it rotates to and fro 

> "it rotates completely".

By "it rotates completely," I mean that when the robot reaches the goal position, instead of aligning smoothly to the desired orientation, it performs unnecessary full spins and eventually aborts the navigation task.

But now i have fixed the entire rotation issue.
it now oscillates back and forth while attempting to achieve the target orientation. It turns slightly to the left, then to the right, and repeats this motion instead of smoothly aligning to the final heading.

[nav2_params.txt](https://github.com/user-attachments/files/19461594/nav2_params.txt)

I am also facing inconsistencies in achieving repeatable positional accuracy. I suspect the cause might be drift accumulation over time in sensor data.
I am using IMU + wheel encoder + LiDAR data for EKF localization.
I have a script that periodically clears the sensor cache and removes older values, keeping only the most recent ones to reduce drift accumulation.
Could you please share your insights on improving long-term positional accuracy , repeatability and whether the cache-clearing approach is recommended, or if there are better techniques to mitigate drift?

Above I have attached nav2_params file.

SteveMacenski, 2025-03-26:
Can you show me a video of this behavior? I'm not clearly understanding you about the issue in your description since it sounds like its rotating continuously (?) but you were able to fix it or into a new problematic behavior (?). What were the settings / code for your first described issue and the newly described issue?

Hitesh-waykole, 2025-03-27:
> but you were able to fix it or into a new problematic behavior

Extremely sorry for not explaining the issue accurately.

Previously on reaching the xy goal the robot used to rotate in order to achieve the goal orientation. 
Currently I have reduced the behaviour partially. 
Now the robot on the reaching the xy goal it oscillates back and forth—it rotates slightly left, then slightly right—repeating this motion as it tries to achieve the exact goal orientation. 

> What were the settings / code for your first described issue and the newly described issue?
Partially fixed it by tuning the param file.

I want to entirely stop this behaviour , the robot should achieve the goal orientation smoothly.

> Could you please share your insights on improving long-term positional accuracy , repeatability and whether the cache-clearing approach is recommended, or if there are better techniques to mitigate drift?

SteveMacenski, 2025-03-27:
> the robot should achieve the goal orientation smoothly

I don't understand how your description of the initial behavior doesn't accomplish that: _Previously on reaching the xy goal the robot used to rotate in order to achieve the goal orientation._

Maybe worth looking at the rotation shim controller, it has a check on the goal checker and when the positional tolerance is met, it rotates to heading, bypassing the DWB / other controllers more complex operation and just only a smooth turn. This was implemented for similar reasons as you're finding: its hard to tune reactive and predictive controllers to both do what you want them to do outside of path tracking and then perform simple in-place rotations with the same configurations. You can easily do it with custom critic functions, but this is far more straight forward and not tuning-based.

Hitesh-waykole, 2025-03-28:
> I don't understand how your description of the initial behavior doesn't accomplish that: _Previously on reaching the xy goal the robot used to rotate in order to achieve the goal orientation._

Yes , the robot used to rotate in order to achieve goal orientation , but it wasnt able to achieve it, and since it wasnt able to reach , Nav2 ends up aborting .

> Maybe worth looking at the rotation shim controller

Ok will update you after trying. 

Thank You.

Hitesh-waykole, 2025-03-31:
@SteveMacenski  Thank you for your time . Issue is resolved . 
Although i need your opinion on a topic.
My robot needs to follow my custom given path strictly and stop at certain given waypoints. 
For this i am using Navigatethroughposes for following the path & Navigatetowaypose for stopping at a waypoint.
I am switching action using a script.
Previously i used FollowPath & FollowWaypoint action for this task. 

Is this approach valid and will let me achieve the task ?

SteveMacenski, 2025-03-31:
>  My robot needs to follow my custom given path strictly and stop at certain given waypoints.

If you want to stop at the waypoints, then the NavigateThroughPoses isn't appropriate since it will navigate _through_ those waypoints, not stop at them. The Waypoint Follower node will do a more 'stop at a waypoint -> trigger some action -> move on' type action, though if you want more flexibility in the navigation task execution and fallback, a similar feature could be easily developed in the behavior tree where you can manage the goals and their associated tasks. The independent server is more for demonstration purposes of the Nav2 high-level action API and useful for alot of MVP, research, or student-level needs.

All ways are valid, it just depends on what you need! 

Closing due to resolving the issue, I'm glad that worked out for you!

closed 2025-03-31 by SteveMacenski