ros-navigation/navigation2 #5011: Code Quality Tooling

Nils-ChristianIseke, 2025-03-24:
## Feature request
I would like to discuss the introduction of an code quality tool to navigation_2 (e.g. [sonarqube](https://www.sonarsource.com/products/sonarqube/)) or an alternative.

#### Feature description
Automatically checking:
-  duplicated code
- coding standards
- code coverage
-  code complexity
- comments
- bugs
- security 

#### Implementation considerations
I would run this as non-mandatory part of circleci. Reviewers / developers who are interested in the analysis can look at it, others can ignore it.

SteveMacenski, 2025-03-24:
Can you highlight some specific parts of this that you'd like to integrate and its value? I'm a little unclear about the offerings listed on the website (AI coding assistants) and the list of features you list here that it provides. Also, do you use this or a similar thing already in your org that you're happy with, or is this a new / experimental addition for this project? 

This tool also does not appear to have any free tier. I don't mind Open Navigation paying for software that adds value, but that value add has to be high enough that its worthwhile.

Nils-ChristianIseke, 2025-03-24:
I am using sonarqube  my organization, and overall i am happy with it.
By introducing it we were able to improve the code quality on legacy code and it also repeatedly points out quality problems in PRs that are not caught by standard linters.

But I also know that I really like automatic feedback on code and, if in doubt, I run one more tool ;). 

I had in mind that sonarqube was free for open source projects, but apparently that only applies to a few users per project. 
That said, if you are interested, I can see if there is something comparable that is free for open source projects, and just open a PR that shows practically what is marked by the tooling on the  nav2 code. And then you can decide if you think nav2 would benefit from such tooling.
Let me know if you'd like me to proceed with this—otherwise, I’m happy to close this issue.

Nils-ChristianIseke, 2025-05-14:
> @Nils-ChristianIseke any updates here?

No, hope i will find some time for it soon!

mini-1235, 2025-09-19:
When I was working on https://github.com/ros-navigation/navigation2/pull/5391, I was thinking about clang-tidy, since it was already available in the ament_lint repository, and I believe it can actually check for:
>duplicated code
coding standards
code coverage
code complexity
comments
bugs
security

as requested in this ticket

Moveit actually uses clang-tidy https://github.com/moveit/moveit2/blob/main/.clang-tidy, and they have a ticket for cpp best practice https://github.com/moveit/moveit2/issues/840. I tested their rules on velocity smoother: https://gist.github.com/mini-1235/e334ca280f3f440b0694c50ffd4019c5. That said, if we want to apply this across our repo, it would likely mean modifying thousands of lines.