ros-navigation/navigation2 #5015: [Docking] Create API to enable / disable detector

SteveMacenski, 2025-03-24:
This would be useful so that the detector running for docking (which may be very computationally expensive, like AI) doesn't need to be always running when not in a docking situation. You can enable it when starting to dock and disable it once done - then the detector can be a task server instead of an always-on node (optionally)

ayushkumar8340, 2025-03-29:
Hi,
I am interested in working on this problem statement. Could you please assign this issue to me and provide further details?

SteveMacenski, 2025-03-31:
@ayushkumar8340 thanks for showing interest!

I think what we want to do here is to:
* Update the Dock Plugin API to include a new "startDetection()" "stopDetection()" methods 
* Update documentation in the readme and nav2 website about them
* Implement these 2 functions in the 2 plugins: Set a service name parameter to enable and disable the task server doing dock detections, with the dock type name as a request field (in case multiple dock types, to enable detections on the the current right one). If none provided (make the default empty string), then we skip and don't do anything, meaning the detector is always on

That would allow the explicit start/end of detections when they're needed for docking in case a high computation cost detector is being used (like AI) 

Does that make sense? 😄

ayushkumar8340, 2025-04-01:
Yup got it. I'll keep you posted about the updates.

ayushkumar8340, 2025-04-01:
I had a few doubts regarding the implementation of this problem statement :

1. As you mentioned, there should be  a mechanism to turn on/off the task server doing docking detection but when i went through the code i found out that both the plugins: simple charging dock and simple non-charging dock subscribes to a topic to get the dock pose and there no such mechanism to trigger the detection node as it is a separate package. 

[image]
Source : https://github.com/open-navigation/opennav_docking/tree/ubr1_testing

So do you want me to add some logic inside the detection node but according to me that would not be a good and scalable solution.
Or do I need to use the std::system() function to start a process (the detector node) and then kill it using the same.

I don't think both the solutions are good enough. What do you recommend ?

2. Does the service which i have to create will have 1 field (Just confirmation): 
a. type -> dock plugin type to enable

SteveMacenski, 2025-04-01:
1. This is true for the current detectors, but not all detectors are that way. Many are based on AI solutions which are heavy to run continuously and should only be enabled when required. 

However, as Fergs points out, we can still subscribe and unsubscribe from that topic to automatically start/stop the detection within our new APIs (rather than calling a service) to do the same for the case of the Apriltag detector. The way I'm suggesting you implement this has service calls (if configured) to enable / disable -- but also would be good to subscribe / unsubscribe as well for those that implement lazy publishers instead! The service calls using a `std_srvs/Trigger` - if the `service_name` is set as an available option. Otherwise, just subscribe / unsubscribe

2. Maybe actually use a `std_srvs/Trigger`. We can have the user set the `service_name` to be matched to the proper detector server interface. We should just put that into the docs.

SteveMacenski, 2025-05-20:
@ayushkumar8340 any updates? :-) Its been a little while and I think this should be relatively straight forward

SteveMacenski, 2025-05-28:
Please! I appreciate it thanks!

commit 4d05d23c (2025-09-16): feat(opennav_docking): add on-demand detector client & unit tests for dock plugins (#5015) (#5218)

* feat(opennav_docking): Add dynamic lifecycle for external detectors

This change introduces an API to dynamically enable and disable external
docking detectors, such as those based on AI, to conserve system
resources when not actively docking.

Key changes:
- Added `startDetectionProcess()` and `stopDetectionProcess()` to the
  docking plugin interface (`ChargingDock` and `NonChargingDock`).
- Implemented this logic in `SimpleChargingDock` and
  `SimpleNonChargingDock` using a `std_srvs/Trigge [... cut]
closed 2025-09-16 by SteveMacenski