ros-navigation/navigation2 #5020: Missing Parameters in MPPI Controller

YBachmann, 2025-03-25:
## Bug report

I noticed that the MPPI controller_server is missing some parameters that are listed in the [nav2 documentation](https://docs.nav2.org/configuration/packages/configuring-mppic.html). 
For example all the acceleration settings (`ax_max`, `ax_min`, `ay_max`, `az_max` are missing). 

Also some parameters that one would expect to exist are neither listed in the documentation nor do they actually exist (for example `ay_min` should exist in my opinion, especially for omnidirectional robots). 

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Computer:
  - i5-9300H
- ROS2 Version:
  - Humble binaries
- Version or commit hash:
  - ros-humble-navigation2	1.1.16-1jammy.20240830.231501
- DDS implementation:
  - cycloneDDS

#### Steps to reproduce issue

``` ros2 param list /controller_server ```

```
yannic:~/projects//ros/yannic_robot_ws$ ros2 param list /controller_server 
  /bond_disable_heartbeat_timeout
  FollowPath.ConstraintCritic.cost_power
  FollowPath.ConstraintCritic.cost_weight
  FollowPath.ConstraintCritic.enabled
  FollowPath.CostCritic.collision_cost
  FollowPath.CostCritic.consider_footprint
  FollowPath.CostCritic.cost_power
  FollowPath.CostCritic.cost_weight
  FollowPath.CostCritic.critical_cost
  FollowPath.CostCritic.enabled
  FollowPath.CostCritic.inflation_layer_name
  FollowPath.CostCritic.near_goal_distance
  FollowPath.GoalAngleCritic.cost_power
  FollowPath.GoalAngleCritic.cost_weight
  FollowPath.GoalAngleCritic.enabled
  FollowPath.GoalAngleCritic.threshold_to_consider
  FollowPath.GoalCritic.cost_power
  FollowPath.GoalCritic.cost_weight
  FollowPath.GoalCritic.enabled
  FollowPath.GoalCritic.threshold_to_consider
  FollowPath.PathAlignCritic.cost_power
  FollowPath.PathAlignCritic.cost_weight
  FollowPath.PathAlignCritic.enabled
  FollowPath.PathAlignCritic.max_path_occupancy_ratio
  FollowPath.PathAlignCritic.offset_from_furthest
  FollowPath.PathAlignCritic.threshold_to_consider
  FollowPath.PathAlignCritic.trajectory_point_step
  FollowPath.PathAlignCritic.use_path_orientations
  FollowPath.PathAngleCritic.cost_power
  FollowPath.PathAngleCritic.cost_weight
  FollowPath.PathAngleCritic.enabled
  FollowPath.PathAngleCritic.forward_preference
  FollowPath.PathAngleCritic.max_angle_to_furthest
  FollowPath.PathAngleCritic.offset_from_furthest
  FollowPath.PathAngleCritic.threshold_to_consider
  FollowPath.PathFollowCritic.cost_power
  FollowPath.PathFollowCritic.cost_weight
  FollowPath.PathFollowCritic.enabled
  FollowPath.PathFollowCritic.offset_from_furthest
  FollowPath.PathFollowCritic.threshold_to_consider
  FollowPath.PreferForwardCritic.cost_power
  FollowPath.PreferForwardCritic.cost_weight
  FollowPath.PreferForwardCritic.enabled
  FollowPath.PreferForwardCritic.threshold_to_consider
  FollowPath.TrajectoryVisualizer.time_step
  FollowPath.TrajectoryVisualizer.trajectory_step
  FollowPath.TwirlingCritic.cost_power
  FollowPath.TwirlingCritic.cost_weight
  FollowPath.TwirlingCritic.enabled
  FollowPath.VelocityDeadbandCritic.cost_power
  FollowPath.VelocityDeadbandCritic.cost_weight
  FollowPath.VelocityDeadbandCritic.deadband_velocities
  FollowPath.VelocityDeadbandCritic.enabled
  FollowPath.batch_size
  FollowPath.critics
  FollowPath.enforce_path_inversion
  FollowPath.gamma
  FollowPath.iteration_count
  FollowPath.max_robot_pose_search_dist
  FollowPath.model_dt
  FollowPath.motion_model
  FollowPath.plugin
  FollowPath.prune_distance
  FollowPath.regenerate_noises
  FollowPath.reset_period
  FollowPath.retry_attempt_limit
  FollowPath.temperature
  FollowPath.time_steps
  FollowPath.transform_tolerance
  FollowPath.visualize
  FollowPath.vx_max
  FollowPath.vx_min
  FollowPath.vx_std
  FollowPath.vy_max
  FollowPath.vy_std
  FollowPath.wz_max
  FollowPath.wz_std
  controller_frequency
  controller_plugins
  controller_server.verbose
  failure_tolerance
  goal_checker_plugins
  min_theta_velocity_threshold
  min_x_velocity_threshold
  min_y_velocity_threshold
  odom_topic
  progress_checker.movement_time_allowance
  progress_checker.plugin
  progress_checker.required_movement_radius
  progress_checker_plugin
  simple_goal_checker.plugin
  simple_goal_checker.stateful
  simple_goal_checker.xy_goal_tolerance
  simple_goal_checker.yaw_goal_tolerance
  speed_limit_topic
  use_sim_time

```

#### Expected behavior
Parameters that are mentioned in the [documentation](https://docs.nav2.org/configuration/packages/configuring-mppic.html_ should actually exist (`ax_max`, `ax_min`, `ay_max`, `az_max`). 

Parameters like `ay_min` (that are not mentioned in the documentation and don't exist) should also be supported. 
#### Actual behavior

#### Reproduction instructions

#### Additional information

SteveMacenski, 2025-03-25:
Ah, yes these are not available in Humble. Humble's very out of date due to the need to maintain a stable ABI/API for Humble users, which prevents us from backporting many features to that distribution at this point. Unfortunately, then that cascades where other features depend on the features that were previously not backportable and so forth. Humble's reasonably out of date with SOTA in Nav2 due to the velocity of feature development.

I suggest upgrading to Jazzy if you want to use the latest features! 

> Parameters like ay_min (that are not mentioned in the documentation and don't exist) should also be supported.

Noted and this is actually in progress in #4954!

SteveMacenski, 2025-04-14:
Merged!

closed 2025-04-14 by SteveMacenski