ros-navigation/navigation2 #5025: parameter 'use_composition:=True' often causes nav2bringup.launch.py initialization to fail

GoesM, 2025-03-27:
**Required Info:**

- Operating System:
  - ubuntu 22.04
- Computer:
  -  NUC or Y9000P(i7,4060)
- ROS2 Version:
  -  humble
- Version or commit hash:
  -  the latest
- DDS implementation:
  -  defaulted
- Nav2 Package:
  -  nav2_bringup
- Have you asked this on Robotics Stack Exchange?
  -  no

#### Tuning / Configuration Goal

I expect to deploy Nav2 to my robot through source code compilation.
As LLVM, clang is a very popular compiler for C/C++ programs,  I use the following configuration to compile nav2

```sh
export CC=/usr/bin/clang
export CXX=/usr/bin/clang++
export MAKEFLAGS="-j4" 
colcon build \
   --cmake-clean-cache \
   --cmake-args \
            -DBUILD_TESTING=OFF \
            -DCMAKE_CXX_FLAGS="${{CMAKE_CXX_FLAGS}} -w -Wno-error -Wno-everything"  \
            -DCMAKE_C_FLAGS="${{CMAKE_C_FLAGS}} -w -Wno-error -Wno-format-security"
```
the following is the version of used clang:
```sh
$ clang --version
Ubuntu clang version 14.0.0-1ubuntu1.1
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
```

#### What I've Tried

1. if I launch nav2 (built from source by clang) by following steps, it failed to initialize:
```sh
export TURTLEBOT3_MODEL=waffle_pi
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/opt/ros/humble/share/turtlebot3_gazebo/models
ros2 launch nav2_bringup tb3_simulation_launch.py
# HIGHLIGHT, here the parameter "use_composition" is set as "True" as defaulted
```

Initialization failure is manifested as:
- Nav2 error:
```
[component_container_isolated-5] [ERROR] [1742750307.678639050] [global_costmap.global_costmap]: Caught exception in callback for transition 10
[component_container_isolated-5] [ERROR] [1742750307.678696138] [global_costmap.global_costmap]: Original error: MultiLibraryClassLoader: Could not create object of class type nav2_costmap_2d::StaticLayer as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
```
- At the same time, all nav2_goals sent will be ejected, and nav2 will not work correctly

2. if I launch nav2  (built from source by clang) by following steps, its initialization also failed:
launch gazebo, nav2_bringup,separately:
```sh
# gazebo
export TURTLEBOT3_MODEL=waffle_pi
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/opt/ros/humble/share/turtlebot3_gazebo/models
ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py
```
```sh
# nav2_bringup
ros2 launch nav2_bringup bringup_launch.py map:=/path/to/map
```

3. if I launch nav2  (built from source by clang) by following steps, it could success its initialization:
launch gazebo, nav2, rviz separately:
```sh
# gazebo
export TURTLEBOT3_MODEL=waffle_pi
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/opt/ros/humble/share/turtlebot3_gazebo/models
ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py
```
```sh
# rviz
ros2 launch nav2_bringup rviz_launch.py
```
```sh
# nav2
ros2 launch nav2_bringup navigation_launch.py
```

4. if I launch nav2 (built from source by clang) by following steps, it could success its initialization:
if we set "use_composition:=False", nav2 could work correctly:
```sh
ros2 launch nav2_bringup bringup_launch.py use_composition:=False
```
OR
```sh
ros2 launch nav2_bringup tb3_simulation_launch.py use_composition:=False
```

5. if I launch nav2 (built from apt-get installation), it could always success

#### What I've guessed
1. This phenomenon should be mainly caused by the  launcher "bringup_launch.py" 
2. "use_composition:=True" conflict with using "clang"/"LLVM" as compiler.

SteveMacenski, 2025-03-27:
Interesting. Not an issue [solved with a reboot](https://github.com/ros-navigation/navigation2/issues/5008#issuecomment-2746913441)? 

I assume nothing's special about the static layer, if you try with other layers as the first layer, do you still see this error? (Obstacle, Voxel, Inflation, etc)

What about if you compile pluginlib locally with the same compiler and settings? Maybe theres some compatibility issue 

This might be something with plugins inside of plugins - do you see an issue with the controller server if you comment out the costmap stuff and try again? If you use DWB or MPPI, there are also plugins inside of other plugins - which itself is inside of the component which is also a plugin. 

---

FYI the reason that `ros2 launch nav2_bringup navigation_launch.py` works is because that has `use_composition` as false, so all cases that it doesn't work are w.r.t. composition and when it does work it is without composition.

GoesM, 2025-03-28:
> Interesting. Not an issue [solved with a reboot](https://github.com/ros-navigation/navigation2/issues/5008#issuecomment-2746913441)?

uhuhuhuh,  last time I wasn't very clear about what went wrong, but I’ve now gathered more information.

> I assume nothing's special about the static layer, if you try with other layers as the first layer, do you still see this error? (Obstacle, Voxel, Inflation, etc)

Sure, I’ve tried using all the other layers as the first layer, and the result remains the same.

> What about if you compile pluginlib locally with the same compiler and settings? Maybe theres some compatibility issue

Which pluginlib are you referring to? Do you mean plugins for the costmap, controller, or planner, like costmap layers, DWB, MPPI, and so on?

Actually, I’ve been using clang/clang++ for compilation for the entire Navigation2 stack.

# Additional information
To clearly demonstrate what happened when this issue occurred, I’ve attached the Nav2 execution logs below.

Everytime I used "use_composition:=True", Nav2 failed to work. Here’s the complete log:

```sh 
# launch by cmd: `ros2 launch nav2_bringup tb3_simulation_launch.py headless:=False use_rviz:=True`
[INFO] [launch]: All log files can be found below /home/ubuntu2204-humble/.ros/log/2025-03-28-11-43-00-604094-GoesM-1042
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [1058]
[INFO] [gzclient-2]: process started with pid [1060]
[INFO] [spawn_entity.py-3]: process started with pid [1062]
[INFO] [robot_state_publisher-4]: process started with pid [1064]
[INFO] [rviz2-5]: process started with pid [1066]
[INFO] [component_container_isolated-6]: process started with pid [1068]
[robot_state_publisher-4] [INFO] [1743133381.081121127] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-4] [INFO] [1743133381.081255573] [robot_state_publisher]: got segment base_link
[robot_state_publisher-4] [INFO] [1743133381.081262283] [robot_state_publisher]: got segment base_scan
[robot_state_publisher-4] [INFO] [1743133381.081264617] [robot_state_publisher]: got segment camera_depth_frame
[robot_state_publisher-4] [INFO] [1743133381.081266643] [robot_state_publisher]: got segment camera_depth_optical_frame
[robot_state_publisher-4] [INFO] [1743133381.081268706] [robot_state_publisher]: got segment camera_link
[robot_state_publisher-4] [INFO] [1743133381.081270826] [robot_state_publisher]: got segment camera_rgb_frame
[robot_state_publisher-4] [INFO] [1743133381.081272756] [robot_state_publisher]: got segment camera_rgb_optical_frame
[robot_state_publisher-4] [INFO] [1743133381.081274664] [robot_state_publisher]: got segment caster_back_left_link
[robot_state_publisher-4] [INFO] [1743133381.081276503] [robot_state_publisher]: got segment caster_back_right_link
[robot_state_publisher-4] [INFO] [1743133381.081278357] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-4] [INFO] [1743133381.081280476] [robot_state_publisher]: got segment wheel_left_link
[robot_state_publisher-4] [INFO] [1743133381.081282508] [robot_state_publisher]: got segment wheel_right_link
[component_container_isolated-6] [INFO] [1743133381.131110203] [nav2_container]: Load Library: /**tb3_ws/install/nav2_map_server/lib/libmap_server_core.so
[rviz2-5] QStandardPaths: wrong permissions on runtime directory /run/user/1000/, 0755 instead of 0700
[component_container_isolated-6] [INFO] [1743133381.288536523] [nav2_container]: Found class: rclcpp_components::NodeFactoryTemplate<nav2_map_server::CostmapFilterInfoServer>
[component_container_isolated-6] [INFO] [1743133381.288584494] [nav2_container]: Found class: rclcpp_components::NodeFactoryTemplate<nav2_map_server::MapSaver>
[component_container_isolated-6] [INFO] [1743133381.288589494] [nav2_container]: Found class: rclcpp_components::NodeFactoryTemplate<nav2_map_server::MapServer>
[component_container_isolated-6] [INFO] [1743133381.288592004] [nav2_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<nav2_map_server::MapServer>
[component_container_isolated-6] [INFO] [1743133381.294264046] [map_server]:
[component_container_isolated-6]        map_server lifecycle node launched.
[component_container_isolated-6]        Waiting on external lifecycle transitions to activate
[component_container_isolated-6]        See https://design.ros2.org/articles/node_lifecycle.html for more information.
[component_container_isolated-6] [INFO] [1743133381.294616154] [map_server]: Creating
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/map_server' in container '/nav2_container'
[component_container_isolated-6] [INFO] [1743133381.295199945] [nav2_container]: Load Library: /**tb3_ws/install/nav2_controller/lib/libcontroller_server_core.so
[component_container_isolated-6] [INFO] [1743133381.345580726] [nav2_container]: Found class: rclcpp_components::NodeFactoryTemplate<nav2_controller::ControllerServer>
[component_container_isolated-6] [INFO] [1743133381.345625262] [nav2_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<nav2_controller::ControllerServer>
[component_container_isolated-6] [INFO] [1743133381.351948953] [controller_server]:
[component_container_isolated-6]        controller_server lifecycle node launched.
[component_container_isolated-6]        Waiting on external lifecycle transitions to activate
[... cut ...]
[component_container_isolated-6] [INFO] [1743133386.632117469] [amcl]: initialPoseReceived
[component_container_isolated-6] [INFO] [1743133386.632211683] [amcl]: Setting pose (0.000000): -1.919 -0.516 -0.171
[rviz2-5] Start navigation
[rviz2-5] [INFO] [1743133391.172369991] [rviz_navigation_dialog_action_client]: NavigateToPose will be called using the BT Navigator's default behavior tree.
[rviz2-5] [ERROR] [1743133391.174587814] [rviz_navigation_dialog_action_client]: Goal was rejected by server
```

SteveMacenski, 2025-03-28:
Ah sorry, the dynamic libraries package is called pluginlib (https://github.com/ros/pluginlib) and I'm asking for you to build that from source with your compiler settings to see if that helps. 

Other things that would be good to try:
- Comment out the costmap creation from the controller server, do the controller plugins also fail?
- Move the configuration in the lifecycle manager's input nodes of the behavior server before the controller server, does that also fail?

That would tell us if there's something unique about the controller server and/or costmap plugins.

Next, if those both are still failing:
- Compile pluginlib and rclcpp_components, see if that fixes it
- If not, compile ROS 2 from source using ros2.repos file (will take about 45 minutes, sorry!) taking care to leave your installed version out of your environmental variables / bashrc -- see if compiling fully using those flags resolves it

I expect the issue is just a cross compilation issue with something in ROS 2 core in gcc with plugins done in clang (not necessarily anything to do with Nav2)

muellerbernd, 2025-04-06:
We have the same "problem" in the [nix overlay for ROS](https://github.com/lopsided98/nix-ros-overlay). See [this PR](https://github.com/lopsided98/nix-ros-overlay/pull/591).

SteveMacenski, 2025-04-14:
@GoesM any update?

SteveMacenski, 2025-04-30:
@GoesM any update? Will close in a few weeks if not

GoesM, 2025-05-01:
Apologies for the delay. We’ve struggled with reinstalling the entire ROS environment with clang/clang++ since we’re not very familiar with it. After numerous failed attempts, we still haven’t resolved the issue.

> Any update? Will close in a few weeks if not.

It seems that, given our current capabilities, it might be too difficult for us to proceed with further validation, so...

SteveMacenski, 2025-05-02:
OK - I'm going to close this then. I'm happy to help / brainstorm / etc but if you're probably going to abandon this effort then I think its worth taking it off the plate.

Thanks for your diligence and tickets, as always! 😄

closed 2025-05-02 by SteveMacenski