ros-navigation/navigation2 #5032: Navigation crashes randomly after canceling and setting a new goal

MFHuniovi, 2025-03-31:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22-04
- Computer:
  -  Hyper V (simulate with gazebo)
  -  i7-10700 (real robot)
- ROS2 Version:
  - ROS2 humble https://docs.ros.org/en/humble/Installation.html
  - Nav2 https://docs.nav2.org/development_guides/build_docs/index.html#for-iron-and-older
- Version or commit hash:
amr-sim@amrsim:$ dpkg-query --show "ros-$ROS_DISTRO-navigation2"
ros-humble-navigation2  1.1.16-1jammy.20240830.231501
amr-sim@amrsim:$ dpkg-query --show "ros-$ROS_DISTRO-nav2-*"
ros-humble-nav2-amcl    1.1.16-1jammy.20240830.193818
ros-humble-nav2-behavior-tree   1.1.16-1jammy.20240830.193910
ros-humble-nav2-behaviors       1.1.16-1jammy.20240830.205037
ros-humble-nav2-bringup 1.1.16-1jammy.20240831.001909
ros-humble-nav2-bt-navigator    1.1.16-1jammy.20240830.205457
ros-humble-nav2-collision-monitor       1.1.16-1jammy.20240830.204654
ros-humble-nav2-common  1.1.16-1jammy.20240824.181243
ros-humble-nav2-constrained-smoother    1.1.16-1jammy.20240830.205908
ros-humble-nav2-controller      1.1.16-1jammy.20240830.205928
ros-humble-nav2-core    1.1.16-1jammy.20240830.204819
ros-humble-nav2-costmap-2d      1.1.16-1jammy.20240830.202726
ros-humble-nav2-dwb-controller  1.1.16-1jammy.20240830.214325
ros-humble-nav2-lifecycle-manager       1.1.16-1jammy.20240830.202342
ros-humble-nav2-map-server      1.1.16-1jammy.20240830.193909
ros-humble-nav2-mppi-controller 1.1.16-1jammy.20240830.205055
ros-humble-nav2-msgs    1.1.16-1jammy.20240824.181735
ros-humble-nav2-navfn-planner   1.1.16-1jammy.20240830.205449
ros-humble-nav2-planner 1.1.16-1jammy.20240830.205543
ros-humble-nav2-regulated-pure-pursuit-controller       1.1.16-1jammy.20240830.205157
ros-humble-nav2-rotation-shim-controller        1.1.16-1jammy.20240830.210830
ros-humble-nav2-rviz-plugins    1.1.16-1jammy.20240830.230452
ros-humble-nav2-simple-commander        1.1.16-1jammy.20240824.182428
ros-humble-nav2-smac-planner    1.1.16-1jammy.20240830.205547
ros-humble-nav2-smoother        1.1.16-1jammy.20240830.214913
ros-humble-nav2-theta-star-planner      1.1.16-1jammy.20240830.205759
ros-humble-nav2-util    1.1.16-1jammy.20240830.192949
ros-humble-nav2-velocity-smoother       1.1.16-1jammy.20240830.204314
ros-humble-nav2-voxel-grid      1.1.16-1jammy.20240824.181738
ros-humble-nav2-waypoint-follower       1.1.16-1jammy.20240830.205813
- DDS implementation:
  - Default RMW

#### Steps to reproduce issue
Launch navigation using:

```ros2 launch nav2_bringup bringup_launch.py ```
with a predefined map and nav2_params.

Start a Python node that sends a navigate_to_pose action to move the robot to point A.
```python3 routine_test.py```
```
import rclpy
from rclpy.action import ActionClient
from rclpy.node import Node
from nav2_msgs.action import NavigateToPose
from geometry_msgs.msg import PoseStamped
import time

class NavigateToPoseClient(Node):

    def __init__(self, pose_a, pose_b):
        super().__init__('navigate_to_pose_client')
        self._action_client = ActionClient(self, NavigateToPose, 'navigate_to_pose')

         # Definir los dos puntos de destino
        self.pose_a = pose_a
        self.pose_b = pose_b
        self.current_goal = self.pose_a  # Para rastrear a qué punto vamos

        self.visits_a = 0  # Contador de visitas al punto A
        self.visits_b = 0  # Contador de visitas al punto B
        self.total_visits = 0  # Total de viajes realizados
        
        # Archivo de logs
        t = time.time()
        logTime = time.strftime('%Y-%m-%d_%H:%M', time.gmtime(t))
        
        # Esperar a que el servidor esté disponible
        self._action_client.wait_for_server()

        # Iniciar la navegación
        self.send_goal(self.pose_a)

    def send_goal(self, pose: PoseStamped):
        goal_msg = NavigateToPose.Goal()
        goal_msg.pose = pose
        self.current_goal = pose

        if self.current_goal == self.pose_a:
            self.visits_a += 1
        else:
            self.visits_b += 1
        self.total_visits += 1

        t = time.time()
        logTime = time.strftime('%Y-%m-%d_%H:%M', time.gmtime(t))
        logMessage = f'[{logTime}]Visitas totales a objetivos[{self.total_visits}] [Punto[{self.pose_a.pose.position.x},{self.pose_a.pose.position.y}]: {self.visits_a} veces, Punto[{self.pose_b.pose.position.x},{self.pose_b.pose.position.y}]: {self.visits_b} veces]'
        
        self.get_logger().info(logMessage)
        
        self.start_time = t

        self.get_logger().info(f'Enviando objetivo: x={pose.pose.position.x}, y={pose.pose.position.y}')

        self._send_goal_future = self._action_client.send_goal_async(goal_msg, feedback_callback=self.feedback_callback)
        self._send_goal_future.add_done_callback(self.goal_response_callback)

    def goal_response_callback(self, future):
        self._goal_handle = future.result()
        if not self._goal_handle.accepted:
            self.get_logger().info('Goal rejected :(')
            return

        self.get_logger().info('Goal accepted :)')

        self._get_result_future = self._goal_handle.get_result_async()
        self._get_result_future.add_done_callback(self.get_result_callback)

    def feedback_callback(self, feedback_msg):
        feedback = feedback_msg.feedback
        
        
        self.get_logger().info(
            f'Posición actual: x={feedback.current_pose.pose.position.x}, y={feedback.current_pose.pose.position.y}, '
            f'Tiempo navegación: {feedback.navigation_time.sec}s, Tiempo restante: {feedback.estimated_time_remaining.sec}s, '
            f'Recuperaciones: {feedback.number_of_recoveries}, Distancia restante: {feedback.distance_remaining:.2f}m'
        )
        
        if time.time() - self.start_time < 3:
            return
            
        if 0.07 < feedback.distance_remaining <= 0.18:
            self.get_logger().info("Se llega al umbral de <0.18 se procede a cancelar")
            cancel_future = self._goal_handle.cancel_goal_async()
            
            cancel_future.add_done_callback(self.cancel_done_callback)  
                    
    def cancel_done_callback(self, future):
        cancel_response = future.result()
        self.get_logger().info("Esperando...")
        self._action_client.wait_for_server()
        self.get_logger().info("Espera terminada!!!")

    def get_result_callback(self, future):
        self.get_logger().info('Goal completed!')
        next_goal = self.pose_b if self.current_goal == self.pose_a else self.pose_a
        self.send_goal(next_goal)
        

def main(args=None):
    rclpy.init(args=args)

    print("Iniciado RoutineTest")
    # Definir dos objetivos PoseStamped
    goal_pose_a = PoseStamped()
    goal_pose_a.header.frame_id = 'map'
    goal_pose_a.header.stamp = rclpy.time.Time().to_msg()
    goal_pose_a.pose.position.x = -1.0
    goal_pose_a.pose.position.y = 1.0
    goal_pose_a.pose.position.z = 0.0
    goal_pose_a.pose.orientation.w = 1.0

    goal_pose_b = PoseStamped()
    goal_pose_b.header.frame_id = 'map'
    goal_pose_b.header.stamp = rclpy.time.Time().to_msg()
    goal_pose_b.pose.position.x = -2.0
    goal_pose_b.pose.position.y = 2.0
    goal_pose_b.pose.position.z = 0.0
    goal_pose_b.pose.orientation.w = 1.0

    node = NavigateToPoseClient(goal_pose_a, goal_pose_b)

    rclpy.spin(node)
    rclpy.shutdown()

if __name__ == '__main__':
    main()

```

When the robot is within <0.18m of point A, cancel the goal and immediately send a new goal to point B.

At a random moment (either shortly after starting or after several hours of execution), navigation crashes.
#### Expected behavior
The robot should reliably navigate to each goal without unexpected crashes.

#### Actual behavior
At an unpredictable moment, navigation crashes. This can happen soon after launching or after hours of execution in both simulation and real-world scenarios.  Additionally, the error message is not always the same, making it difficult to identify a clear pattern.

[image]

[image]

[image]

[image]

[image]

#### Reproduction instructions
The issue is hard to reproduce consistently, but it involves canceling a goal and sending a new one in quick succession. Attached are screenshots and a core dump trace showing the crash.

#### Additional information
I followed the instructions from this post (https://askubuntu.com/questions/434431/how-can-i-read-a-crash-file-from-var-crash) to obtain the trace of the crashes I experienced. I am attaching traces from different crashes (although I did not obtain crash files for all the errors shown in the screenshots, the execution process was always the same).

[image]

[image]

[image]

[image]

I need help understanding what is happening—whether it's a bug, a misuse of the navigation stack, or if canceling and sending a new goal so quickly is not recommended. Any guidance on reproducing the issue, debugging it, or identifying relevant files to check would be greatly appreciated. 

Thanks in advance!

SteveMacenski, 2025-04-02:
[This seems to point to your own Bt node crashing](https://private-user-images.githubusercontent.com/71136753/428545438-aa6f788e-fa6e-457a-8236-e4449468acea.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDM1NTI1NzYsIm5iZiI6MTc0MzU1MjI3NiwicGF0aCI6Ii83MTEzNjc1My80Mjg1NDU0MzgtYWE2Zjc4OGUtZmE2ZS00NTdhLTgyMzYtZTQ0NDk0NjhhY2VhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA0MDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNDAyVDAwMDQzNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ4NDI4NTgzYmI4MzM1MWMyODk2Yjc4NGM4MWJkOGI0NTVjZjY0YzgxOTM2ZDIyMWViN2Q1YjE2NWQ0MWViOTYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.Zc9vcXR_XySgyjJZSFgVA9hYSMLzSi9jsDeo2byS2lI) `portenta_control.cpp`

 But, we'd know more if you compiled with `-g` so we can see the full trace on crash to see what the offending lines are. We have a [guide here](https://docs.nav2.org/tutorials/docs/get_backtrace.html) that describes it (make sure to do Nav2 + your code).

MFHuniovi, 2025-04-03:
I have left all night the computer simulating without any custom plugin and it did not die, so I understand that the problem is in one of my plugins. Do you have any guide or recommendations for me to develop the plugins correctly and avoid the navigation to die?

Thanks

SteveMacenski, 2025-04-04:
This is likely part of your issue. That node is a shared resource for all BT nodes in your tree. Some of which may be `spin_until_future_complete`. You're not allowed to spin a node in multiple executors simultaneously so if this thread's `spin_some` overlaps with the `spin_until_future_complete` of another node, then that would cause an issue. It throws an exception, and if that exception was caught in a BT node, then that would kill the navigation task in error. If it gets throws in this thread, then there's no `catch` for the thread and it probably brings down the process. 

```

    spinning_thread_ = std::make_unique<std::thread>([this]() {
        rclcpp::Rate rate(10); // 10 Hz
        while (rclcpp::ok()) {
            rclcpp::spin_some(node_);
            rate.sleep();
        }
    });
```

MFHuniovi, 2025-04-09:
Hi, sorry for replying a bit late.
I've been going through the documentation to use callback_group and executor correctly. I've also been checking BT plugins examples and behaviorTree.cpp comments, and it seems that the problem disappeared in some plugins, because right now I have 3 plugins of my own where 2 of them have been running up to 5h in a row without error when normally it stops between 20 minutes and 1h. But the problem keeps occurring randomly with the other plugin and the navigation disappears and this time it didn't show anything in the console or generate a file in /var/crash/<file>.crash. This plugin only calls a service from a "central"  node which in turn makes a call to change a local_costmap parameter.

```
void MAIN_NODE_CLASS::InitializeFootprintSetter() {
    parameter_footprint_client_ = std::make_shared<rclcpp::AsyncParametersClient>(this, "local_costmap/local_costmap");
    if (!parameter_footprint_client_->wait_for_service(std::chrono::seconds(5))) {
        RCLCPP_ERROR(this->get_logger(), "Service not available. Make sure the navigation node is running.");
    }
}

void MAIN_NODE_CLASS::SetFootprintCallback(const std::shared_ptr<mss_types::srv::SetString::Request> request,
                                          std::shared_ptr<mss_types::srv::SetString::Response> response)
{
    std::string footprint = request->data;

    RCLCPP_INFO(this->get_logger(), "Received request to set footprint: %s", footprint.c_str());

    if (!parameter_footprint_client_->wait_for_service(std::chrono::seconds(5))) {
        response->result = false;
        response->message = "Parameter service is not available.";
        RCLCPP_ERROR(this->get_logger(), "%s", response->message.c_str());
        return;
    }

    try {
        parameter_footprint_client_->set_parameters({
            rclcpp::Parameter("footprint", footprint)
        });

        response->result = true;
        response->message = "Footprint set successfully.";
        RCLCPP_INFO(this->get_logger(), "%s", response->message.c_str());
    } catch (const std::exception &e) {
        response->result = false;
        response->message = "Failed to set footprint: " + std::string(e.what());
        RCLCPP_ERROR(this->get_logger(), "%s", response->message.c_str());
    }
}
```

 So I think I have a problem in the following action plugin:

```
#include "action/set_footprint_action.hpp"

namespace nav2_behavior_tree
{

SetFootprintAction::SetFootprintAction(const std::string &name, const BT::NodeConfiguration &config)
    : BT::AsyncActionNode(name, config), current_footprint_(""), spinning_thread_(nullptr)
{
    node_ = this->config().blackboard->get<rclcpp::Node::SharedPtr>("node");
    RCLCPP_INFO(node_->get_logger(), "SetFootprintAction::SetFootprintAction => Instantiating action");

    callback_group_ = node_->create_callback_group(rclcpp::CallbackGroupType::MutuallyExclusive, false);
    callback_group_executor_.add_callback_group(callback_group_, node_->get_node_base_interface());
    
    // Create the service client
    setClient_ = node_->create_client<mss_types::srv::SetString>(
        "/setFootprint",
        rmw_qos_profile_services_default,
        callback_group_);
    
    // Create QoS compatible with the publisher
    rclcpp::QoS qos_profile(rclcpp::QoSInitialization::from_rmw(rmw_qos_profile_default));
    qos_profile.reliability(RMW_QOS_POLICY_RELIABILITY_RELIABLE);
    qos_profile.durability(RMW_QOS_POLICY_DURABILITY_VOLATILE);
    qos_profile.history(RMW_QOS_POLICY_HISTORY_KEEP_LAST).keep_last(10);
    
[... 146 lines cut ...]

```
#ifdef USE_BTCPP3_OLD_NAMES
using AsyncActionNode = ThreadedAction;
#endif
/**
 * @brief The ThreadedAction executes the tick in a different thread.
 *
 * IMPORTANT: this action is quite hard to implement correctly.
 * Please make sure that you know what you are doing.
 *
 * - In your overriden tick() method, you must check periodically
 *   the result of the method isHaltRequested() and stop your execution accordingly.
 *
 * - in the overriden halt() method, you can do some cleanup, but do not forget to
 *   invoke the base class method ThreadedAction::halt();
 *
 * - remember, with few exceptions, a halted ThreadedAction must return NodeStatus::IDLE.
 *
 * For a complete example, look at __AsyncActionTest__ in action_test_node.h in the folder test.
 *
 * NOTE: when the thread is completed, i.e. the tick() returns its status,
 * a TreeNode::emitWakeUpSignal() will be called.
 */
```

SteveMacenski, 2025-04-09:
I think this is better suited on Robotics Stack Exchange now that we know its an issue with your custom BT nodes. Please follow up there and I'll try to give your code a look and see if I can identify a new issue. 

From a quick glance, It sounds like you may have deadlocked, something important to remember is that the parameter setting object calls a service to a remote node, so it could deadlock if you call a service with the same executor within an callback (which itself is blocking the `spin` thread being executed). I don't know if that's the issue, but that's my first immediate thought.