ros-navigation/navigation2 #5033: Prevent RoundRobin wrap-around in recovery sequences

DylanDeCoeyer-Quimesis, 2025-03-31:
**Issue:**
The RoundRobin node currently wraps around to its first child after the last one succeeds. When used as recovery within a RecoveryNode (e.g., [here](https://github.com/ros-navigation/navigation2/blob/main/nav2_bt_navigator/behavior_trees/nav_to_pose_with_consistent_replanning_and_if_path_becomes_invalid.xml#L36)), its index can become misaligned with the RecoveryNode’s retry counter. This happens because RoundRobin skips children that return FAILURE, while the RecoveryNode assumes each attempt corresponds to a distinct child. If the RecoveryNode’s number_of_retries equals the number of RoundRobin children, there is a risk that RoundRobin wraps around and executes some children more than once.

**Proposed Enhancement:**
Introduce a boolean parameter that, when enabled, causes RoundRobin to return FAILURE instead of wrapping around. This would signal to the RecoveryNode that all recovery actions have been exhausted, allowing the failure to propagate appropriately.

SteveMacenski, 2025-04-02:
> its index can become misaligned with the RecoveryNode’s retry counter.

I think that is the design intent of the RoundRobin node. Its supposed to return success when the first one returns success ([docs](https://docs.nav2.org/behavior_trees/overview/nav2_specific_nodes.html#control-roundrobin)). The RecoveryNode's counter is counting the number of times a successful recovery was completed, whereas the RoundRobin's current state is just one which part of its children actions it is currently trying to attempt (which can include failures).

If you wanted the number of attempts to equal the number of retires for the RecoveryNode, wouldn't it be better to drop the RoundRobbin entirely and just use a `ForceSuccessNode` so that even failures are reported back to the RecoveryNode as a success? The force success is to make sure we increment the recovery counter on failures and try again instead of exiting ([see code](https://github.com/ros-navigation/navigation2/blob/main/nav2_behavior_tree/plugins/control/recovery_node.cpp#L94-L106))

DylanDeCoeyer-Quimesis, 2025-04-03:
> If you wanted the number of attempts to equal the number of retires for the RecoveryNode, wouldn't it be better to drop the RoundRobbin entirely and just use a ForceSuccessNode so that even failures [...]

I'm not sure to understand, how could a ForceSuccess replace the RoundRobin? I have multiple recovery actions, and I need to execute a different one at every attempt. I see no other way than using the RoundRobin node whose purpose is exactly this one.

> If all children fail, we should be already returning FAILURE.

Yes, that's correct. But here, I would like to have the RoundRobin **also** return FAILURE when it is ticked and its current index equals the number of children. It means "I have no more child to tick, I cannot do anything". The index could be reset in the `halt()` function.

This is a behavior that would be helpful for our use-case. But again, I suggest to have a boolean parameter to enable or disable this behavior, and it could be disabled by default.

> Its worth noting I have a PR open right now https://github.com/ros-navigation/navigation2/pull/5027

I think this PR is good but it does not help with my issue. I just want to ensure that each recovery action is performed once and only once, even if one of them fails. Currently, for every recovery action failure (skipped by the RoundRobin), I have a recovery action which is executed twice due to the wrapping.

SteveMacenski, 2025-04-08:
Sorry I had some typos in my thoughts on `ForceSuccess` - however after further thought I agree with you so I'm going to just leave it as 'oops'. I was thinking another BT node that returns `SUCCESS` and `FAILURE` in the tick children in a row fashion could have its return overrided to always be `SUCCESS`, but I'm not sure other than `RoundRobin` that exists in the BT.CPP ecosystem.

A param for enabling wrap around would be a good! Please submit a PR. I think that even makes sense to set in all the Nav2 BT XMLs as well to be `false` so that after all are attempted, we exit rather than wrapping around. Should we update any of the retry counts in the recovery node to match?

SteveMacenski, 2025-05-03:
@DylanDeCoeyer-Quimesis are you planning on submitting a patch?

SteveMacenski, 2025-06-26:
@DylanDeCoeyer-Quimesis just another gentle ping :-) 

Just out of curiosity - to see if it does this well and maybe all you need to do is just test it instead:

@claude implement this feature. Provide a parameter that is default false about whether the RoundRobin behavior tree node in the `nav2_behavior_tree` package should wrap around its children or not. The current behavior wraps around and we no longer want this to be the default behavior.

commit a59e4e16 (2025-06-26): Add wrap_around parameter to RoundRobin node to control wrap-around behavior

- Add boolean wrap_around parameter (default: false) to RoundRobin node
- When wrap_around=false, node returns FAILURE instead of wrapping to first child
- Updated XML node definition to include new parameter
- Existing behavior tree XMLs will use new non-wrap-around behavior by default
- Addresses issue where RoundRobin index can become misaligned with RecoveryNode retry counter

Fixes #5033