ros-navigation/navigation2 #5041: Callback not executed in IsBatteryCharging BT

ajtudela, 2025-04-02:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 24.04
- ROS2 Version: Jazzy / Rolling
  - Jazzy / rolling
- Version or commit hash:
  - Source
- DDS implementation:
  - Cyclone

##
I have an problem since we moved to Jazzy, but its also happening in Rolling. Both installed from source.

I have a behavior tree where the battery level is checked before trying to undock the robot, similar to this [one](https://github.com/ros-navigation/navigation2/blob/main/nav2_docking/opennav_docking_bt/behavior_trees/application_example.xml): 

But I have to call the NavigateToPose action twice before the robot starts to move. 
- On the first call, IsBatteryCharging returns FALSE, so the other branch of the BT is executed. As the robot is docked, the global result is a failure.
- On the second call, IsBatteryCharging returns SUCCESS, performing the undocking action and moving as usual. 

After checking the code [here](https://github.com/ros-navigation/navigation2/blob/main/nav2_behavior_tree/plugins/condition/is_battery_charging_condition.cpp), I noticed that the **batteryCallback** is never executed in the first call, but it is in the second one. I think it's because this **spin_some** doesn't spin the first time.

https://github.com/ros-navigation/navigation2/blob/08268f9922f92511f12ad9b951148d1c1d30096b/nav2_behavior_tree/plugins/condition/is_battery_charging_condition.cpp#L66-L72

After reading other code, I found a similar problem fixed with a double call:
https://github.com/ros-navigation/navigation2/blob/08268f9922f92511f12ad9b951148d1c1d30096b/nav2_behavior_tree/plugins/decorator/goal_updater_node.cpp#L95-L97

I'm not sure if this is fixed or not because putting another **spin_some** in the **IsBatteryCharging** BT fixed the problem.

SteveMacenski, 2025-04-02:
Linking: https://github.com/ros-navigation/navigation2/issues/4567 we should NOT revert this yet

ajtudela, 2025-04-02:
> If you still see this, that's a problem. Is your Jazzy/Rolling up to date? 

Yes, I updated this morning.

> If so, can you follow up with a new rclcpp ticket on this and link those PRs that this is still an issue?

Sure!

> I think that should just be applied here as well. Call spin_some twice, the first time with a duration of 1ns.
> - is_battery_charging_condition
> - is_battery_low_condition
> - XYZ_selector_node's

Agreed. I'll open a PR tomorrow to provide a temporary fix while rclcpp is reviewed.

closed 2025-04-07 by SteveMacenski