ros-navigation/navigation2 #5042: [nav2_bringup] localization includes the map server

dvdmc, 2025-04-02:
Currently, when using the bringup launch with `use_localization = False`, the map server is not loaded. Thus, the global planner returns:

```bash
[global_costmap.global_costmap]: Can't update static costmap layer, no map received
```

Actually, I don't even need a map. What is the intended way of configuring the global costmap to not require a map for planning?

SteveMacenski, 2025-04-02:
You'd need to remove the static layer from the costmap configurations in that case. Usually when localization is off, SLAM is on, so that is providing a static map. If not using localization nor SLAM, then you need to remove the use of a occupancy grid :-)

closed 2025-04-02 by SteveMacenski

dvdmc, 2025-04-03:
Thanks! Can this be done only with the params file? Where exactly?

SteveMacenski, 2025-04-03:
In your `global_costmap` and `local_costmap` params in your yaml file, remove the `static_layer` from the plugins list.