ros-navigation/navigation2 #5054: Robot failing to rotate to goal

Flamethr0wer, 2025-04-04:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Computer:
  - Raspberry Pi 4B
- ROS2 Version:
  - Humble binaries
- Version or commit hash:
  - apt binaries: ros-humble-navigation2 1.1.17-1jammy.20241128.124843
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
Jutt set a goal_pose and let the robot navigate to it. My current params file is this:
```
amcl:
  ros__parameters:
    use_sim_time: False
    alpha1: 0.5
    alpha2: 0.2
    alpha3: 0.3
    alpha4: 0.5
    alpha5: 0.2
    base_frame_id: "base_footprint"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.8
    beam_skip_threshold: 0.3
    do_beamskip: true
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: -1.0
    laser_min_range: -1.0
    laser_model_type: "likelihood_field"
    max_beams: 60
    max_particles: 2000
    min_particles: 500
    odom_frame_id: "odom"
    pf_err: 0.05
    pf_z: 0.99
    recovery_alpha_fast: 0.2
    recovery_alpha_slow: 0.002
    resample_interval: 1
    robot_model_type: "nav2_amcl::DifferentialMotionModel"
    save_pose_rate: 0.5
    sigma_hit: 0.2
    tf_broadcast: true
    transform_tolerance: 1.0
    update_min_a: 0.05
    update_min_d: 0.05
    z_hit: 0.7
    z_max: 0.05
    z_rand: 0.4
    z_short: 0.05
    set_initial_pose: true
    initial_pose:
      x: 0.616
      y: -2.717
      z: 0.0    # Typically 0.0 for a 2D plane
      yaw: 0.0  # Replace with your desired initial orientation in radians
    scan_topic: scan

bt_navigator:
  ros__parameters:
    use_sim_time: False
    global_frame: map
    robot_base_frame: base_link
[... 325 lines cut ...]
    smoothing_frequency: 20.0
    scale_velocities: False
    feedback: "OPEN_LOOP"
    max_velocity: [0.3, 0.0, 1.0]
    min_velocity: [-0.3, 0.0, -1.0]
    max_accel: [2.5, 0.0, 3.2]
    max_decel: [-2.5, 0.0, -3.2]
    odom_topic: "odom"
    odom_duration: 0.1
    deadband_velocity: [0.0, 0.0, 0.0]
    velocity_timeout: 1.0
```

#### Expected behavior
When giving it a goal (6.71, 0.455 with orientation z = -0.707 and w = 0.707 in my case), the robot should navigate to it and then align with the requested orientation.

#### Actual behavior
Instead, the robot stops near the goal and starts printing these messages many times in the terminal:
```
[ERROR] [DWBLocalPlanner]: 1.00: RotateToGoal/Nonrotation command near goal. 
[WARN] [controller_server]: No valid trajectories out of 420!
```

#### Additional information
I have tried setting the RotateToGoal critic's scale to 1000 and lookahead_time to 0.1 as suggested here: https://answers.ros.org/question/354523/ but that didn't solve it. I also tried decreasing the slowing_factor a bit, but it didn't work either. I also changed the position of the goal in case it was some inflation overlap, but that didn't seem to affect it.

SteveMacenski, 2025-04-04:
please look at our configuration docs and defaults in `nav2_bringup` for Nav2 resources for system tuning. If you have further questions, please lookup similar questions and ask on Robotics Stack Exchange if not answered already.

closed 2025-04-04 by SteveMacenski

Flamethr0wer, 2025-04-04:
@SteveMacenski I think I've looked at every single configuration guide out there and no combination of parameters has even come close to getting rid of these errors. Also, I did post a question on Stack Exchange and have only come here after having bumped it for almost a week now. I really don't think it's about "tuning".

Flamethr0wer, 2025-04-08:
@SteveMacenski Please I need help I still can't solve this error no matter what I do and I'm still bumping the SE question to no avail. I don't know what to do anymore

SteveMacenski, 2025-04-08:
I just copy+pasted the params in the [DWB Controller config guide page](https://docs.nav2.org/configuration/packages/configuring-dwb-controller.html#example) and things seem to be working fine - this is something to do with your configuration if that does not work - please checkout the standard TB3/4 bringup sim files in `nav2_bringup` to verify to yourself that these are working correctly.

Flamethr0wer, 2025-04-08:
@SteveMacenski Those seem to work... But then what might be causing it? I haven't seen anything that problematic in my files, they're pretty close to the default...

Flamethr0wer, 2025-04-11:
> please look at our configuration docs and defaults in `nav2_bringup` for Nav2 resources for system tuning. If you have further questions, please lookup similar questions and ask on Robotics Stack Exchange if not answered already.

I have tried using all the default critic values just in case, but it's still not working.

Flamethr0wer, 2025-04-11:
> please look at our configuration docs and defaults in `nav2_bringup` for Nav2 resources for system tuning. If you have further questions, please lookup similar questions and ask on Robotics Stack Exchange if not answered already.

@SteveMacenski I have tried using all the default critic values just in case, but it's still not working.

Bob-YsPan, 2025-07-04:
@Flamethr0wer seems like the `min_vel_x`, and `min_speed_theta`, when I try to increase these values a little bit to prevent my robot base cannot moving when the velocity is below a certain value (About 0.05 m/s for x axis, 0.5 rad/s for theta. But it seems like more examples just keep 0 and will cause the robot to get stuck because the moving speed is too small.), but it will cause the same problem and cannot approach the goal.

I am still looking for a solution to this problem.

Flamethr0wer, 2025-07-04:
@Bob-YsPan Yeah what I ended up doing was setting min_vel_x and min_vel_xy to 0.0. I believe the demand for a non-zero linear velocity was preventing the robot from simply turning in place at the goal's location. I had set them to a small value to avoid stalling the motors, but it appears it was a bad idea.