ros-navigation/navigation2 #5060: Is it possible/advisable to run nav2_costmap_2d by itself without the rest of Nav2

pitosalas, 2025-04-05:
For my purposes, all I need is a 2dcostmap based on /scan and anchored in the /odom tf. 

I've been trying therefore to create a node that uses just 2dcostmap. On the belief that it will be simpler and faster than running the whole stack.

But then I am questioning my approach. As the developers would you say that using just that package (https://github.com/ros-navigation/navigation2/tree/main/nav2_costmap_2d) is a good idea or would it be under the umbrella of theoretically possible but untested and therefore unsupported.

If what I am doing is a good idea, then any tips?

SteveMacenski, 2025-04-07:
Sure thing, there's already a [Costmap 2D node](https://github.com/ros-navigation/navigation2/blob/main/nav2_costmap_2d/src/costmap_2d_node.cpp) in the costmap package, there shouldn't be any need for your to create a node at all. 

No issues / concerns. The only thing is to understand why the costmap nodes are embedded in the planner and controller servers in the first place. This was done to have zero-latency access to the internal data structure for planning and control use to ensure safety and the most up to date knowledge of the environment to generate plans and controls from. When you control at, say, 100hz that's every 10ms. If your message latency on a rather large data structure like a costmap is 2-3ms, that can be a pretty big delay in how you respond to near-by obstacles since the local costmap is typically in the rolling local odometry frame.

So, I put them in the most sensitive servers for that reason. However, if you don't need the planners/controllers then its totally a reasonable choice to use the costmap by itself. 

It may be a good improvement / contribution back to the stack actually to make the costmaps parameterized. Set a flag about whether to use a costmap subscriber or the actual costmap inside of the server if folks have the desire for whatever reason to distribute them and/or use a different implementation with the existing planner and controller algorithms. Food for thought.

pitosalas, 2025-04-07:
Thanks. I am still coming up to speed on understanding of Nav2... E.g. The "servers"... I see mention of "containers" in the code are they them or related? I have not yet figured them out. Also when I tried to run it by itself I got errors that were more configuration related so it never quite ran. Is there an example you can point me to to show me how to run it by itself? I If needed I can show you my work so far, I have spent a good deal of time learning this but it is quite complex and there's a learning curve (as you know).

SteveMacenski, 2025-04-07:
The various nodes running are the "servers" (planner server, controller server, behavior server, etc). The containers are the process management of the servers into a single process for IPC and performance optimization purposes. 

If you're having configuration related issues, your best bet is Robotics Stack Exchange for Q&A. We don't provide that here in the issue tracker.

closed 2025-04-14 by SteveMacenski