ros-navigation/navigation2 #5063: Please remove the dependency on ros-gz-sim (cleanup for people using real hardware)

brean, 2025-04-07:
## Bug report
I'm not sure if its really a bug, more of a suggested change: 

**Please remove the dependency on ros-gz-sim.**

turtlebot3_navigation2 depends on nav2_bringup which depends on ros_gz_sim and ros_gz_bridge.

I run only on real hardware and do not need gazebo on my machine.

- Operating System:
  - Ubuntu 24.04 Server
- Computer:
  - Raspberry Pi 3b+
- ROS2 Version:
  - jazzy
- Version or commit hash:
  - 1.3.4-1noble.20250124.142240
- DDS implementation:
  - N/A

#### Steps to reproduce issue
try to uninstall all gz-sim-packages, it will try to uninstall nav2 as well:
```
sudo apt remove ros-jazzy-ros-gz-sim
```

#### Expected behavior
nav2_bringup should not depend on ros_gz_sim and ros_gz_bridge

#### Actual behavior
See https://github.com/ros-navigation/navigation2/blob/main/nav2_bringup/package.xml#L23-24

#### Reproduction instructions
None

#### Additional information
None

SteveMacenski, 2025-04-07:
> turtlebot3_navigation2

This is not a Nav2 package, you'd have to take up their dependency choices and what demos they provide with them 😄 

> nav2_bringup which depends on ros_gz_sim and ros_gz_bridge.

We have many launch files that depend on GZ so that is a legitimate dependency of that package and `nav2_system_tests` which both contain simulation files. If you're running on a physical robot, you can exclude the installation of `nav2_bringup` and `nav2_system_tests` to prevent them from installing simulation assets on your computer. Nav2 bringup actually isn't included in the `navigation2` metapackage, you have to select the install of that yourself if doing `apt install ros-humble-navigation2` (another `-nav2-bringup` install step). If they are dependencies of TB3 packages you use, then rosdep would also identify them, but you should select whether that's the right dependency for your needs.

Keep in mind that you can also use `--skip-keys=SKIP_KEYS` in your `rosdep install` step to skip certain keys (like these packages) if you choose. That's a totally reasonable way to handle this in my opinion so you don't need to copy source and don't need to manually uninstall packages later. [We actually do this ourselves](https://github.com/ros-navigation/navigation2/blob/e63fecc64320f20d9a81e4782a416ed12672aa8f/Dockerfile#L77-L78) in our CI to speed things up with SLAM Toolbox. Its also used in the [core ROS 2 install](https://docs.ros.org/en/humble/Installation/Alternatives/Ubuntu-Install-Binary.html#installing-the-missing-dependencies) as well.

Sorry to hear about your issue, but I don't think this is actionable for Nav2. GZ is required for nav2_bringup and nav2_system_tests.

brean, 2025-04-08:
Hi @SteveMacenski,
thanks a lot for the detailed response.
I was missing this part of the nav2_bringup readme:
```
It is our expectation for an application specific robot system that you're mirroring nav2_bringup package and modifying it for your specific maps/robots/bringup needs.
```
and understand it that you would just use it.
So I'll close this for now and suggest to build a turtlebot3_nav2_bringup-package, if I find the time even build it myself.

closed 2025-04-08 by brean