ros-navigation/navigation2 #5064: planner_server: Path PoseStamped headers don't contain frame_id

EnzoGhisoni, 2025-04-07:
## Bug report
The PoseStamped messages of the nav_msgs/msg/Path published by the planner_server have empty headers (no timestamp and frame_id indicated). It causes issue for the visualization on FoxGlove Studio for example.
The header is filled for the Path but not each PoseStamped.

**Required Info:**
- Operating System:
  - Ubuntu 22.04
- Computer:
  - Desktop
- ROS2 Version:
  - Humble
- Version or commit hash:
  - apt: 1.1.17-1jammy
- DDS implementation:
  - Cyclone DDS

#### Steps to reproduce issue
It can be verified by sending a goal on rviz2 and echo the topic /plan 

#### Actual behavior
```
$ ros2 topic echo /plan
header:
  stamp:
    sec: 1744032547
    nanosec: 974855611
  frame_id: map
poses:
- header:
    stamp:
      sec: 0
      nanosec: 0
    frame_id: ''
  pose:
    position:
      x: 4.5
      y: 0.0
      z: 0.0
    orientation:
      x: 0.0
      y: 0.0
      z: 0.0
      w: 1.0
- header:
    stamp:
      sec: 0
      nanosec: 0
    frame_id: ''
  pose:
    position:
      x: 4.7816162109375
      y: 0.1631011962890625
      z: 0.0
    orientation:
      x: 0.0
      y: 0.0
      z: 0.0
      w: 1.0
```

As the headers are empty, it causes issues with Foxglove Studio integration as it requires the frame_id of the PoseStamped to handle the visualization.

#### Expected behavior
It would make sense to fill in the header for a stamped messages.
Is this behavior intentional or can I send a PR to fill the path's PoseStamped headers?

SteveMacenski, 2025-04-07:
>  can I send a PR to fill the path's PoseStamped headers?

Sure thing! I think of the timestamps belonging to a time-series path with times for each pose (a kind of poor-man's trajectory), but populating them all with the same timestamp as the main message header is totally fine. 

Just make sure to do so against all the planners (navfn, theta star, and the 3x smac planners) -- but it should be < 15 minutes of work.

SteveMacenski, 2025-04-17:
Merged!

closed 2025-04-17 by SteveMacenski