ros-navigation/navigation2 #5066: Docker server Controller not subscribing to namespaced topics

SirFireByte, 2025-04-08:
## Bug report

**Required Info:**

- Operating System:
  - 
- Computer: Ubuntu 24.04
  - 
- ROS2 Version: Jazzy
  - 
- Version or commit hash: 1.3.5-1noble.20250210.222234
  - 
- DDS implementation: cyclon
  - 

#### Steps to reproduce issue

While trying to debug why the loacal costmap is not getting published when namespacing my robot(solved in [here](https://github.com/ros-navigation/navigation2/issues/4905#issuecomment-2706339656)), i saw, that the Docker Server subscribed to not namespace topics of the local_costmap.To reproduce this launch the tb3 simulation with a namespace
```
ros2 launch nav2_bringup tb3_simulation_launch.py namespace:=tb3_0
```

#### Expected behavior
The Node should subscribe subscribe to the namespaced local_costmap.
#### Actual behavior
When Running `ros2 node info /tb3_0/docking_server` subscribs to non-namespaced topics, while the needed data is published on namespaced topics
``` /tb3_0/docking_server
  Subscribers:
    /clock: rosgraph_msgs/msg/Clock
    /local_costmap/costmap_raw: nav2_msgs/msg/Costmap
    /local_costmap/costmap_raw_updates: nav2_msgs/msg/CostmapUpdate
    /local_costmap/published_footprint: geometry_msgs/msg/PolygonStamped ```
#### Additional information

I believe this is due to changes to the nav2_param.yaml file when adding collision detection for docking in #4752.
```
 controller:
      k_phi: 3.0
      k_delta: 2.0
      v_linear_min: 0.15
      v_linear_max: 0.15
      use_collision_detection: true
      costmap_topic: "/local_costmap/costmap_raw"
      footprint_topic: "/local_costmap/published_footprint"
      transform_tolerance: 0.1
      projection_time: 5.0
      simulation_step: 0.1
      dock_collision_threshold: 0.3
```

SteveMacenski, 2025-04-08:
@SirFireByte just remove the leading `/` so that the namespace will be applied -- like so: https://github.com/ros-navigation/navigation2/pull/5068

Does that resolve your issue? If so, I'll backport to Jazzy too

closed 2025-04-08 by SteveMacenski

SirFireByte, 2025-04-08:
Removing the leading "/" did remap it properly, however i did not test the actual docking as i wasn't using it. I just found it while debugging other issues.

SteveMacenski, 2025-04-08:
OK - its been backported!