ros-navigation/navigation2 #5070: Store last known possition using AMCL directly!

techtasie, 2025-04-09:
## Feature request
After restarting nav2 the robot should know it's last known position.
#### Feature description

I’m currently looking for a way to persist the last known pose of the robot so that, after restarting the Nav2 stack, the robot can be automatically localized again without having to set an initial pose manually. In our use case, we typically don’t move the robots by hand, so restoring the last pose would be very practical.

#### Implementation considerations

Even if AMCL crashes, it should be restarted automatically, and the robot should keep its position. Ideally, this involves writing to a file in a way that is resistant to corruption.

I would be interested in Implementing it myself once all the details are clear.

SteveMacenski, 2025-04-09:
This would be a really nice feature! Rather than building it into the AMCL node directly, I think this makes sense as a separate node that can spawn on bringup and once it has the initial pose set and accepted, it can exit. That way its a nice modular utility (that maybe its pattern can be repeated in the future for other similar state-persisting variables).

We could create a node that takes as a parameter the file directory to persist state data information within. Then, we subscribe to the AMCL pose topic and dump it to that file periodically. On startup, if the file is detected, we take its value and send it over to AMCL's set pose topic (so we have ACK that it was accepted). 

Another param could be whether to only dump the pose to file, set the pose on startup, or both. 

If this is designed well, this could be abstract of AMCL set-pose specifically and could be later applied to other topics state variables that should be persisted in some applications.

Vedant87, 2025-04-10:
Hello @SteveMacenski and @techtasie , I would like to help.

SteveMacenski, 2025-04-11:
@Vedant87 Great! Do you have any questions based on this conversation so you may get started?

Vedant87, 2025-04-11:
I've understood the proposed implementation. And I'll get back to you soon with any questions.

pranavk-2003, 2025-04-11:
How does it sound if we make this as a service? the service and be called whenever there's a need to restore the last state of the robot.The saving works the same as saving the pose of the robot periodically.The launch file takes the location of the saved file as a parameter and starts the node which saves the pose. when the service is called, it could simply set the pose to what the last know pose of the robot was as in the saved file.

pranavk-2003, 2025-04-11:
I tried making it as a separate package and well i got till here which is working , but i think making it a service would be better

https://github.com/user-attachments/assets/f25a147c-96e1-4b26-8c94-9678f36e0492

techtasie, 2025-04-11:
I think the best approach is to write the position to a file. To avoid corruption, I’d first write the data to a temporary file, then  replace the original file with the temp file. This works reliably on both Windows and Linux, since replacing a file is an atomic operation. If a power loss occurs during the write, modern journaling file systems like ext4 or NTFS will prevent data corruption by ensuring file system consistency.

pranavk-2003, 2025-04-13:
> I don't fully undersatnd what you mean by the question _How does it sound if we make this as a service_. If we want to reinitialize the robot on startup, it would need to trigger on initialization. I think the storage to file would need to happen automatically at some configurable time interval. How it calls AMCL to set the pose should be a service, but I'm not sure what other service should exist if we want this to initialize the robot automatically to its previous state.

What i meant was , there would be services like start_pose_saver,stop_pose_saver etc.i did made a final version on it.Here's a video demonstration.i think its polished enough.Can this be added to navigation2 repo? @SteveMacenski 

https://github.com/user-attachments/assets/d1957ce7-8ebc-4c41-bf8a-1de95310a511

SteveMacenski, 2025-04-15:
I think having options that it: (a) starts with the recording on startup rather than waiting for a start service and (b) whether to load the initial pose from file on startup automatically rather than waiting for an explicit call. 

I think that would round things off for contribution back! I'm thinking it could go into a couple of places - possibly a new package within [navigation2_auxiliary](https://github.com/ros-navigation/navigation2_auxiliary), its own repository, or nav2_util. If we had a larger collection of these kinds of tools, I think we could make a new package within Nav2 itself with the collection!

pranavk-2003, 2025-04-17:
> I think having options that it: (a) starts with the recording on startup rather than waiting for a start service and (b) whether to load the initial pose from file on startup automatically rather [...]

Great ! Makes sense to make saving pose and load initial pose automatic.What i did is keep the manual services , as it might be useful while debugging and have these services handy instead of restarting the node everytime and added two parameters which automate the pose saving and pose restoring.

I think it would be the best if a new package named "nav2_toolkit" can be made which shall contain all packages which add quality-of-life / automation features like this.If its now feasible to add a new package now ,it suits better in nav2_util.Can i proceed to make a PR?Let me know where shall this package be placed.Pleased to contribute to nav2!!

SteveMacenski, 2025-04-17:
I agree having the services would be nice, the automatic setting initial pose on startup I think should just be a param option, though might make sense to be default on (then subsequent service requests can be sent to set it again if other things get restarted). 

Maybe a `nav2_toolkit` would make sense. I think the `BaseFootprintPublisher` in nav2_util could be moved over there as well as another good common-place utility. 

Can you we brain storm a few other tools that might be nice to have included in this toolkit that we're missing? It would be nice to introduce this with a small handful of things to start off with.

pranavk-2003, 2025-04-18:
Yes , automatic setting initial pose is a param. but wouldn't it conflict if this param is default on and it could happen that it loads a pose from a previous session of simulation? for this i think it'll be better if kept default off, and if navigation happens to crash , or a need to relocalize the robot on startup, we just can relaunch the node with this param set to true and if we dont wish to relaunch the node again, the services can anytime be used for relocalisation.

 I’d love to help seed a new nav2_toolkit package with this and a few other small utility nodes. I agree that the BaseFootprintPublisher would be a great fit there too.To get started with i think we can have 

- System Resource monitor: Tracks , logs and alerts for usage of system resources relevant to robot navigation performance. Would be useful to monitor performance on target devices.
- Goal Repeater : This could be an extension to the pose restore , which would be a function which would track the goal progress and if theres a problem before "Goal is Succeeded" then can repeat the last goal after the recovery.
- Map validator : A node which monitors the current map and checks for issues like All-zero maps,Corrupt or empty maps being loaded.
- Benchmark Planners : given a map  and a goal , it would benchmark different Planners in nav2.

Not sure about the feasibility of these ideas but if implemented i think they can be handy. 

More Complex :

- Map switcher : Basic idea is to switch maps for a robot which operates in a multi-level environment as it goes from one level to another.
- Planner switcher : Switch global planner at runtime accessing certain conditions.

techtasie, 2025-04-19:
@pranavk-2003 many of these things are already included in the core nav2 stack. The goal Repeater and Planer switcher are  already realized with the [behavior tree](https://docs.nav2.org/behavior_trees/overview/detailed_behavior_tree_walkthrough.html). The Map switcher is already a [service](https://docs.nav2.org/behavior_trees/overview/detailed_behavior_tree_walkthrough.html) offered by the map server.

I think the System Resource monitor is an issue that in my opinion should not be solved by ros2 nav2 but by your usual system monitoring tools. 

But I really like the idea of a benchmarking tool. Especially when doing recordings for a paper, it is quite cumbersome to get performance metrics. Our current approach is to record rosbags and then run some Python scripts to analyse them after the fact. But maybe a simple node that records navigation performance metrics, like: average speed, recovery behaviour count, path driven vs. path planned, etc., would probably be quite useful.

SteveMacenski, 2025-04-22:
> Yes , automatic setting initial pose is a param. but wouldn't it conflict if this param is default on and it could happen that it loads a pose from a previous session of simulation? 

Ah, yes, if we have this new node on by default, I agree with that. I guess I didn't imagine having this part of our default launch configuration, but if we do, I agree for simulation it would need to be off. But, that could be default on and we set the parameter in our nav2 yaml file to be off for the simulation case. 

In that case, I think we should make a new nav2_bringup launch file for the toolkit so that we separate out the tools from the other system bringups so they're easy to enable / disable by a runtime launch parameter. I'm really liking this idea! 

Another idea that I had that comes up reasonably often is a "pause/resume" behavior where you can pause (cancel) navigation and resume (send goal again) behavior. I've always recommended people implement this either in their BT XML (which is easy enough) or making a utility like this. It would be good to just have the utility available. 

> System Resource monitor

You may be able to strip that out of my `opennav_amd_demos` repository and add additional system metrics. I do tend to agree with @techtasie that this might not be a great Nav2 tool and there are some other ROS 2 utilities in the ecosystem that handle this. But, some other monitors might be nice if they're Nav2 or mobility specific metrics.  

> Goal Repeater

Can you clarify this one more? I'm not sure I understand. 

> Map validator

I think the map server error codes / lifecycle handles this? 

> Benchmark Planners

We actually have that in `tools/` already for offline use. I don't think that should be in the toolkit since that's an offline testing utility, not runtime. 

> Planner switcher 

The servers already handle this actually and there's BT nodes for it as well. I'm not sure there's much use for an additional utility 

> Map switcher

This would just be calling a service. I suppose we could expose a script that just takes in a new filepath to send to the service as a CLI command, but not much use for runtime deployed use, more debugging / lab use.

pranavk-2003, 2025-04-23:
> In that case, I think we should make a new nav2_bringup launch file for the toolkit so that we separate out the tools from the other system bringups so they're easy to enable / disable by a runtime launch parameter. I'm really liking this idea!

> Another idea that I had that comes up reasonably often is a "pause/resume" behavior where you can pause (cancel) navigation and resume (send goal again) behavior. I've always recommended people [...]

These are great thoughts. Surely must go forward with them.

>opennav_amd_demos 

will check this out , please pin any additions links if any for the same

>Can you clarify this one more? I'm not sure I understand.

Basically you can compare this with the pose_restorer.That means if navigation crashes in between a goal is being executed.after restarting navigation, after the robot relocalizes,we can again call a service which would execute the goal which was not successful previously.

>I think the map server error codes / lifecycle handles this?

>Benchmark Planners

>We actually have that in tools/ already for offline use. I don't think that should be in the toolkit since that's an offline testing utility, not runtime.

>Planner switcher

>The servers already handle this actually and there's BT nodes for it as well. I'm not sure there's much use for an additional utility

I guess yes , i explored and yes these already exists.

So final tools for `nav_toolkit` would be :

- pose_saver & pose_restore with added launch file in nav2_bringup.
- resume and pause goals (can add goal repeater to this)
- nav2_resource_monitor
- also BaseFootprintPublisher as you said earlier.

Looks good to introduce `nav2_toolkit` with these?What i think is it would be good if we introduce with **BaseFootprintPublisher** and **pose_saver & pose_restore**. The reason behind this is that as community uses these tools , we can get issues which would improve these faster and then actively work on the remaining tools.

techtasie, 2025-04-23:
> Basically you can compare this with the pose_restorer.That means if navigation crashes in between a goal is being executed.after restarting navigation, after the robot relocalizes,we can again call a service which would execute the goal which was not successful previously.

Then I misunderstood your idea. This sounds good to me :+1:

SteveMacenski, 2025-04-23:
Maybe we should skip the system metrics piece unless we can come up with a few mobility-specific ones.

I think something like the pause/resume behavior would potentially be a better candidate than the Goal Repeater. Possibly this node could be built to listen and store for resume consistently so that it could be used without a specific `pause` command. What do you think? That would do the repeater-like behavior but targeted more for pause/resume. If we have an old goal on file and asked to resume (without a pause commanded) then it could be done. I think of that more as a backup or use between emergency reboots than 'if Nav2 crashes' because Nav2 should never crash.  

Sounds good to me!

techtasie, 2025-04-23:
> Maybe we should skip the system metrics piece unless we can come up with a few mobility-specific ones.

The best ones I can come up with is: Localization Confidence, Lifecycle Manager Heath, failed goal count etc...

But I feel like most of them are not directly useful expect for Localization Confidence and this is available already....

> I think something like the pause/resume behavior would potentially be a better candidate than the Goal Repeater. Possibly this node could be built to listen and store for resume consistently so that [...]

I agree this was my intention behind the storing of the last position.

> Sounds good to me! I think these should be ideally C++ so that they can be used run-time on the robots. I know file IO in C++ is more annoying, but luckily AI code gen tools are nice to get that boilerplate 'I have a filepath, store this data to that file in a protected way' going.

I would refer to my post from earlier about how to write to a file, since AI tends to work best when giving implementation instructions along. This is probably the the best approach that is guaranteed to worki n all edge cases.
> I’d first write the data to a temporary file, then replace the original file with the temp file. This works reliably on both Windows and Linux, since replacing a file is an atomic operation. If a [...]

pranavk-2003, 2025-04-25:
> I think something like the pause/resume behavior would potentially be a better candidate than the Goal Repeater. Possibly this node could be built to listen and store for resume consistently so that [...]

Sounds good! but i still think having an option of pausing with all this would be ideal.This would make it pretty flexible and useful instead of it only having purpose of repeating a goal.

So final tools (Updated) for nav_toolkit would be :

- [ ] pose_saver & pose_restore with added launch file in nav2_bringup.
- [ ] resume and pause goals (can add goal repeater to this)
- [ ] also BaseFootprintPublisher as you said earlier.

SteveMacenski, 2025-04-25:
Great! I think this is a good addition to the stack!

pranavk-2003, 2025-04-26:
Hey, had a quick question , how shall we move forward with addition to the Stack? make a new branch and then merge? or have a PR directly after all the tools are ready.What i think would be better is to make a new branch and then have these tools made there.As pose saver is almost done , it can be added to the branch right now.Also BaseFootprintPublisher needs to be just moved can me moved to the branch as well.

SteveMacenski, 2025-04-28:
Yeah, so you make a branch on your fork and then submit a PR back to `ros-navigation/navigation2`. Then we review and merge through GitHub's UX there after a series of CI jobs run to check on code quality and functionality. 

I think you should just submit a draft PR with a TODO check list of what is needed before I should give it a full detailed review. But, I can give you a review with broad strokes before then in case there are big obvious things that we need to address (like missing features, architecture, or test coverage). If you have the first 2 nodes in the toolkit, I think that's a good time to open the draft PR for us to take a look together while you work on the last one :-)

SteveMacenski, 2025-04-30:
Great, I'll take a look!

I think just the relocation would be good. That's all I think should be done, maybe a namespace needs to be updated in the files for the new package, but otherwise it should be a simple translation.

pranavk-2003, 2025-05-02:
> Another idea that I had that comes up reasonably often is a "pause/resume" behavior where you can pause (cancel) navigation and resume (send goal again) behavior. I've always recommended people [...]

I was trying to implement this but cant get it working / cant decide which approach should be followed , i would appreciate a briefing about it.

SteveMacenski, 2025-05-02:
@pranavk-2003 can you talk a bit about what direction you've gone and what issues you ran into? :-)

pranavk-2003, 2025-05-03:
I am not that familiar with advance Nav2 and BT.Sowhat i tried first is using services to get the pause resume behaviour to test the logic.(Didn't work) and i think because as its a action server,cant use a method i used in pose_persistent by subscribing the topics?still unsure.Then got started with having a BT and  still learning it, but feel lost.So if i could get a brief idea about the approach which shall be used , most preferred, i'll cope up.

SteveMacenski, 2025-05-06:
The action servers have "hidden" topics for the request, feedback, and response that the action client and server communicate. Actions are implemented in ROS 2 and not a type within DDS so its all done over topics that you can introspect. You can see this yourself if you use `ros2 topic list --include-hidden-topics`. So, you could potentially listen to this topic and store the last goal that's processing. When a response is sent, you can delete that as a completed non-pending goal. 

So when you get a 'pause', you can use that goal to 'resume'. I'm actually not sure what the QoS settings for those action server hidden topics are by default. If they're transient local, then you could actually obtain the goal _when the pause service is called_ rather than always listening. 

So then you would have to do something in the 'pause' service like:
- Log the pause, with the goal info for debugging
- Call the action server to cancel the current goal (and you should have the current goal info to do so)

On resume:
- Log the resume with respective information 
- Call the action server with the original goal with updated timestamp.

Then, you can create BT nodes to trigger those pause/resume actions. 

---

Separately, we could implement a pause / resume within the Behavior Tree itself.

In this case, we have a node that exposes a pause and resume service. When its in the paused state, then this node returns something that within the BT logic just skips doing anything. Else, it returns a value that lets the BT continue. 

When the BT paused state starts, we'd just need to cancel any running nodes -- and lucky for us that all of the action servers have a Cancel BT node that you can simply call at that point. So maybe the pause/resume BT node could be a decorator that when its a new pause, does a sequence of these cancelations. 

This has the added benefit of not requiring any topic magic or storing values, it just uses the values in the BT blackboard already and we don't need to resend actions. The downside is that it wouldn't persist between boots. 

---

Arguably, given how easy this one conceptually is to do, we may find it beneficial to offer both for those that like one over another. I would be interested in community feedback if everyone just likes one over the other so that maybe we don't need to offer both, but both are definitely possible. Maybe worth posting on discourse to ask for feedback?

techtasie, 2025-07-18:
@pranavk-2003 any updates on your side otherwise I would take over and finish up the implementation.

pranavk-2003, 2025-07-18:
> [@pranavk-2003](https://github.com/pranavk-2003) any updates on your side otherwise I would take over and finish up the implementation.

yes , i am almost done with the pose_saver & pose_restore.If you can help with the pause and resume tool, would be great!

mini-1235, 2026-01-20:
@pranavk-2003 any updates? @techtasie would you be interested in taking this over?

SteveMacenski, 2026-01-20:
Related: https://github.com/ros-navigation/navigation2/issues/5883

SteveMacenski, 2026-01-29:
https://github.com/ros-navigation/navigation2/pull/5922 resolves

closed 2026-01-29 by SteveMacenski

techtasie, 2026-01-30:
@SteveMacenski THANKS 👍