ros-navigation/navigation2 #5072: Constrained smoother cost function shape

LinusTxtonomy, 2025-04-09:
## Feature request
The constrained smoother has a cost function in ceres, that is not aligned with the documentation of ceres and causes poor convergence. The documentation can be found [on the ceres page](http://ceres-solver.org/nnls_tutorial.html). The cost function squares all residuals even though ceres does that internally and it should not be done by the user. This causes the linearization being bad which in turn makes the trust region smaller and slows down the convergence a lot in scenarios where the costmap is causing the highest residual values.

#### Feature description
Squaring of residuals need to be removed in all residuals. For example remove the extra multiplication by value here:
https://github.com/ros-navigation/navigation2/blob/d518dd372f8bf649652e4ca1a2d6d148e2400acc/nav2_constrained_smoother/include/nav2_constrained_smoother/smoother_cost_function.hpp#L213

#### Implementation considerations
Since this changes the shape of the cost function, the weights also need to be adjusted. Is there an example that can be used to tune the parameters?

SteveMacenski, 2025-04-09:
OK - want to give that a test to make sure it improves performance and quality? Happy for the contribution :-) Please just make sure to highlight the changes in the PR (runtime, quality, etc) 

For an example for tuning, I don't have a particular one in mind. I've used some warehouse maps and the TB3 sandbox world in `nav2_bringup` to make sure I've tuned against both smaller confined spaces and larger open spaces.

SteveMacenski, 2025-05-03:
@LinusTxtonomy can you give me an updated timeline for a PR to fix this after you have a chance to test?

SteveMacenski, 2025-05-16:
@LinusTxtonomy any update? I think this is very sensible, but I just need you to verify testing that this works well (better performance, better quality).

LinusTxtonomy, 2025-05-17:
> [@LinusTxtonomy](https://github.com/LinusTxtonomy) any update? I think this is very sensible, but I just need you to verify testing that this works well (better performance, better quality).

Sorry for the late response, I have done some testing but was not at home to finish and it and create a PR. I will do so next week.