ros-navigation/navigation2 #5082: Nav2 Route Server Improvements

SteveMacenski, 2025-04-12:
The following are a list of known possible improvements or extensions of the Nav2 Route Server that I would like the community and/or users of this work to help implement for their own application needs.

## Provide a 'force to go through these edges' capability

Its been requested in the requirements section of this work to make it possible to force the robot to go through certain edges as part of its routing solution. 

## Stop / clearance -- Open-RMF inspired

Have an option to set clearance for each edge for lock-outs of shared resources. Right now, we can implement this using the DynamicEdgeCost service in the EdgeScorer, but there are probably some good add-ons for full feature parity and/or providing a service client util for submitting and modifying these for a fleet in an easy way. 

## Show lanes effectively implemented using this feature, with demo

This work can be used for creating lanes for the robot to stay within. Show a demonstration of this with the lanes being marked by edges in the graph and the metadata containing the maximum distance it is permitted to be away from the centerline as the lane's width. Use this in conjunction with the BT to ensure enforcement of this constraint. 

## Built in path / route turning radius at intersections

Rather than post-processing the route path with the smoother server, it would be ideal to have the server provide a minimum turning radius option to make turns use a fixed turning radius to smooth out turns when closely tracking the route graph. 

## Dynamic parameter handling

This is relatively straight forward: allow use of dynamic parameters, taking care to  mutex lock resources so that  changes cannot happen during an active request. 
 
## Edges have non-straight paths. 

Its been requested to have the edges be non-straight paths to symbolize more complex features. This can be implemented with a metadata field of the path and/or vectorized representation. When we reconstruct the route path and/or during Edge Scoring, we can use this path to construct the path to take and compute the length for scoring, respectively. We just need to take care to properly transform it into the current frame or make conventions about how these are represented. 

## Persist blocked edges between calls and the timemarker betwen sessions

Discuss solutions to persist some data between planning requests and/or sessions. Things like the time markers of how long it takes to go through an edge for later improved scoring or the blocked state of certain edges may be nice to persist for more than a single task request. 

## Demo showcasing ComputeAndTrack route option, not just ComputeRoute

Basically the same as the 2 demo BT XMLs in the `nav2_bt_navigator` package, just now with route operations to do something during the task, like change speeds and handling the fact that that server is long-running like the controller server.

## Tutorial showing how to add route operations to the geojson file and create the new route operation plugin to use that metadata

Related to above, also show how to edge a graph file to include a custom route operation, write that route operation, and show it at work so folks can more easily extend this work to their application specific needs

VivekVRobo, 2026-10-08:
I'd like to take item 5 (dynamic parameter handling) from this issue.

I'll keep the scope limited to the Route Server's existing parameters:

1. identify parameters that are safe to update at runtime,
2. add a dynamic parameter callback with validation,
3. protect shared state with the appropriate mutex so parameter changes cannot race an active route request,
4. preserve current startup/default behavior,
5. add focused tests for accepted/rejected updates and concurrent request safety where practical.

I won't touch item 8 or the other Route Server improvements in this PR.

I'll prepare this in a fork first and come back with the implementation and test results before opening the upstream PR.

SteveMacenski, 2026-10-08:
Great, thanks! That makes sense :-) Feel free to look at other Nav2 servers / utilities for examples of the common design patterns for how we handle / test dynamic parameters. I look forward to it!

yexy1816, 2026-11-02:
Hi Steve, I'm very interested in working on item 3: "Show lanes effectively implemented using this feature, with demo."

I'm thinking of implementing it in the following stages:

1. Automatically generate an additional costmap layer based on the defined route graph, with a configurable "maximum deviation" parameter for each edge. Areas outside the permitted corridor would be assigned a lethal cost. I'm wondering whether this could be achieved by extending or dynamically generating keepout zones using the existing KeepoutFilter.

2. Add a function to the controller server to reduce linear velocity or adjust angular velocity when the robot approaches the lane boundary. Ideally, this would be independent of the specific local controller being used.

3. For the BT, check if the robot is still in the boundary, introduce recovery behaviour to guide the robot back towards the route centreline if it deviates beyond the permitted boundary.

4. Define a demo in simulation, maybe using some auto-generated route graph? 

Separately, I'd also be interested in improving the usability of the RViz Route Tool. Some ideas include copying and pasting existing routes, dragging and dropping nodes and edges to adjust their positions, and improving the graph saving and loading workflow.

Please let me know if this is feasible for the current architecture, and please let me know what you think!

SteveMacenski, 2026-11-03:
Hi- great! 

1. Perhaps yes! Also could be an added constraint sent only to the MPPI critics to mark trajectory samples that leave the bounds as illegal (honestly might be probably good to do both for flexibility of solution). I think there's a currently open PR for a costmap layer that never quite got over the line which you may or may not want to take inspiration from. Review the discussion. I imagine this would be for the local costmap layer only? 

2. Is this common from your experience? If so, that's a nice feature, but also don't want to overcomplicate things for speculative features. This may be a reason for MPPI critic(s) actually. 

3. 👍 

4. Sure, or one of the ones we have already for the TB4 worlds

yexy1816, 2026-11-04:
Hi Steve, thank you so much for your suggestions!

I was thinking of a costmap layer generated by the route graph that could be applied to either the global or local costmap or both, similar to the Keepout Filter. From my experience with the route server, the robot can sometimes take "shortcuts" rather than following the defined route. Applying corridor constraints to the global costmap could help prevent this. I think it would be useful if the user has the choice of adding the layer in global/local costmap. One limitation would be first mile and last mile navigation, where the robot or goal may be outside the route graph. However this might be desirable for applications where the robot is expected to remain strictly within predefined routes? 

I'm reviewing PR #5605 and can see why you suggested implementing this in the local costmap. I'll take a look at the implementation and if I can think of a more efficient way. I'm also interested in exploring an MPPI critic as an complementary approach. I'll experiment with both in my fork :).

SteveMacenski, 2026-11-04:
Agreed - having a global version would be nice too so that you could do a route graph search to find the general strategy for navigation. Then you could still do global planning, but then its constrained to what's possible within the bounds of the searched route. I actually have a project in mind right now that could use this feature, so I could even test this out 'for real'. 

Separately, the local costmap for path tracking use in the controller. 

> One limitation would be first mile and last mile navigation, where the robot or goal may be outside the route graph.

We could do something similar to the keepout zone. I believe we added a feature earlier this year that checks if the robot is in the keepout zone. If so, then we actually make the lethal cost into just below lethal cost, so that it allows the robot to navigate, but heavily incentivizes it to get back into the groove as immediately as possible. That prevents it from getting stuck. I think a similar thing could be applied here. Or, in the behavior tree where the first- and last-mile are happening, we can send a disablement/enablement when not currently on the graph. I figure why not offer both! The second option is just a simple service call & a 40 line BT node. Probably worth having the ability to toggle it regardless.

We just need to focus on efficiency for this - so in the global costmap we need to take some care :-)

yexy1816, 2026-11-04:
Thank you for pointing the direction! It all make sense, I like the idea of having a "global" layer which doesn't stuck the robot when it accidentally enters the keep out area, will let you know once I made some progress!

SteveMacenski, 2026-11-05:
Feel free to ping me on slack too if you have any quick questions!

yexy1816, 2026-11-12:
Created draft PR #6781 will keep working on this.