ros-navigation/navigation2 #5086: Default Behavior Tree design

mini-1235, 2025-04-13:
As noted in https://www.behaviortree.dev/docs/nodes-library/SequenceNode, the Reactive family restarts the entire sequence if its child is return RUNNING.
Thus, in the following code, `GoalUpdated` will only be checked if the `RoundRobin` returns RUNNING after the first tick
```cpp
<ReactiveFallback name="RecoveryFallback">
  <GoalUpdated/>
  <RoundRobin name="RecoveryActions">
    <Sequence name="ClearingActions">
      <ClearEntireCostmap name="ClearLocalCostmap-Subtree" service_name="local_costmap/clear_entirely_local_costmap"/>
      <ClearEntireCostmap name="ClearGlobalCostmap-Subtree" service_name="global_costmap/clear_entirely_global_costmap"/>
    </Sequence>
    <Spin name="SpinRecovery" spin_dist="1.57" error_code_id="{spin_error_code}" error_msg="{spin_error_msg}"/>
    <Wait name="WaitRecovery" wait_duration="5.0" error_code_id="{wait_error_code}" error_msg="{wait_error_msg}"/>
    <BackUp name="BackUpRecovery" backup_dist="0.30" backup_speed="0.15" error_code_id="{backup_error_code}" error_msg="{backup_error_msg}"/>
  </RoundRobin>
</ReactiveFallback>
```
However, as I checked the implementation of `ClearEntireCostmap`, `Spin`, `Wait`, `Backup`, only `ClearEntireCostmap` will return RUNNING if the future timeouts. In this case, I felt like it is more suitable to have `GoalUpdated` abort the `RoundRobin` block once the goal has been updated, even when it has not found a successful action.
In order to do this, we will need each child to return RUNNING before moving to its sibling, I think there are similar nodes named `AsyncSequence`, `AsyncFallback` starting from BT 4.6. https://www.behaviortree.dev/docs/migration/

P.S. I wouldn’t call this a feature request, as it is possible that there is a misunderstanding on my part

mini-1235, 2025-04-15:
Suppose when we first try to clear costmap, the sequence returns FAILURE, then `RoundRobin` will return nothing and moves to the next child
https://github.com/ros-navigation/navigation2/blob/20b6fe58c83e5d260855a412b331d45ddf5923bd/nav2_behavior_tree/plugins/control/round_robin_node.cpp#L60-L64
At this point, it will try to do `Spin` without checking whether the goal is updated or not, isn't it?

SteveMacenski, 2025-04-15:
That's the intra-tick movement. There is a whole thread about that which culminated in this ticket https://github.com/BehaviorTree/BehaviorTree.CPP/issues/954, but that's a bit of work in progress. 

>  At this point, it will try to do Spin without checking whether the goal is updated or not, isn't it?

It will, but then `Spin` will be called, return RUNNING within the same `tick`. Then next tick comes along and then the `GoalUpdated` is checked. At that point, if the goal is updated, then we'd `halt` that Spin action. So it is delayed by at most 1 `tick` frequency, but that's a quirk of how BT.CPP handles many of its control flow nodes and there's not consensus on the best practices (yet). Though I agree that the async method is probably best, I want the consensus to move everything globally to use that.

mini-1235, 2025-04-15:
Ok! Thanks for the explanation and pointing me to the ticket @SteveMacenski, I will close this now

closed 2025-04-15 by mini-1235