ros-navigation/navigation2 #5088: MPPI GoalCritic and Path Inversion Bug

alex-roba, 2025-04-15:
## Bug report

**Required Info:**

- Operating System:
  - ubuntu 24.04
- Computer:
  - 13th Gen Intel NUC
- ROS2 Version:
  - Jazzy
- Version or commit hash:
  - main repository
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
The MPPI controller's GoalCritic behavior, modified in PR #4822, causes the robot to ignore local path inversions and focus solely on the global goal.

1. Set up a Nav2 environment with the MPPI controller.
2. Create a path with an inversion point (e.g., in a narrow aisle requiring a 180-degree turn).
3. Send a goal pose with an orientation opposite to the robot's current direction.
4. Observe the robot bypassing the inversion point and moving directly to the final goal.

#### Expected behavior
The robot should respect path inversions, navigating to intermediary goals (e.g., inversion points) when required, before approaching the final goal.

#### Actual behavior
The robot to ignore local path inversions and focus solely on the global goal.

#### Additional information
Proposed Solution:
Default to using the "real" (global) goal for GoalCritic scoring and other critics.

If a path inversion is detected and inversions are considered (based on a parameter), use the intermediary goal (end of the pruned path or inversion point).

SteveMacenski, 2025-04-15:
I agree that this is an issue.  See more context in this thread: https://github.com/ros-navigation/navigation2/pull/4822#issuecomment-2795578162 for the issue and possible solutions. 

Anyone in the community open to helping with this? It involves adding the parameter for enforcing path inversions and using the global goal pose vs the path's end pose considering path inversions when MPPI is used with a feasible planner like Smac Planner Hybrid-A*.

BriceRenaudeau, 2025-04-16:
@alex-roba , Thanks for creating the issue.

BriceRenaudeau, 2025-04-16:
The quickest solution is to revert the faulty MR #4822

redvinaa, 2025-04-17:
@BriceRenaudeau reverting #4822 would reintroduce the bug that it solved :) Instead, you could open a PR with the solution that @SteveMacenski suggested.

BriceRenaudeau, 2025-04-17:
Yes, I understand, for me it was a feature, not a bug. 👍

SteveMacenski, 2025-04-17:
I've also marked it with help-wanted. I've been noticing many of these tickets have had community members jump in to help too, so I've been leaving nice bite-sized things like this open for them to contribute and grow. If this isn't something @BriceRenaudeau @alex-roba @redvinaa can do, at some point I'll address it myself but wanted to give open opportunity to the community to be involved before I did.

brayanpa, 2025-04-21:
Hi! I'm currently working with the Smac Planner Hybrid-A* and encountered this issue while testing the MPPI controller, which led me to switch to the RPP controller instead. I'd like to take a closer look and work on a definitive fix so that the MPPI controller can be used reliably. If you're okay with it, I’ll go ahead and start working on this.

SteveMacenski, 2025-04-22:
Please do @brayanpa !

brayanpa, 2025-04-25:
Hi! I've opened a pull request with a proposed fix. Although it's a simple solution, I've tested it both in simulation and on the real robot, and it seems to fully address the issues.

@SteveMacenski let me know if this is close to what you had in mind or if any adjustments are needed. I'm happy to make changes and will be monitoring the thread.

SteveMacenski, 2025-04-25:
Great! I'll take a look early next week - probably Monday

closed 2025-05-15 by SteveMacenski