ros-navigation/navigation2 #5091: Active collision avoidance in Assisted Teleop using the VFH algorithm

exMachina316, 2025-04-16:
## Feature request

Reference: https://github.com/orgs/ros-navigation/discussions/4819

#### Feature description  
This feature proposes enhancing the existing **Assisted Teleoperation** behavior in Nav2 by incorporating a **Vector Field Histogram (VFH)-style obstacle avoidance** algorithm. The goal is to enable the robot to **steer away from obstacles** rather than simply **reducing velocity** or stopping when a collision is imminent. This will provide smoother, more intuitive control during teleoperation, especially in semi-structured or cluttered environments.

This approach segments the robot's forward field of view into polar regions and dynamically selects safe steering directions based on real-time laser scan data, allowing for more **robust and responsive obstacle avoidance** during manual control.

---

#### Implementation considerations  
- **Core Logic**:
  - A reusable utility should be developed to convert **laser scan data into polar regions** with associated metadata (e.g., occupancy, safety cost).
  - This utility can then be used in both the **Assisted Teleop behavior** and potentially in a **controller plugin** or other behaviors (e.g., semi-autonomous exploration).

- **Algorithm**:
  - Based on the VFH method, generate a polar histogram representing free and occupied directions.
  - If the current velocity vector lies in an occupied region, adjust the heading towards the nearest viable direction.
  - Weight selection by proximity to goal direction and safety cost.

- **Pros**:
  - More natural teleoperation experience (e.g., remote driving).
  - Improved safety in cluttered or dynamic environments.
  - Reusable logic across other plugins or behaviors.

- **Cons**:
  - Slight increase in computational load.
  - Must be carefully tuned to avoid oscillations in narrow corridors.

- **Design Suggestion**:
  - Still under development.

exMachina316, 2025-04-16:
@SteveMacenski sorry for the delays but I was caught up in academic work. I have some free time now and would like to begin work on this feature.

exMachina316, 2025-04-16:
I have attempted to contact the original author of the paper and am currently awaiting permission to use the paper as the implementation base. I would like some guidance from the maintainers on how the permissions and licensing is to be handled as I am relatively new to this kind of formalities.

BriceRenaudeau, 2025-04-16:
You can use the DWB to achieve this behavior.

SteveMacenski, 2025-04-24:
@BriceRenaudeau have an example or some more info the user can work from in order to create this behavior and contribute it to the assisted teleop behavior? Sounds like you have some first-hand experience. I assume it is along the lines of an obstacle critic married with a critic trying to stick as close to the original control request as possible but collision free. Any more subtle detail or information you can provide would be great. 

Something else to consider is that maybe we want to use only direct current measurements for teleop rather than costmaps. Something like polar-coordinate representation of the sensors in the current snap-shot seems to make sense as a proxy for costmaps, though I suppose a critic could be made that uses that rather than the costmap itself (there's no tie of DWB that needs to use a costmap-based collision checker). Putting that info in a costmap also wouldn't be the worst thing in the world either; we have plenty more compute resources than was possible when techniques like VFH were originally developed. Using DWB also comes with the benefit of making sure the output is feasible, so that's worth considering.

BriceRenaudeau, 2025-04-24:
I have tried several solutions to achieve robust assisted teleoperation for non-expert operators.

The sensor-based solution using polar or Cartesian coordinates is efficient, but you must code everything; it's very robot-related.
It's a good playground for specific algorithms.

If the objective is reliability and easy dev/tuning, I recommend using the Nav2 tools : costmap + DWB.
You can create a node that takes the user input and modifies it.
1. Compute the expected trajectory from the user input
2. Use DWB `computeVelocityCommands()` with the expected trajectory
3. Filter new_cmd_vel to fit your limits then publish

This architecture allows to use separate costmap plugin for different sensors and map, and DWB critics!

exMachina316, 2025-04-24:
Thank you @SteveMacenski and @BriceRenaudeau for your participation and comments. 

The implementation design I came up with was to generate a Cartesian histogram and a polar obstacle map on the map server from scan data at a high frequency. The assisted teleop node can use this polar obstacle map and input velocity to calculate the appropriate contol velocity. 

I appreciate the idea of using DWB's robust architecture and existing infrastructure to build this functionality but it does tend to be compute intensive when it does not need to be. 
Lmk your thoughts on this approach.

SteveMacenski, 2025-04-29:
I think its worth probably taking a step back and doing a survey of techniques that could be helpful to deploy for assisted teleoperation that deflects the control input based on collision (or other criteria?) to maintain intent but adjust the output to be safer. 

Some examples of expected behavior:
* If driving towards a doorway misaligned, adjust the velocity to have better margin from the walls / door frame
* if driving towards an obstacle/wall that may clip the robot in an otherwise open space, slightly adjust the velocity away from the obstacle to give it some margin

I think we should look into the available options in research, industry, other open-source projects, etc and come to some conclusions about directions we would like to go, rather than starting with 1-2 technology solutions and trying to fit them to the problem. 

For example, there's also Artificial Potential Fields, things like ORCA & PVO. Also MPC, optimization techniques, control barrier functions, etc.

---

A benefit of DWB is that it can use a costmap that is going to consider more than just a 2D laser scanner (3D lidar, RGBD camera, sonar, etc) in a sensor fusion method widely used in Nav2. It also can be dynamically feasible due to the sampling. In contrast, it will be somewhat more computationally expensive and due to reliance on a costmap, it may not represent the current state of the world in highly dynamic situations where 10ms update rate may be too slow. 

VFH or similar instead is relatively computationally light, but it only would work with 2D lidars or having to create additional complexity to fuse other sensors in, while also not providing dynamically feasible outputs. But, using live data makes sure it has the most up to date knowledge of the world possible. 

Though... DWB, VFH, APF, or others could be set up to use either live-data or costmap from the local costmap topic.