ros-navigation/navigation2 #5093: Follow Path controller always does a rotataion after reaching the goal

MeghaMaheshwari, 2025-04-17:
I sent a path to my follow path and set the orientation to (0,0,0,1). So my path has x, y and orientation set to (0,0,0,1).  Yet after the robot reaches the goal, it does a roughly 180 rotation. Why does it do so and how can i prevent the rotation? This is my configuration. I tried by removing the RotateToGoal and yet it still does the rotation at the end

FollowPath:
      plugin: "dwb_core::DWBLocalPlanner"
      debug_trajectory_details: True
      min_vel_x: -0.2
      min_vel_y: -0.1
      max_vel_x: 0.5
      max_vel_y: 0.1
      max_vel_theta: 1.0
      min_speed_xy: 0.0
      max_speed_xy: 0.2
      min_speed_theta: 0.0
      acc_lim_x: 2.5
      acc_lim_y: 2.5
      acc_lim_theta: 3.2
      decel_lim_x: -2.5
      decel_lim_y: -2.5
      decel_lim_theta: -3.2
      vx_samples: 20
      vy_samples: 20
      vtheta_samples: 20
      sim_time: 1.7
      linear_granularity: 0.05
      angular_granularity: 0.025
      transform_tolerance: 0.2
      xy_goal_tolerance: 0.1
      trans_stopped_velocity: 0.25
      short_circuit_trajectory_evaluation: True
      stateful: True
      critics: [
        "RotateToGoal", 
        "Oscillation", 
        "BaseObstacle", 
        "GoalAlign", 
        "PathAlign", 
        "PathDist", 
        "GoalDist"
        ]
      BaseObstacle.scale: 20.0
      PathAlign.scale: 32.0
      PathAlign.forward_point_distance: 0.4
      GoalAlign.scale: 32.0
      GoalAlign.forward_point_distance: 0.4
      PathDist.scale: 32.0
      GoalDist.scale: 32.0
      RotateToGoal.scale: 32.0
      RotateToGoal.slowing_factor: 5.0
      RotateToGoal.lookahead_time: -1.0 

## Bug report

**Required Info:**

- Operating System:
  - 
- Computer:
  - 
- ROS2 Version:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

#### Reproduction instructions

#### Additional information

SteveMacenski, 2025-04-17:
`0,0,0,1` means a zero degree yaw setting, that is a meaningful orientation. You should set your goal checker plugin to have a yaw tolerance of 2PI if you want to ignore the orientation of the path's goal pose and return success at any heading, so that once it meets the XY Tolerance, the Yaw Tolerance is always met.

MeghaMaheshwari, 2025-04-17:
this is my configuration 

```
controller_server:
  ros__parameters:
    use_sim_time: True
    controller_frequency: 20.0
    min_x_velocity_threshold: 0.001
    min_y_velocity_threshold: 0.001
    min_theta_velocity_threshold: 0.001
    failure_tolerance: 0.3
    progress_checker_plugin: "progress_checker"
    goal_checker_plugins: ["general_goal_checker"]
    controller_plugins: ["FollowPath"]
    # Progress checker parameters
    progress_checker:
      plugin: "nav2_controller::SimpleProgressChecker"
      required_movement_radius: 0.5
      movement_time_allowance: 10.0
    general_goal_checker:
      plugin: "nav2_controller::SimpleGoalChecker"
      xy_goal_tolerance: 0.25
      yaw_goal_tolerance: 0.25
```

SteveMacenski, 2025-04-17:
`yaw_goal_tolerance` your yaw tolerance is set to 0.25 radians (~15 deg). Set this to 2PI if you want this to always be met

MeghaMaheshwari, 2025-04-17:
ok let me try that

MeghaMaheshwari, 2025-04-17:
Follow up: I am assuming setting it to π (3.14) is sufficient because a difference of π radians in either direction covers all possible orientations (since +π and -π represent the same orientation) or does this explicitly need to be 2PI

MeghaMaheshwari, 2025-04-17:
setting to pi works
yaw_goal_tolerance: 3.14159265359

Thanks a lot for the quick support

closed 2025-04-18 by SteveMacenski