ros-navigation/navigation2 #5094: Use of stateful goal checker in Rotation Shim Controller

Ayush1285, 2025-04-18:
## Feature request

#### Feature description
Problem:
The rotation shim controller uses normal tolerance-based goal checking for the x-y goal. Once the tolerance is matched, it starts the rotation cycle.
Code: https://github.com/ros-navigation/navigation2/blob/5423eba732ed16b4f97a7353027c1f444b6edc60/nav2_rotation_shim_controller/include/nav2_rotation_shim_controller/tools/utils.hpp#L33C1-L33C41
The problem we are facing is that due to localisation shifts, it goes back into primary controller mode if the x-y shift is outside the tolerance, causing back-and-forth Rotation and translation switches. 

#### Implementation considerations

Solution:
Since goal_checker already implements the stateful feature, and goal_checker pointer is available in the `withinPositionGoalTolerance`,  we can use `isGoalReached` function of the goal checker instead of comparing ourselves. `isGoalReached` will take care of the stateful feature. Ref:   https://github.com/ros-navigation/navigation2/blob/5423eba732ed16b4f97a7353027c1f444b6edc60/nav2_controller/plugins/simple_goal_checker.cpp#L109

SteveMacenski, 2025-04-18:
Please submit a PR and I can take a look! It may be that this was implemented in RPP before it was available in the Goal Checker API so it was strapped in place and we missed a potential usecase spot. 

Note that `isGoalReached` looks at the orientation as well in that implementation, not just the translation. The RPP use of `withinPositionGoalTolerance` checks just the XY to know if we want to enact a rotation, so we need to only look at XY (and also have a stateful setting to fix the problem you mention, either through the GoalChecker use or our own implementation within RPP if need be).

SteveMacenski, 2025-05-03:
@Ayush1285 any update?

closed 2025-05-16 by SteveMacenski