ros-navigation/navigation2 #5095: base_link vs base_footprint

dan9thsense, 2025-04-19:
nav2.yaml seems inconsistent in the use of base_link vs base_footprint. The urdf file shows base_footprint as a child of base_link.

for base_frame_id:
amcl, collision_monitor, loopback_simulator use base_footprint

for robot_base_frame: 
bt_navigator, local_costmap, global_costmap, behavior_server use base_link

for base_frame:
docking_server uses base_link

But it seems that all of these should use the same link. In particular, shouldn't the costmaps be using base_footprint? 

Am I missing something here?

dan9thsense, 2025-04-22:
Sounds good to me. What about the question of whether base_footprint is a parent (as in turtlebot3 and most of ROS1) or a child (as in turtlebot4 and a lot of ROS2) of base_link?

SteveMacenski, 2025-04-23:
This answer is correct: https://robotics.stackexchange.com/questions/87014/base-link-vs-base-footprint

As to which should be the parent, that's is a good question and I suppose a matter of philosophy. I personally use base_link as be the root parent and base_footprint should be a child of it, but there's legitimate room for disagreement there depending on how you think about it.

dan9thsense, 2025-04-23:
Right, but it would be good to pick one because most of the nav2 docs have odom both published and broadcast with base_link while other docs have it going to base_footprint. If base_footprint is a child of base_link, it could end up with two parents in the tf tree

SteveMacenski, 2025-04-23:
>  If base_footprint is a child of base_link, it could end up with two parents in the tf tree.

How do? If its like below, then there are no double parents. 

```
base_link ---> ... robot frames
               \---> base_footprint
```

> since base_footprint should be flat on the ground with no pitch or roll with respect to odom, it makes more sense to me for it to be the parent of base_link

That is one way to look at it.  The other way is that we want full 6DOF representations of the robot to generally speaking use, but then for some applications we want the ground projected version of it with Z,Pitch,Roll removed. Timing delays from the node publishing the flat-ground projected base footprint would impact the entire system's timing if base footprint was then globally used. 

I think its OK to leave them as inconsistent in terms of their TF trees in the URDF. That shouldn't impact homologating the parameter names and values for the purposes of `nav2_bringup`'s yaml file.

dan9thsense, 2025-04-24:
Fair point. I agree it's good to get base_footprint out of the way of the main tf tree chain. That's a good enough argument that I'm going to use base_footprint as a child of base_link moving forward. It's also nice to be setup the way Nav2 expects, with odom going to base_link.

closed 2025-04-24 by dan9thsense