ros-navigation/navigation2 #5096: Breaking Namespace Substition for Params on costmap on purpose?

techtasie, 2025-04-19:
## Bug report
If using the Costmap, it gets added with an additional  [sub-namespace](https://github.com/ros-navigation/navigation2/blob/5423eba732ed16b4f97a7353027c1f444b6edc60/nav2_costmap_2d/src/costmap_2d_ros.cpp#L85). This causes the ros2 parameter namespace substitution to not work properly: https://github.com/ros2/rcl/blob/cc72fc1769ce9e6557c0f01efde982b3212e77b0/rcl/src/rcl/expand_topic_name.c#L99C55-L99C59

I also learned about this functionality quite recently: [proposal-namespace-shorthand-support-in-ros-2-parameter-values](https://discourse.ros.org/t/proposal-namespace-shorthand-support-in-ros-2-parameter-values)

Is there any reason why the namespace of the Costmap can't just be the normal namespace, and the topics and services get a "~/" as a prefix? This would then, by the rcl, be resolved to `/<namespace>/<node_name>`, which is exactly the same as it currently is for the topics, but would allow using the {namespace} in a parameter file and then have the correct top-level namespace.

This would mean changes like this [example](https://github.com/ros-navigation/navigation2/blob/5423eba732ed16b4f97a7353027c1f444b6edc60/nav2_costmap_2d/src/costmap_2d_ros.cpp#L219):

```
footprint_pub_ = create_publisher<geometry_msgs::msg::PolygonStamped>(
    "published_footprint", rclcpp::SystemDefaultsQoS());
```
This does resolves currently to: `/<top_level_namespace>/<node_name>/published_footprint`
After removing the `add_namespaces` it would need to be changed to:
```
footprint_pub_ = create_publisher<geometry_msgs::msg::PolygonStamped>(
    "~/published_footprint", rclcpp::SystemDefaultsQoS());
```
Which would also be resolved to `/{namespace}/{node_name}/published_footprint`. From the outside it would look the same except that the ros2 node base services like `/<namespace>/<node_name>/<node_name>/get_parameter_types`  would be changed to `/<namespace>/<node_name>/get_parameter_types`

I am interested to hear if this is just an oversight or if there is no intention of supporting the `{namespace}` parameter substitution.

**Required Info:**

- Operating System:
  - 
- Computer: Fedora 42 on Lenovo Ideapad
  - 
- ROS2 Version: Jazzy
  - 
- Version or commit hash: current jazzy release
  - 
- DDS implementation:
  - 
  - zenoh

#### Steps to reproduce issue

Just run a local_costmap with: 
```
       obstacle_layer:
          scan_lidar:
            topic: /{namespace}/scan

```
then do a `ros2 node info /{namespace}/local_costmap/local_costmap`

#### Expected behavior
It should subscribe to `/{namespace}/scan`
#### Actual behavior
It will subscribe to the `/{namespace}/local_costmap/scan`.
#### Reproduction instructions

#### Additional information
I would be interested to implement this if there is interested on your side.

SteveMacenski, 2025-04-22:
This is a complicated topic but the namespace is important for the sub-node use within the planner and controller servers for more reasons than just the topics names. 

Unfortunately, this is a wontfix since alot of subtle behavior relies on this configuration. You can really easily just support namespaces by not providing an absolute topic, ie `scan` rather than `/{namespace}/scan`. There's no reason to add the `/{namespace}` when not using absolute topics (prepended `/`). The namespace will be automatically added.

closed 2025-04-22 by SteveMacenski

techtasie, 2025-04-23:
@SteveMacenski what is the intended way of handling namespaces tf frames?

We are currently overriding the frame configs like: `base_frame_id` for each namespace by providing an [override config](https://github.com/carologistics/robotino_navigation/blob/master/robotino_navigation/config/robotinobase1_mps_map_gen.yaml). Is there also a better way of doing it for frame id's?

techtasie, 2025-04-23:
@SteveMacenski maybe adding a namespace / multi-robot section to the [Configuration guide](https://docs.nav2.org/configuration/index.html#configuration-guide) would be useful. If I understood how to handle the frame_id's properly I would be interested in adding a section if this is wanted...

SteveMacenski, 2025-04-23:
If you look at Nav2's launch files, we remap `/tf` to `tf` for all nodes so that namespaces apply to the TF topic as well. 

The cloned and unique launch files show how it can be done :-)

techtasie, 2025-04-23:
@SteveMacenski Yeah, that is great if you want to have separate TF trees for different robots.

We are currently working on multi-agent navigation, where we namespace all topics except the TF, since we want to have one large TF tree with all the information from all the robots (there is something to be said about maybe filtering certain parts of the TF to sync for network load reasons, but currently we are not having issues on that side).

So we are currently namespacing TF frame names (except for the map, since the map is the same for all).

But I gather from your answer that there is currently no intent to support this kind of setup within the Nav2 parameters...

SteveMacenski, 2025-04-23:
Having separate topics is more practical than having them share a TF topic - as that would be the only topic globally shared between all robots and it would involve alot of N:N transport of TF data from each robot to each other robot that would not be very practical on hardware. Simulation would work fine. 

I'd suggest changing to using separate TF topics for each robot as you would for all other topics.

techtasie, 2025-04-23:
@SteveMacenski but with rmw_zenoh we can decide to only share the TF topic from each robot to our central planning server then this would be an N:1 syncing.

NOTE: We migrated from Fawkes (our old software stack that predates ros1) to ros2 therefore some of our design patterns could be influenced by the way that fawkes did it.

I nonetheless think it's quite reasonable to synchronize all robot TFs with a central agent, as it enables straightforward lookups—like calculating distances between robots—and supports tasks such as merging cost maps or laser scans centrally for analysis.

SteveMacenski, 2025-04-23:
We haven't designed Nav2 to be tied to any particular RMW vendor, so that design wouldn't be something we'd want for default Nav2 behavior to tie it to required behavior of Zenoh (let alone Zenoh was only released earlier this year). 

I'm not really understanding what this has to do with the namespacing topics from before though. The frame IDs used in TF aren't much related to the `scan` sensor topics. You can still do substitutions in the frame IDs, just knowing the pattern that Costmap2D uses.

techtasie, 2025-04-24:
@SteveMacenski it was related in the sense that you had a better way of handling topic namespaces, and I was wondering if you also have a better way of handling frame namespaces.

After consideration, we decided that the Nav2 approach of using separate TFs is probably better than our current approach.

Thank you for the nudge in the right direction.