ros-navigation/navigation2 #5098: Regulated Pure Pursuit (RPP) controller should consider allow_reversing / velocity sign change detection with fixed curvature lookahead calculation, too

moooeeeep, 2025-04-22:
## Bug report

**Required Info:**

- ROS2 Version:
  - humble and jazzy binaries (and higher)
- Version:
  - humble_main
  - iron
  - jazzy
  - main

#### Steps to reproduce issue

* Run a RPP controller with fixed curvature lookahead distance enabled,
* Provide a `REEDS_SHEPP` path with reversing maneuvers,
* Watch behavior.

#### Expected behavior

Controller is able to follow the path.

#### Actual behavior

Controller performs weird steering maneuvers when a velocity sign change is within the configured curvature lookahead distance.

#### Reproduction instructions

Parameters file to reproduce:
```
amcl:
  ros__parameters:
    alpha1: 0.2
    alpha2: 0.2
    alpha3: 0.2
    alpha4: 0.2
    alpha5: 0.2
    base_frame_id: "base_footprint"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: false
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: 100.0
    laser_min_range: -1.0
    laser_model_type: "likelihood_field"
    max_beams: 60
    max_particles: 2000
    min_particles: 500
    odom_frame_id: "odom"
    pf_err: 0.05
    pf_z: 0.99
    recovery_alpha_fast: 0.0
    recovery_alpha_slow: 0.0
    resample_interval: 1
    robot_model_type: "nav2_amcl::DifferentialMotionModel"
    save_pose_rate: 0.5
    sigma_hit: 0.2
    tf_broadcast: true
    transform_tolerance: 1.0
    update_min_a: 0.2
    update_min_d: 0.25
    z_hit: 0.5
    z_max: 0.05
    z_rand: 0.5
    z_short: 0.05
    scan_topic: scan

[... 365 lines cut ...]
      costmap_topic: "/local_costmap/costmap_raw"
      footprint_topic: "/local_costmap/published_footprint"
      transform_tolerance: 0.1
      projection_time: 5.0
      simulation_step: 0.1
      dock_collision_threshold: 0.3

loopback_simulator:
  ros__parameters:
    base_frame_id: "base_footprint"
    odom_frame_id: "odom"
    map_frame_id: "map"
    scan_frame_id: "base_scan"  # tb4_loopback_simulator.launch.py remaps to 'rplidar_link'
    update_duration: 0.02
```

Can run this, e.g., like this:

    ros2 launch nav2_bringup tb4_loopback_simulation.launch.py params_file:=nav2_params.yaml

#### Additional information

Or is this feature not meant to be used like this?

Does it even make sense, to use a different lookahead point for velocity and steering?

SteveMacenski, 2025-04-30:
I think I see what's happening. `findVelocitySignChange` finds the max lookahead point before a path discontinuity due to directional changes and sets `lookahead_dist` if there's one nearer than the lookahead distance otherwise found. 

When using the curvature fixed distance for the curvature regulation & `shouldRotateToPath`, we use `params_->curvature_lookahead_dist` and simply ignore the `lookahead_dist` that was thresholded by the path directionality change. 

> weird steering maneuvers 

What are these behaviors, can you show a video or describe them? It it something like its slowing down alot or stopping and rotating repeatedly?

It seems to be like this should be updated to be:

```
  if (params_->use_fixed_curvature_lookahead) {
    double curv_lookahead_dist = params_->curvature_lookahead_dist < dist_to_cusp ? dist_to_cusp : params_->curvature_lookahead_dist;
    auto curvature_lookahead_pose = getLookAheadPoint(
      curv_lookahead_dist,
      transformed_plan, params_->interpolate_curvature_after_goal);
    rotate_to_path_carrot_pose = curvature_lookahead_pose;
    regulation_curvature = calculateCurvature(curvature_lookahead_pose.pose.position);
    curvature_carrot_pub_->publish(createCarrotMsg(curvature_lookahead_pose));
  }
```

And the `dist_to_cusp` be moved to be in the method scope (set to `0` `if (!params_->allow_reversing)`. 

Can you test that out and let me know if that solves your problem?

doisyg, 2025-05-01:
> Separately, in looking at this, I noticed that `params_->interpolate_curvature_after_goal` is only used in the `getLookAheadPoint` for the curvature lookahead. Should we enable that for the normal case too [@doisyg](https://github.com/doisyg) ?

(I will probably need to refresh a bit my memorry for a complete answer)
Do you mean enabling interpolation for the `carrot_pose` computation ? If so, I don't think so, we need it to not overshoot the goal for computing `x_vel_sign` properly

moooeeeep, 2025-05-05:
> What are these behaviors, can you show a video or describe them?

Bad behavior (green dot is curvature lookahead point, black dot is "normal" lookahead point):

https://github.com/user-attachments/assets/2d1cdb1c-60c5-4f29-986e-8cbfeb389009

The curvature lookahead point is jumping to odd positions when it goes beyond the cusp dist, thus producing weird steering commands.

With these changes it seems to work (jazzy branch, although this could probably be implemented more elegantly):

```diff
diff --git a/nav2_regulated_pure_pursuit_controller/src/regulated_pure_pursuit_controller.cpp b/nav2_regulated_pure_pursuit_controller/src/regulated_pure_pursuit_controller.cpp
index edcd6db5..8567dd5a 100644
--- a/nav2_regulated_pure_pursuit_controller/src/regulated_pure_pursuit_controller.cpp
+++ b/nav2_regulated_pure_pursuit_controller/src/regulated_pure_pursuit_controller.cpp
@@ -187,6 +187,7 @@ geometry_msgs::msg::TwistStamped RegulatedPurePursuitController::computeVelocity
 
   // Find look ahead distance and point on path and publish
   double lookahead_dist = getLookAheadDistance(speed);
+  double curv_lookahead_dist = params_->curvature_lookahead_dist;
 
   // Check for reverse driving
   if (params_->allow_reversing) {
@@ -197,6 +198,9 @@ geometry_msgs::msg::TwistStamped RegulatedPurePursuitController::computeVelocity
     if (dist_to_cusp < lookahead_dist) {
       lookahead_dist = dist_to_cusp;
     }
+    if (dist_to_cusp < curv_lookahead_dist) {
+      curv_lookahead_dist = dist_to_cusp;
+    }
   }
 
   // Get the particular point on the path at the lookahead distance
@@ -211,7 +215,7 @@ geometry_msgs::msg::TwistStamped RegulatedPurePursuitController::computeVelocity
   double regulation_curvature = lookahead_curvature;
   if (params_->use_fixed_curvature_lookahead) {
     auto curvature_lookahead_pose = getLookAheadPoint(
-      params_->curvature_lookahead_dist,
+      curv_lookahead_dist,
       transformed_plan, params_->interpolate_curvature_after_goal);
     rotate_to_path_carrot_pose = curvature_lookahead_pose;
     regulation_curvature = calculateCurvature(curvature_lookahead_pose.pose.position);

```

SteveMacenski, 2025-05-07:
@moooeeeep that looks perfect, can you open a PR?

SteveMacenski, 2025-05-08:
Merged, thanks for the report and patch!

closed 2025-05-08 by SteveMacenski