ros-navigation/navigation2 #5103: ros-rolling-behaviortree-cpp package not found during rosdep install

gopidusanapudi, 2025-04-24:
Description: 
 I'm trying to build Nav2 using the Dockerfile or on a Rolling distro system, and rosdep fails when attempting to install the ros-rolling-behaviortree-cpp package. It gives the error:

E: Unable to locate package ros-rolling-behaviortree-cpp

This results in the following error during rosdep install:
apt: command [apt-get install -y -qq ros-rolling-behaviortree-cpp] failed

Expected Behavior:
 The dependency should be resolved successfully, either by installing the correct package from apt or by providing fallback instructions to install from source.

 Actual Behavior: 
 The build process fails due to an unresolved dependency via rosdep.

mini-1235, 2025-04-24:
I think this is related to https://github.com/BehaviorTree/BehaviorTree.CPP/issues/960

gopidusanapudi, 2025-04-25:
Thanks for the quick response!

Yes, here are the steps I followed that led to the error:

Reproduction Instructions
Opened the navigation2 workspace in VS Code using Dev Containers

Used the VS Code Command Palette (Ctrl+Shift+P) → Dev Containers: Rebuild and Reopen in Container
then i got the above error

SteveMacenski, 2025-04-25:
Closing since issue was resolved by https://github.com/ros-navigation/navigation2/pull/5104 (thanks @ruffsl !) 

Try rebuilding from scratch after removing the install/build spaces and resourcing `/opt/ros`. Our CI is turning over and so are my local builds so I assume your workspace is just in a funky state

closed 2025-04-25 by SteveMacenski