ros-navigation/navigation2 #5107: Questions regarding main branch

saudintantri, 2025-04-26:
Hi,

I am looking to use the main branch of this repository primarily because of the nice set of features that have been introduced.

I have a few questions:

1. How stable is the main branch at the moment? Is there a previous "release" on the main branch which I can use?
2. For using the main branch on humble, there is an open PR and an untouched branch named "humble_main" but that branch is frozen at a commit 3 weeks ago. Should a PR that compiles the main branch on humble be rebased to the current developments in the main branch? Are there any breaking changes?
3. Is the "humble_main" branch a one-time thing? Or will future commits to main also be available on the humble_main branch?
4. I am aware that features will be frozen for the kilted release of navigation2. When is this set to happen? Will this first release of kilted be the most "Stable" version of main?
5. When I run `ament_uncrustify` on the main branch on a rolling docker, I don't get any code divergences. However, when I run the same "main" branch on a humble docker, I get quite a few. Why is this? Has the code formatting changed from humble to rolling? Should a PR that compiles main on humble run ament_uncrustify on humble so that it's clean? 

Please let me know if I am unclear, thank you for your time.

coderwyvern, 2025-04-27:
Other than my questions, I would like to share that I have had progress on compiling the main branch on humble.

Firstly, my mate raised a PR here https://github.com/ros2/launch/pull/864 to backport `EqualsSusbstitution` and `NotEqualsSubstitutions`. We will need this merged for this PR to work.

https://github.com/ros-navigation/navigation2/pull/5108 needs to be merged first. This rebases the `humble_main` with the recent developments in `main` and runs the linting to fix some failing tests.

Then, the actual changes are built on top of that in https://github.com/ros-navigation/navigation2/pull/5112. I raised 2 PRs separately so that the actual changes are clearly visible on the second one once the first one is merged.

-> I have tested on the nav2 minimal tb4 sim and it seems to work fine.
-> No failing system tests other than the one reported https://github.com/ros-navigation/navigation2/issues/5111. Which also seems to fail on humble.
-> Will test on a real robot tomorrow.

SteveMacenski, 2025-04-28:
1. Generally speaking I consider it pretty stable functionality wise, but there's no promise of that. It is the main development branch where new work comes in and there are occasional regressions and there are frequent breaks in ABI/API as we're building newer and better systems. 
2. That work is not complete yet, that branch in the repository is where the PR will eventually land once completed
3. TBD. It won't be a 1:1 every commit situation since we will not be able to automatically backport main changes to that branch due to the major divergence needed to make the main branch work on humble in the first place being done in that PR. However, I will be very open to any community member(s) that want to help maintain the status of that branch. Maybe at the end of each month taking a couple of hours and backporting main's changes for the month into that humble compliant branch. Otherwise, I might make a call more on the ~quarterly basis for engineers with a vested interest to help me out. This will probably not receive much of my attention since Humble is so outdated at this point. 
4. TBD. I haven't maintained a very particular deadline but I usually do it around mid-May when the release candidate for the release is cut so that we can test our Nav2 candidate against it. We usually have some big features that we want to land around the release timeframe as well, so I'm willing to delay it a couple of weeks if we need more time to get those in before we release into a new distribution.
5. The uncrustified policy has changed over time.
6. You'd have to walk those updates back to work on Humble I suppose.

P.S. You may rather use Jazzy instead. Our `main` branch works on Jazzy as-is for the last couple of months.

SteveMacenski, 2025-04-28:
Let me know about your robot testing :-)

coderwyvern, 2025-04-29:
Thanks for you response. Will update you about the robot testing once that's done.

SteveMacenski, 2025-05-01:
I'll follow up in the PR. But otherwise I think this ticket is good to close? 

(will reopen if requested and there's more topics on the original subject; we have another ticket on the humble main bit)

closed 2025-05-01 by SteveMacenski