ros-navigation/navigation2 #5111: Failing test in nav2_collision_monitor

suchetanrs, 2025-04-27:
Hello,
I compiled the main branch on rolling. After a clean build, I ran `colcon test` and all the tests passed successfully. After this, I ran `colcon test` again and it seems like there were some failure cases in `nav2_collision_monitor`.

I am attaching a file with the output of `colcon test-result --verbose` below. Is this some sort of flaky test?

SteveMacenski, 2025-04-28:
Does it happen again? It might just be a flaky test. I don't see that in our CI as a failure that's happened anytime in the recent history.

suchetanrs, 2025-04-29:
Hello, yes it seems to happen again.

I used this PR #5112 and ran the tests on humble. It happens on that as well. 
I'll try to come up with a fix for this if there is one :)

padhupradheep, 2025-05-08:
Hey @suchetanrs any updates?

suchetanrs, 2025-05-09:
Hey @padhupradheep , I haven't found time to get back to this yet. I'll get to it this weekend and let you know in case of progress.

suchetanrs, 2025-05-12:
I tried multiple times but I was not able to recreate this again. I am not sure why it happened in the first place. 

Please feel free to close this ticket. I'll open a new one if I am able to recreate it again and have some test log files to assist.

closed 2025-05-12 by padhupradheep