ros-navigation/navigation2 #5113: The code does not compile in the devcontainer - uninitialized dock value

msadowski, 2025-04-28:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04 (but running a devcontainer)
- Computer:
  - Not relevant
- ROS2 Version:
  - Rolling through devcontainer 
- Version or commit hash:
  - 4b9b7110b1ddbe7c579bee58728bad1eb1e0836d
- DDS implementation:
  - N/A

#### Steps to reproduce issue
I just tried setting up a devcontainer on my end, but I'm getting the following compilation error:

```
--- stderr: opennav_docking                   
In file included from /opt/ros/rolling/include/rclcpp/rclcpp/copy_all_parameter_values.hpp:25,
                 from /opt/ros/rolling/include/rclcpp/rclcpp/rclcpp.hpp:171,
                 from /opt/overlay_ws/src/navigation2/nav2_docking/opennav_docking/include/opennav_docking/docking_server.hpp:25,
                 from /opt/overlay_ws/src/navigation2/nav2_docking/opennav_docking/src/docking_server.cpp:16:
In constructor ‘rclcpp::Parameter::Parameter(const std::string&, ValueTypeT) [with ValueTypeT = bool]’,
    inlined from ‘bool rclcpp_lifecycle::LifecycleNode::get_parameter(const std::string&, ParameterT&) const [with ParameterT = bool]’ at /opt/ros/rolling/include/rclcpp_lifecycle/rclcpp_lifecycle/lifecycle_node_impl.hpp:268:21,
    inlined from ‘virtual nav2_util::CallbackReturn opennav_docking::DockingServer::on_configure(const rclcpp_lifecycle::State&)’ at /opt/overlay_ws/src/navigation2/nav2_docking/opennav_docking/src/docking_server.cpp:65:22:
/opt/ros/rolling/include/rclcpp/rclcpp/parameter.hpp:70:21: error: ‘dock_backwards’ may be used uninitialized [-Werror=maybe-uninitialized]
   70 |   : Parameter(name, ParameterValue(value))
      |                     ^~~~~~~~~~~~~~~~~~~~~
/opt/overlay_ws/src/navigation2/nav2_docking/opennav_docking/src/docking_server.cpp: In member function ‘virtual nav2_util::CallbackReturn opennav_docking::DockingServer::on_configure(const rclcpp_lifecycle::State&)’:
/opt/overlay_ws/src/navigation2/nav2_docking/opennav_docking/src/docking_server.cpp:63:8: note: ‘dock_backwards’ was declared here
   63 |   bool dock_backwards;
      |        ^~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
gmake[2]: *** [CMakeFiles/opennav_docking_core.dir/build.make:76: CMakeFiles/opennav_docking_core.dir/src/docking_server.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:225: CMakeFiles/opennav_docking_core.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
``` 

This seems to be introduced in this PR: https://github.com/ros-navigation/navigation2/pull/5079. @ajtudela I think the value was not initialized for a reason but because it's not initialized, it results in the above compilation error, even though there is a try/catch to signal the value was deprecated. 

#### Expected behavior

The code in devcontainer compiles

#### Actual behavior

It does not compile

#### Reproduction instructions

Pull latest main, and rebuild the docker container without cache

#### Additional information

SteveMacenski, 2025-04-28:
Solved in #5114

closed 2025-04-28 by SteveMacenski

ajtudela, 2025-04-29:
Thanks for the catch. As a parameter, it could be initialized without any problem.

It's funny, I haven't got those compilation error on any of my pcs.

Thanks for the quick fix @SteveMacenski