ros-navigation/navigation2 #5115: Move AMCL (and other standalone packages) to submodules

coxep, 2025-04-30:
## Feature request

#### Feature description

I would like to be able to fork AMCL and stay abreast of upstream changes without having to pull in all of the nav2 stack. Submodules would be a good way to both continue to deploy AMCL (and other packages) as nav2 packages while allowing the community to track and contribute to individual packages.

#### Implementation considerations

SteveMacenski, 2025-04-30:
I don't disagree that having some breakout for AMCL would be nice, but I don't want to do that until there is a robust and tested alternative either in `ros-navigation` or a third party supported released solution. IMO Navigation and Localization are different tasks and it feels odd to me AMCL is in Nav2 to begin with. 

Are any of the changes to AMCL able to be contributed back? If so, that's a better solution so that you don't need to maintain a fork. 

>  thought saw a recent discussion that said you can't really use submodules and still release the packages into the ROS build farm

This is true, AMCL would need to have its own release runs made in that case separately, which increases the maintainer burden.

SteveMacenski, 2025-05-08:
Closing due to lack of response from the author and this seems like a `wontfix` situation

closed 2025-05-08 by SteveMacenski