ros-navigation/navigation2 #5117: Cannot prevent SMAC from planning long backwards patchs

karl-schulz, 2025-05-01:
**Required Info:**

- Operating System:
  - Ubuntu 24
- Computer:
  -  Ryzen 9 7940-HS
- ROS2 Version:
  - Jazzy
- Version or commit hash:
  - jazzy
- DDS implementation:
  - Cyclone
- Nav2 Package:
  - SMAC
- Have you asked this on Robotics Stack Exchange?
  - No

#### Tuning / Configuration Goal

I want to make the robot not reverse unless absolutely neccessary.
* The robot is in front of an obstacle? -> reverse
* But if the goal is more than 1-2m behind the robot and everything is in free space, it should turn towards the goal instead of driving backwards.

How i test this:
* I put the robot at the location depicted in the image. The navigation starts below, with the orientation of the poses pointing to the top right.
* I then call NavigateThroughPoses with 3 poses on the path.
* What happens: The robot drives backwards to the first of the poses.

This should not happen:

[image]

#### What I've Tried

I expected a very high `reverse_penalty` (I tested 200) to basically prevent backwards navigation. Am I missing something? Is the search space maybe too small, so that non-backwards expansions are not explored enough?

Otherwise I tested nothing (because of lack of ideas). The problem is reliably reproducible.

Relevant parameters:
```yaml

global_costmap:
  global_costmap:
    ros__parameters:
      global_frame: map
      map_topic: /no_global_map_topic
      always_send_full_costmap: True
      rolling_window: True # Required if we boot the robot far away from the mission location
      track_unknown_space: False
      publish_frequency: 5.0
      update_frequency: 10.0
      resolution: 0.05
      width: 50
      height: 50
      lethal_cost_threshold: 99
      robot_base_frame: base_footprint
      footprint: '[ [0.201, 0.411], [0.201, -0.411], [-0.981, -0.411], [-0.981, 0.411] ]'
      footprint_padding: 0.0
      use_maximum: True
      plugins: ... (removed for brevity)

planner_server:
  ros__parameters:
    expected_planner_frequency: 20.0
    planner_plugins: ["SMAC"]
    SMAC:
      plugin: "nav2_smac_planner::SmacPlannerHybrid"
      tolerance: 0.2 # tolerance for planning if unable to reach exact pose, in meters, for 2D node
      downsample_costmap: false # whether or not to downsample the map
      downsampling_factor: 1 # multiplier for the resolution of the costmap layer (e.g. 2 on a 5cm costmap would be 10cm)
      allow_unknown: true # allow traveling in unknown space
      max_iterations: 1000000 # (default 1.000.000) maximum total iterations to search for before failing (in case unreachable), set to -1 to disable
      max_on_approach_iterations: 1000 # maximum number of iterations to attempt to reach goal once in tolerance, 2D only
      max_planning_time: 1.5 # max time in s for planner to plan, smooth, and upsample. Will scale maximum smoothing and upsampling times based on remaining time after planning.
      motion_model_for_search: "REEDS_SHEPP" # 2D Moore, Von Neumann; Hybrid Dubin, Redds-Shepp; State Lattice set internally
      angle_quantization_bins: 64 # For Hybrid/Lattice nodes: Number of angle bins for search, must be 1 for 2D node (no angle search)
      analytic_expansion_ratio: 3.5 # For Hybrid/Lattice nodes: The ratio to attempt analytic expansions during search for final approach.
      analytic_expansion_max_length: 3.0 # For Hybrid/Lattice nodes: The maximum length of the analytic expansion to be considered valid to prevent unsafe shortcutting (in meters). This should be scaled with minimum turning radius and be no less than 4-5x the minimum radius
      minimum_turning_radius: 0.1 # For Hybrid/Lattice nodes: minimum turning radius in m of path / vehicle
      reverse_penalty: 5.0 # For Reeds-Shepp model: penalty to apply if motion is reversing, must be => 1
      change_penalty: 0.0 # For Hybrid/Lattice nodes: penalty to apply if motion is changing directions, must be >= 0
      non_straight_penalty: 3.0 # For Hybrid/Lattice nodes: penalty to apply if motion is non-straight, must be => 1
      retrospective_penalty: 0.015
      cost_penalty: 5.5 # For Hybrid/Lattice nodes: penalty to apply to higher cost areas when adding into the obstacle map dynamic programming distance expansion heuristic. This drives the robot more towards the center of passages. A value between 1.3 - 3.5 is reasonable.
      lookup_table_size: 10.0 # For Hybrid/Lattice nodes: Size of the dubin/reeds-sheep distance window to cache, in meters.
      cache_obstacle_heuristic: False # For Hybrid/Lattice nodes: Cache the obstacle map dynamic programming distance expansion heuristic between subsequent replannings of the same goal location. Dramatically speeds up replanning performance (40x) if costmap is largely static.
      debug_visualizations: false
      smooth_path: True
      smoother:
        max_iterations: 1000
        w_smooth: 0.3
        w_data: 0.2
        tolerance: 1.0e-10

```

SteveMacenski, 2025-05-02:
> But if the goal is more than 1-2m behind the robot and everything is in free space, it should turn towards the goal instead of driving backwards.

`analytic_expansion_max_length` is the parameter you're looking for. It defines how long a possible analytic expansion is allowed to be to be considered viable. Since using the Reeds-Shepp model, it can reverse as part of the final approach process which bypasses the search cost for reversing `reverse_penality` (which I see you've already set as very high, so I don't think that's your issue).

If you want 1-2m, you can set that to 1-2m and it will not accept any analytic expansion larger than that number - which could include some reversing. As a side note: the shorter you make this, the more effort goes into search before it can 'shortcut' to find your exact solution to the goal. Setting this to 1-2m is probably fine and you won't see much impact on your planning times, I don't believe, but beware setting it _too_ low.

SteveMacenski, 2025-05-08:
@karl-schulz any update?

karl-schulz, 2025-05-14:
Hi Steve,

Thank you for the elaborate answer. In a short test, this definitely resolved the issue. I think we can close this issue and reopen in case our experiments show that this is not the solution.

Thanks again!

closed 2025-05-14 by tonynajjar