ros-navigation/navigation2 #5132: Follow path fails to make progress

MeghaMaheshwari, 2025-05-06:
I have a case where follow path enters a reverse V shaped area and the robot is standing at the edge of one of the two tips as shown below (as shown in the figure below, the dash line is the obstacle, consider it like a hallway)

---------------------------------------------------------------
       (obstacle)

             *
    
*(robot)              *

The robot drives towards the tip of V  and than follow path terminates saying

[controller_server-8][controller_server]: Failed to make progress
[controller_server-8] [controller_server]: [follow_path] [ActionServer] Aborting handle.

There is sufficient space between the robot and the obstacle to turn and go to the other end of the V. Not sure why follow path fails to make progress.

this is my configuration for follow path

`controller_plugins: ["FollowPath"]
    # Progress checker parameters
    progress_checker:
      plugin: "nav2_controller::SimpleProgressChecker"
      required_movement_radius: 0.5
      movement_time_allowance: 10.0
    general_goal_checker:
      plugin: "nav2_controller::SimpleGoalChecker"
      xy_goal_tolerance: 0.25
      yaw_goal_tolerance: 3.141
    FollowPath:
      plugin: "dwb_core::DWBLocalPlanner"
      debug_trajectory_details: True
      min_vel_x: -0.2
      min_vel_y: -0.1
      max_vel_x: 0.5
      max_vel_y: 0.1
      max_vel_theta: 1.0
      min_speed_xy: 0.0
      max_speed_xy: 0.2
      min_speed_theta: 0.0
      acc_lim_x: 2.5
      acc_lim_y: 2.5
      acc_lim_theta: 3.2
      decel_lim_x: -2.5
      decel_lim_y: -2.5
      decel_lim_theta: -3.2
      vx_samples: 20
      vy_samples: 20
      vtheta_samples: 20
      sim_time: 1.7
      linear_granularity: 0.05
      angular_granularity: 0.025
      transform_tolerance: 0.2
      xy_goal_tolerance: 0.1
      trans_stopped_velocity: 0.25
      short_circuit_trajectory_evaluation: True
      stateful: True
      critics: [
        "RotateToGoal", 
        "Oscillation", 
        "BaseObstacle", 
        "GoalAlign", 
        "PathAlign", 
        "PathDist", 
        "GoalDist"
        ]
      BaseObstacle.scale: 20.0
      PathAlign.scale: 32.0
      PathAlign.forward_point_distance: 0.4
      GoalAlign.scale: 20.0
      GoalAlign.forward_point_distance: 0.4
      PathDist.scale: 20.0
      GoalDist.scale: 20.0
      RotateToGoal.scale: 20.0
      RotateToGoal.slowing_factor: 5.0
      RotateToGoal.lookahead_time: -1.0 `

There if 

## Bug report

**Required Info:**

- Operating System:
  - 
- Computer:
  - 
- ROS2 Version:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

#### Reproduction instructions

#### Additional information

SteveMacenski, 2025-05-06:
Please ask on Robotics stack exchange, we do not offer configuration advice in the issue tracker

closed 2025-05-06 by SteveMacenski

MeghaMaheshwari, 2025-05-06:
Do you think the issue is mostly related to configuration?

SteveMacenski, 2025-05-06:
Most likely, yes!