ros-navigation/navigation2 #5133: [Smac] State lattice planner: angle flips with allow_reverse_expansion enabled

slazarev8, 2025-05-06:
## Bug report

**Required Info:**

Operating System:
Ubuntu 22.04
ROS2 Version:
Humble
Version or commit hash:
Humble
DDS implementation:
FastDDS
Nav2 Package:
Smac

Hello! I've noticed an issue when using the SMAC lattice planner with a diff robot. In most cases, everything works fine, but in dificult scenarios that require maneuvering (for example, when the robot is near a wall, and we need to turn around), the final plan's orientation may be flipped 180 degrees from the requested orientation.

After debugging and examining the SMAC source code, I believe the root of the problem may be in these locations:

1. [NodeLattice::GetIndex](https://github.com/ros-navigation/navigation2/blob/a291975e15ff28752cd166aad0234ef910ba2fc1/nav2_smac_planner/include/nav2_smac_planner/node_lattice.hpp#L282) does not take the `_backwards` flag into account
2. [NodeLattice::getNeighbors](https://github.com/ros-navigation/navigation2/blob/a291975e15ff28752cd166aad0234ef910ba2fc1/nav2_smac_planner/src/node_lattice.cpp#L557) can mark the goal node as backward

As a result, when the desired trajectory is not simple and the heuristic is not perfect for this particular case, the flipped trajectory can be selected during A* path calculation. I've observed that both the `isGoal` condition and the `_best_heuristic_node.first < getToleranceHeuristic()` check can result in these incorrectly flipped plans being chosen.

Also, If I disable allow_reverse_expansion flag, the problem does not reproduce. 

Some extra logs from planner I've added: 
The requested goal: 
```
[planner_server-11] [INFO] [1746567908.368662407] [planner_server]: goal th: 0.016815
[planner_server-11] [INFO] [1746567908.373920331] [planner_server]: goal is back: 0
[planner_server-11] [INFO] [1746567908.369878992] [planner_server]: Index goal: 5730272  //Index of the goal not at the SetGoal method
```
goal node that A* found and path backtrace: 
```
[planner_server-11] [INFO] [1746567908.405222967] [planner_server]: A* found goal: 1
[planner_server-11] [INFO] [1746567908.405227768] [planner_server]: th goal: 0.000000
[planner_server-11] [INFO] [1746567908.405231161] [planner_server]: th node: 16.000000
[planner_server-11] [INFO] [1746567908.405234835] [planner_server]: BackTracePath
[planner_server-11] [INFO] [1746567908.405237915] [planner_server]: Curr node ind = 5730272 //Index of the node A* take as final, the same as in Goal set, but angle is 180 deg. rotated
[planner_server-11] [INFO] [1746567908.405242513] [planner_server]: curr node is back: 1 //The last(goal) node is backward
...
[planner_server-11] [INFO] [1746567908.405644396] [planner_server]: curr path point.x 294.000000
[planner_server-11] [INFO] [1746567908.405648013] [planner_server]: curr path point.y 300.000000
[planner_server-11] [INFO] [1746567908.405652532] [planner_server]: curr path point.th 3.141593
[planner_server-11] [INFO] [1746567908.405656570] [planner_server]: curr path point.x 295.000000
[planner_server-11] [INFO] [1746567908.405660716] [planner_server]: curr path point.y 300.000000
[planner_server-11] [INFO] [1746567908.405665022] [planner_server]: curr path point.th 3.141593
[planner_server-11] [INFO] [1746567908.405668315] [planner_server]: curr path point.x 296.000000
[planner_server-11] [INFO] [1746567908.405671892] [planner_server]: curr path point.y 300.000000
[planner_server-11] [INFO] [1746567908.405675999] [planner_server]: curr path point.th 3.141593
```
The resulting plan:
```

[planner_server-11] [INFO] [1746567908.406408422] [planner_server]: Pose #0: Position(x=-4.674999, y=-5.024999, z=0.000000) Orientation(yaw=-3.141593)
[planner_server-11] [INFO] [1746567908.406423291] [planner_server]: Pose #1: Position(x=-4.724808, y=-5.026018, z=0.000000) Orientation(yaw=-3.112840)
...
[planner_server-11] [INFO] [1746567908.406652179] [planner_server]: Pose #27: Position(x=-5.974999, y=-5.124999, z=0.000000) Orientation(yaw=-2.896614)
[planner_server-11] [INFO] [1746567908.406660567] [planner_server]: Pose #28: Position(x=-5.974999, y=-5.124999, z=0.000000) Orientation(yaw=-3.141593)

```

#### Steps to reproduce issue
1. Locate a robot near the wall 
2. Send a request to a planner to move forward with 180 degree rotated goal pose 
3. Check the path thetas 

#### Expected behavior

At least the last pose has angle equals to the requested angle

#### Actual behavior

The last pose has a 180 degree rotated angle. 

#### Reproduction instructions
```
planner_server:
  ros__parameters:
    expected_planner_frequency: 5.0
    planner_plugins: ["GridBased"]
    smoother_plugins: ["SimpleSmoother"]
    global_frame: map
    robot_base_frame: base_link
    GridBased:
      robot_base_frame: base_link
      plugin: "nav2_smac_planner/SmacPlannerLattice"
      tolerance: 0.005
      downsample_costmap: false
      lattice_filepath: "my_lattice.json"
      downsampling_factor: 1
      allow_unknown: true
      max_iterations: 1000000
      max_on_approach_iterations: 10000
      max_planning_time: 3.0
      non_straight_penalty: 1.5
      cost_penalty: 2.0
      rotation_penalty: 5.0
      reverse_penalty: 2.5
      allow_primitive_interpolation: true
      change_penalty: 0.3
      allow_reverse_expansion: true
      retrospective_penalty: 0.05
      analytic_expansion_ratio: 1.5
      analytic_expansion_max_length: 7.0
      cache_obstacle_heuristic: true
      smooth_path: true
      smoother:
        smoother:
          max_iterations: 1000
          w_smooth: 1.5
          w_data: 2.2
          tolerance: 0.1
          do_refinement: false
          refinement_num: 2
```
The red arrow show the requested pose, but the plan's orientation is flipped here. 
[image]

#### Additional information

slazarev8, 2025-05-07:
Trying to fix the issue, I'v added a small work-around: 
1. Added a bool **is_goal_** to the lattice node, which is false by default and true only for goal node 
2. In NodeLattice::getNeighbors I skip all backward nodes before goal 

[image]
Not sure this is a good approach, but it fixes the issue for me, but need more tests.

slazarev8, 2025-05-07:
Sure it's not a good approach, but I found a work-around:
1. I'v added is_goal_ flag to the lattice node, which is false by default and true only for goal node 
2. Inside NodeLattice::getNeighbors I skip all backward goal nodes: 

[image]
More testing needs to be done, but it seems to work.

SteveMacenski, 2025-05-07:
> the final plan's orientation may be flipped 180 degrees from the requested orientation.

Huh, interesting! Does this happen when you set the tolerance to `0.0`? I see you have `0.005` which gives the solution some room to pick something near-by but not exact. I think what you're pointing out is a real bug, but just sanity checking its not something more superficial 😄 

Otherwise - fascinating. I can see how this happens but also how exceedingly rare it would be given the analytic expansion should typically be the thing finding the exact goal pose. 

Would it be better to change the implemenation of `isGoal` in A* to consider the backwards state? Similar with the heuristic calculation for the closest H value. 

Does this also impact the Hybrid-A* planner's reversing logic?

slazarev8, 2025-05-08:
> Does this happen when you set the tolerance to `0.0`

Zero tolerance does not help, I've just checked. And as I mentioned, A* completes calculation with exact check IsGoal(), so that's not the case. 

> but also how exceedingly rare it would be given the analytic expansion should typically be the thing finding the exact goal pose.

You are right, I also noticed that if I disable analytic expansion (just as experiments), the issue happens much more frequently. But as I see, it often fails if I want to plan a rotation movements in the narrow corridors. Or like in a case from my attached screenshot - a movement with 180 deg. rotation near the wall, this case is 100% reproducible even with enabled expansion. 

About my fix: I see at the path backtrace that near goal segments are always build by analytics extension, so it covers up the flaw of my fix. And after testing in simulation, I didn't find any problems.. But yes, if I disable analytics extension, the planners fails to build the path in my corner-case scenario. But I find this behaviour is better than 180 flipping, and any way, analytics extension should be enabled. 

> Would it be better to change the implemenation of `isGoal` in A* to consider the backwards state?
It looks like more correct way to fix an issue, and it was my first attempt, but now LaticeNode == operator is just index comparing, and a* checks just equality of the nodes. I had tried to change Index calculation method for the LaticeNode and take _backward flag into account, but It did not help, I think it broke some other things =(  Maybe I can try one more time later.

> Does this also impact the Hybrid-A* planner's reversing logic?

I don't have configured Hybrid, the default version failed to build a plan in my corner case, but I can't see backwards flag in the Hybrid node, so I think it's just lattice planner issue.

SteveMacenski, 2025-05-09:
> LaticeNode == operator is just index comparing, and a* checks just equality of the nodes

We can change that comparison to be a bit more than just `node == goal`. We can add in a `backwards` check as well to check that they're the same (hint: the goal is never backwards).

 > I had tried to change Index calculation method for the LaticeNode and take _backward flag into account

I think that would also work! I think also `getCoords` may also need to be updated as well as `getIndex`? You bring up a good point here in that maybe we should want to update these as well to get the 'right' coordinates / index for heuristic and other computations. Its possible that we're doing incorrect collision checking in the Node and analytic expansions orientations (the primitives are correct, since [they take into account backwards](https://github.com/ros-navigation/navigation2/blob/a291975e15ff28752cd166aad0234ef910ba2fc1/nav2_smac_planner/src/node_lattice.cpp#L271-L272), but [the end pose doesn't!](https://github.com/ros-navigation/navigation2/blob/a291975e15ff28752cd166aad0234ef910ba2fc1/nav2_smac_planner/src/node_lattice.cpp#L236-L237)), graph populations in the `NodeGetter`, and heuristic computations (as you pointed out) in this case.

slazarev8, 2025-05-09:
> I **think** considering the backwards-ness in `motion_projection.theta` would resolve this issue since it would get keyed to the right index. We'd just have to remove the [backward-ness logic here and move it up](https://github.com/ros-navigation/navigation2/blob/a291975e15ff28752cd166aad0234ef910ba2fc1/nav2_smac_planner/src/node_lattice.cpp#L535-L541). Then, the `setPose` angle is always just the newly-fixed `motion_projection.theta`

I think this is a good idea, I tested it a little and everything seems good, no 180 degree error reproduced. But there is one thing I can't realize: in theory, this change should improve A* convergence, because it fixes heuristic cost calculation for backward nodes, but I've provided some perf test and don't see any difference comparing with my dirty fix. Or mb the difference is tiny and that's why I don't see it?

SteveMacenski, 2025-05-09:
Can you open a PR with your batch? We  can get this in ASAP.

I wouldn't expect that big of a difference in general cases, but I think I could design a few edge cases where it would be pronounced. Since the search is generally matching the robot in the forward direction since `reverse_penalty > 0`, the reverse expansions usually aren't going towards the goal, but away from the goal, regardless of the orientation the index implies. Its still spatially behind the current node being expanded and that'll have a worse heuristic value spatially regardless. so when we continue to expand, the forward ones continue to be preferred.

slazarev8, 2025-05-09:
> Can you open a PR with your batch? We can get this in ASAP.
>
> I wouldn't expect that big of a difference in general cases, but I think I could design a few edge cases where it would be pronounced. [...]

Sure, just opened PR. 
Could you clarify, pls, to which branch should I merge? I've opened PR to main, but I think we(at least me =) ) need this fix for humble, so I need to open 2 PRs or how to deliver it?

SteveMacenski, 2025-05-09:
@slazarev8 yes, please also open against humble and I can merge. I can autobackport to Jazzy but humble's so out of date that won't be possible.

Did you test this and happy with it as the final solution?

closed 2025-05-10 by SteveMacenski

slazarev8, 2025-05-10:
> Did you test this and happy with it as the final solution?
Sure, I see no problems with it, thx for advice!

I've opened the same PR for humble_main, but there are some code style divergence fails for files I did't change, not sure that the problem is on my side.. https://github.com/ros-navigation/navigation2/pull/5143

SteveMacenski, 2025-05-10:
Fyi: that’s not the humble distribution branch. ‘humble’ is - that’s another branch for compatability of main using Humble but is still a work in progress so you shouldn’t open the PR against that