ros-navigation/navigation2 #5136: The maximum speed of Nav2 is limited to 0.5m/s

lymcs, 2025-05-08:
I used Nav2 for navigation and found that by adjusting the maximum speed with the DWB controller, when the speed is less than 0.5m/s, the speed of the car can match the parameter settings. When the max-vel-x is set to be greater than 0.5m/s, the navigation speed of the car can only reach 0.5m/s. I feel that there is a place that limits the maximum navigation speed, but I cannot find it, so I am seeking help. Thank you very much

padhupradheep, 2025-05-08:
Hey, this sounds to be an issue with the configs of your vehicle. You need to figure out where you can increase the speed limit of the vehicle. 

This is a bug and feature tracker, please use robotics stackexchange for better help from the community. Also make sure, which distribution you are using, and what vehicle, github repo, whichever info that could help the community to answer! 

Thanks!

closed 2025-05-08 by padhupradheep

lymcs, 2025-05-09:
> 嘿,这听起来是你车辆配置的问题。你需要弄清楚在哪里可以提高车辆的限速。
> 
> 这是一个错误和功能跟踪器,请使用机器人堆栈交换,以获得社区的更好帮助。还请确保,您正在使用哪个发行版,以及什么车辆,github repo,无论哪些信息可以帮助社区回答!
> 
> 谢谢!

Hello, thank you very much for your reply. After repeated testing, it is not a problem with the settings of my vehicle, nor is it due to simulation. Later, I used the official Nave2 Burger and Waffie for simulation, and changed the maximum linear speed in the DWB controller to IE1.0. However, it is confusing that the navigation speed of the car is still at a maximum of 0.5m/s and will not increase above 0.5. I always feel that it is affected by which speed limit in Nava2. Can you tell me specifically where in the code the maximum speed limit for car navigation is set?