ros-navigation/navigation2 #5145: Directional planning constraints as masks

muhidabid, 2025-05-11:
## Feature implementation help

#### Feature description
I am working on implementing **directional obstacles** or **one-way zones** using Nav2, where a robot can move through a defined area only in a specific direction (e.g., east at 90 degrees but not in any other direction). This can be useful in environments with conveyor belts, narrow one-way hallways,  one-way doors, or traffic-style constraints.

Although similar concepts like **SpeedFilter**, **KeepoutFilter**, and **Inverse Keepout** exist, I did not find a direct solution for enforcing *direction-based* motion constraints during **global path planning**. 

According to my understanding, this cannot be achieved by simply creating a costmap filter since filters simply add to the master grid. To make the planner aware of directional constraints in a .pgm mask, I will have to create a planner plugin.

#### Old Related Issues
  - The idea of a `DirectedLanesFilter` was discussed in [issue #1263](https://github.com/ros-planning/navigation2/issues/1263), where direction encoding via RB-colorspace was proposed.
  - Later, the concept of "preferred lanes" was implemented differently using inverse keepout in [issue #1522](https://github.com/ros-planning/navigation2/issues/1522).
  - Another related request was in [issue #3395](https://github.com/ros-planning/navigation2/issues/3395) about “one-direction-lane” costmaps. It was noted that directionality can be handled using [Route Graphs](https://docs.nav2.org/tutorials/docs/route_graph_generation.html#route-graph-generation) and that a directional mask is possible and could be a good contribution.

---
What would be the best approach for implementing this? Any suggestions and guidance would be greatly appreciated.

BCKSELFDRIVEWORLD, 2025-05-12:
I believe this is a very important feature. Instead of completely blocking an area, we can treat it as a preferred direction zone

The idea is to define a specific corridor and assign directional costs based on how the robot moves between neighboring cells. Since A* uses Moore neighborhood (8-connected grid), each movement has an implicit direction (east, northeast, north).

In these corridors, we can create a costmap plugin where each cell holds a preferred heading (e.g., east = 0°, west = 180°). When the planner evaluates movement from one cell to another, it can compute an angular difference between the movement direction and the preferred direction, and assign a directional_cost accordingly.

This cost can be added to the existing A* cost formula. While A* typically computes:

total_cost = A * traversal_cost + B * heuristic_cost

We can extend it to:

total_cost = A * traversal_cost + B * heuristic_cost + C * directional_cost

Where directional_cost is calculated based on the heading difference. This will guide the robot to follow preferred directions in specific corridors, but still allow passage in the reverse direction when necessary — just with a higher cost.

If the weights are well-tuned, this could become a very powerful tool for structured navigation, especially in narrow spaces or multi-robot environments where directional flow control improves safety and efficiency.

Maybe like this: (idk)

[image]

SteveMacenski, 2025-05-12:
How would that be communicated to the planners / controllers for use? The planners only know about the master costmap that is the combination of all of the other layers, including filters, into a single cost. We currently do not have the APIs for transition costs from A->B in seach, only the absolute cost of B (which may have been expanded from A).

That would involve a major API change, either by having filters be accessible directly by the planner/control algorithms to make these kinds of transition queries to use in its search behavior or by changing the `getCosts` APIs from using a cell to check to the `from_cell` and `to_cell`. 

I actually think this is a better API than what we do now and I've heard from other companies that they do similar things when we've chatted about API updates to replace Costmap2D with another solution. But, it is a major change for costmap2D that's been around for a long time. 

I also wonder how noisy just checking the orientation would be. A moore neighborhood is only 8 orientations (and also not even evenly distributed every 45 degrees). If we score based on distance from a target orientation, then the path would have zig-zags introduced. A better solution might be to only penalize orientations that are outside of a reasonable bound from the traversal direction (i.e. > 90 degrees off). 

---

@muhidabid another thing to mention is now there is a Route Server in Nav2 which could be easily used to represent directionality and lanes using the edges. The edges are naturally directional which would only let you go down a certain direction during global search only (and others can be set as bidirectional if you choose, even with different costs for each way if you want asymmetry). Depending on your applications, that may be a better choice and was designed to fill a large amount of the gap that a freespace planner with directionality knowledge would. It just requires that you have a structured-ish space that you can naturally represent as a graph (logistics, warehouse, grocery store, etc). Which, it sounds like for your needs and examples would fit well.

muhidabid, 2025-05-21:
I want to clarify that I'm currently exploring this problem due to a need that arose during my internship. In that context, directionality constraints were already being handled using the **Route Server**, which indeed worked well to a certain extent. However, the use of Route Server was also somewhat limiting for more nuanced spatial and directional constraints in unstructured or semi-structured environments - hence the interest in developing a more flexible solution that could work with general navigation planners.

I'm considering implementing the directionality feature described earlier as a more general-purpose capability - possibly as an **open-source plugin** or, if it aligns with the project direction, contributing it back to this repository. 

Could you advise on:

* Whether this kind of contribution would make sense in the context of this repo?
* What would be a good starting point for structuring this as a plugin or upstream contribution?
* Any known pitfalls or architectural limitations I should keep in mind while exploring this path?

Thank you for your input @BCKSELFDRIVEWORLD @SteveMacenski .

SteveMacenski, 2025-05-21:
> Whether this kind of contribution would make sense in the context of this repo?

Sure!  The big questions would be:
- What kind of plugin? Costmap, Planner, etc. Perhaps a constraint plugin within a planner? 
- Which planners would we want to target? All, Smac, NavFn, etc. NavFn is a special case with potential fields that would be hard to edit without just editing costs in the cost grid (or transition functions for `getCost` as mentioned above). The others are more pure A*'s that we can apply weighted costs to their traversal cost function. Thinking about how we want the constraint applied will tell us what planners can or cannot easily support with this.  
- How are the directions represented and constraints enforced to do the directionality bits while not also creating poor behavior? Just doing N/E/S/W would make any spaces that are at 45 degree angles jagged or drunken. I think it should be able to represent 360 space. 

I think it might make sense to survey a bit of the ways this is implemented today in industry / research and summarize it here before jumping into a specific implementation off the bat. It seems likely folks have done and talked about things like this before that we can leverage their experience to know 'what works' and how.

> What would be a good starting point for structuring this as a plugin or upstream contribution?
> Any known pitfalls or architectural limitations I should keep in mind while exploring this path?

I think my comment above on a survey / trade study would be the best starting point, that's what I would do. Having enough information to see 'how its done' or provide some ideas to build off of will make it more sensible about how / where its implemented and how we can best integrate it into Nav2's algorithms.

BCKSELFDRIVEWORLD, 2025-05-21:
For A*-based planners, my idea is this: Even though Steve is not a big fan of adding new layers, I still want to start with a costmap layer (sorry Steve 😊). The user can draw polygons on the map and set a direction (for example, as an angle in degrees). We can save these direction values in a separate direction_map, just like how we use the normal costmap.

Then, in A*, when moving from one cell to another, we can check if the movement direction matches the allowed direction of that cell. If it does not match, we can add extra cost or even block the movement. As I tried to show above (with my bad prototype), we can use Moore neighborhood to compare the directions and set the cost. This gives us flexibility: we can block directions completely or just make them less preferred.

In my own tests, this worked with the Smac planner, even with a simple design.

It’s also important to avoid zigzag or strange paths at corners. For this, we can set a threshold or similarity range, so the robot does not zigzag at 45-degree angles or other edge cases.

**In summary:**

1. The user draws polygons and sets a direction.
2. We store direction values in a direction_map.
3. In A*, we change the traversal cost based on direction.
4. We use a threshold to prevent zigzag paths.

I think this way is both flexible and easy to integrate into planners.

One important thing is the user should tune the cost values well. It would help to give users some min/max value suggestions and a simple guide, so the robot doesn’t act strange.

Right now, my idea only supports one direction for layer, which can be a limitation. Also, with Nav2, it is not easy to set different directions for each polygon directly. A real solution would be to have a map editor where you can set direction for each polygon and send this information to the planner.

When I have time after work and school, I want to help with the [nav2_map_editor](https://github.com/TomasHricina/navigation2/tree/main/nav2_map_editor). If we can finish the editor, it will be much easier and more flexible to set different directions for each polygon and make the robot use these in planning.

SteveMacenski, 2025-05-21:
I want us to start from base principles before we map a solution space to a problem. We should know what the ways its possible to do this with first before deciding the mechanism within Nav2's architecture it should belong to. I don't disagree that's a strong potential candidate, but its important to weigh the options thoughtfully.

And for the record, I don't mind new costmap layers 😉 

> Then, in A*, when moving from one cell to another, we can check if the movement direction matches the allowed direction of that cell. 

This statement does alot of heavy lifting we would need to dissect:
* How would you get the directional information in the A* from the costmap API? 
* How would you query the costmap API with that transition function `A->B`? There's no API like that.  
* Do all of the Theta/A* implementations have the last and current cell information in the same scope in order to call that transition function where the costs are being computed?

All of these things seem possible, but there's design conversations to be had that can't be jumped to. Most of that information and/or APIs don't exist yet. 

> Right now, my idea only supports one direction for layer

Definitely, so thinking about requirements and design is important 😉 We should be able to represent it in a RGB image since that's enough information for a 2D vector and possibly a weight. 

> When I have time after work and school, I want to help with the [nav2_map_editor](https://github.com/TomasHricina/navigation2/tree/main/nav2_map_editor). If we can finish the editor, it will be much [...]

Sounds great! Before going into the directional editor mode, I think deciding how we need it formatted would be good :-)

SteveMacenski, 2025-09-26:
Some thoughts from an analysis for smac planner directional mask addition from another thread. Note that I don't think necessarily this is the way we should be handling directional constraints in general in Nav2, however this is a blueprint for strapping it onto the smac planner if need be 

--- 

We know at any given time the expansion that a neighbor came from since we have the neighborhood in the code. Therefore, we can map for any index in the neighborhood to the orientation that this expansion takes place. That buys us two things: (1) we do not need to track parents nor perform major surgery on the way this code functions and (2) we can just think about scoring for a pose, knowing the direction of movement. That would then still be combined in the same way as before so the deconfliction would just be from the actual new_cost > existing_cost logic for whether this expansion is considered "good enough" after combination to override a directional mask violation. That puts this into a nice structure that there's a tunable parameter for the weighting of the orientation masks w.r.t. distance to know if entering one, even slightly, is worthwhile for a shortcut -- or completely disabling it when set very high. 
 
What we'd want to do is keep this block of code [1], either add/multiply afterwards or within the formulation based on the index of the neighbor, which we know is i we know which mask to look up a cost in for use. I'd recommend doing a multiplicative cost travel_cost *= direction_mask[i]->at(x,y) * weight . You'd have to own the overhead of the directional mask lookups but I think that one is unavoidable. Expect this heuristic to nearly double in runtime since those lookups are actually the heaviest operations. 
 
[https://github.com/ros-navigation/navigation2/blob/main/nav2_smac_planner/src/node_hybrid.cpp#L671-…](https://github.com/ros-navigation/navigation2/blob/main/nav2_smac_planner/src/node_hybrid.cpp#L671-L679)
 
Then you can use the same multiplicative formulation in the traversal cost function so that should keep the Heuristic and Traversal cost functions aligned so the expansion doesn't blow up. 
 
I think this is the best way since it doesn't modify much code and doesn't add any more into memory than strictly required (the masks + subsequent lookups). This could be pluginized too or add in a virtual method  that can be overriden by an application to add something like that in. I'd recommend having a costmap layer that has these masks and manages them for generality, then in a function use the costmap to dynamic cast to find that layer and then query it for information