ros-navigation/navigation2 #5159: rotation values comparison in simple progress checker

gunasekaran-goatrobotics, 2025-05-14:
## Feature request

#### Feature description
- The current simple progress checker only checks the x and y values from the pose topic to compare the radius value
- Need rotation value comparison for the movement when the differential robot does the in-place rotation.
#### Implementation considerations
- Additional parameter in simple progress checker `yaw_tolerance`
-  So the robot rotation movement calculation is based on checking against time.
- This will help in an in-place robot rotation.

gunasekaran-goatrobotics, 2025-05-14:
@SteveMacenski, your thoughts on this?

SteveMacenski, 2025-05-15:
Wait, this already exists in the `PoseProgressChecker` - see here: https://github.com/ros-navigation/navigation2/blob/main/nav2_controller/plugins/pose_progress_checker.cpp 

That sound be your solution :-)

gunasekaran-goatrobotics, 2025-05-15:
Thanks! @SteveMacenski