ros-navigation/navigation2 #5161: Lifecycle nodes loaded as composable nodes in a container are unable to communicate with the lifecycle manager

GPrathap, 2025-05-15:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 22.04

- ROS2 Version:
  -Humble

I have several lifecycle nodes loaded as composable nodes in a container. Then, I use the lifecycle manager to bring up all the nodes

    from launch import LaunchDescription
    from launch.actions import DeclareLaunchArgument
    from launch.substitutions import LaunchConfiguration
    from launch_ros.actions import ComposableNodeContainer, Node
    from launch_ros.descriptions import ComposableNode

    def generate_launch_description():
        use_sim_time = LaunchConfiguration('use_sim_time')

        return LaunchDescription([
            DeclareLaunchArgument(
                'use_sim_time',
                default_value='false',
                description='Use simulation time if true'
            ),

            # Composable Node Container
            ComposableNodeContainer(
                name='sensor_container',
                namespace='',
                package='rclcpp_components',
                executable='component_container',
                output='screen',
                composable_node_descriptions=[
                    ComposableNode(
                        package='imu_preprocessor',
                        plugin='aide_imu_preprocessing::ImuPreprocessor',
                        name='imu_preprocessor_node',
                        parameters=[{'use_sim_time': use_sim_time}]
                    )
                ]
            ),

            # Lifecycle Manager to manage lifecycle state transitions
            Node(
                package='nav2_lifecycle_manager',
                executable='lifecycle_manager',
                name='lifecycle_manager_sensors',
                output='screen',
                parameters=[
                    {'autostart': True},
                    {'node_names': ['sensor_container/imu_preprocessor_node']},
                    {'bond_timeout': 10.0},
                    {'attempt_respawn_reconnection': True},
                    {'retry_on_failure': True}
                ]
            )
        ])

Those nodes start successfully, but due to namespace issues, the lifecycle manager is unable to retrieve the status of each lifecycle node

        [INFO] [launch]: All log files can be found below /home/op/.ros/log/2025-05-15-12-11-22-609782-op-Dell-G16-7630-907199
        [INFO] [launch]: Default logging verbosity is set to INFO
        [INFO] [component_container-1]: process started with pid [907210]
        [INFO] [lifecycle_manager-2]: process started with pid [907212]
        [lifecycle_manager-2] [INFO] [1747307482.668951759] [lifecycle_manager_sensors]: Creating
        [lifecycle_manager-2] [INFO] [1747307482.669757600] [lifecycle_manager_sensors]: Creating and initializing lifecycle service clients
        [component_container-1] [INFO] [1747307482.909456536] [sensor_container]: Load Library: /home/op/imu_ws/install/imu_preprocessor/lib/libimu_preprocessor_component.so
        [component_container-1] [INFO] [1747307482.915015197] [sensor_container]: Found class: rclcpp_components::NodeFactoryTemplate<aide_imu_preprocessing::ImuPreprocessor>
        [component_container-1] [INFO] [1747307482.915051099] [sensor_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<aide_imu_preprocessing::ImuPreprocessor>
        [component_container-1] [INFO] [1747307482.917613496] [imu_preprocessor_node]: Initializing IMU Preprocessor
        [INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/imu_preprocessor_node' in container '/sensor_container'
        [lifecycle_manager-2] [INFO] [1747307484.670338132] [lifecycle_manager_sensors]: Waiting for service sensor_container/imu_preprocessor_node/get_state...
        ^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
        ^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT) again, ignoring...
        [lifecycle_manager-2] [INFO] [1747307485.688357535] [rclcpp]: signal_handler(signum=2)
        [lifecycle_manager-2] [INFO] [1747307485.688621706] [lifecycle_manager_sensors]: Running Nav2 LifecycleManager rcl preshutdown (lifecycle_manager_sensors)
        [lifecycle_manager-2] [INFO] [1747307485.688835270] [lifecycle_manager_sensors]: Waiting for service sensor_container/imu_preprocessor_node/get_state...

When I check the ROS service list, the lifecycle node services are prefixed with their namespaces, but not with the container name sensor_container. As a result, the lifecycle manager is unable to communicate with the nodes. 

`ros2 service list`

`/imu_preprocessor_node/get_state`

Now sure how to resolve this?

SteveMacenski, 2025-05-15:
Checkout Nav2's bringup for examples, we do this all of the time in our default bringup, including with namespaces :-) 

https://github.com/ros-navigation/navigation2/blob/main/nav2_bringup/launch/navigation_launch.py

These are all pushed into a namespace https://github.com/ros-navigation/navigation2/blob/main/nav2_bringup/launch/bringup_launch.py#L147

GPrathap, 2025-05-16:
Hi @SteveMacenski , the issue was 

    void LifecycleNode::createBond()
    {
      if (bond_heartbeat_period > 0.0) {
        RCLCPP_INFO(get_logger(), "Creating bond (%s) to lifecycle manager.", this->get_name());

        bond_ = std::make_unique<bond::Bond>(
          std::string("bond"),
          this->get_name(),
          shared_from_this());

        bond_->setHeartbeatPeriod(bond_heartbeat_period);
        bond_->setHeartbeatTimeout(4.0);
        bond_->start();
      }
    }

I did not know we needed this bit, after adding this and starting to createBond() when activating node, all went fine :) 

Thank yoouu,
Geesara

SteveMacenski, 2025-05-16:
No problem 😉

closed 2025-05-16 by SteveMacenski