ros-navigation/navigation2 #5175: humble_main branch not compatible with ROS 2 Humble (e.g., nav2_util build errors)

MohammedMuthuFasil, 2025-05-19:
Required Info:

    Operating System:

        Ubuntu 22.04

    Computer:

        13th Gen Intel NUC (or your actual system, please replace if different)

    ROS2 Version:

        ROS 2 Humble (installed via binaries)

    Version or commit hash:

        humble_main branch cloned on 2025-05-19
        Commit: 

    DDS implementation:

        rmw_cyclonedds_cpp

Steps to reproduce issue

    Clone the Navigation2 repository and import dependencies:

mkdir -p ~/nav2_ws/src && cd ~/nav2_ws/src
git clone -b humble_main https://github.com/ros-planning/navigation2.git
vcs import < navigation2/tools/ros2_navigation.repos

Install dependencies:

sudo apt update
rosdep install --from-paths . --ignore-src -r -y

Build with colcon:

    cd ~/nav2_ws
    colcon build

Expected behavior

Navigation2 builds successfully on ROS 2 Humble without errors.
Actual behavior

Build fails at nav2_util or similar packages with errors such as:

error: ‘rcl_service_introspection_state_t’ was not declared in this scope
error: ‘RCL_SERVICE_INTROSPECTION_METADATA’ was not declared in this scope
error: cannot convert ‘rclcpp::SystemDefaultsQoS’ to ‘const rmw_qos_profile_t&’
error: ‘configure_introspection’ is not a member of ‘rclcpp::Client<...>’

These errors suggest that humble_main is using newer ROS 2 APIs not available in Humble.
Reproduction instructions

Follow the steps above on a fresh ROS 2 Humble install. Build fails consistently at the same point.
Additional information

This issue seems to be caused by usage of service introspection features introduced in later ROS 2 versions (e.g., Iron or Rolling).

    Can the maintainers confirm if humble_main is still expected to support the official Humble release?

    If not, it may help to update the documentation or README to reflect this, or provide a compatible tag or branch for Humble users.

Let me know if you want this exported as a .md file or if you’d like me to help submit it on GitHub.

SteveMacenski, 2025-05-19:
See #5112. This is not complete yet

closed 2025-05-19 by SteveMacenski

MohammedMuthuFasil, 2025-05-20:
How to get run nav2_dock on humble?