ros-navigation/navigation2 #5188: nav2_smac_planner::SmacPlannerHybrid Generates Weird Path

wang-fire, 2025-05-23:
## Bug report

**Required Info:**

- Operating System: Ubuntu22.04
  - 
- Computer: X86_64
  - 
- ROS2 Version:ros-humble
  - 
- Version or commit hash:humble_main:latest
  - 
- DDS implementation:
  - 

#### Steps to reproduce issue

```
Simulation Environment: Turtlebot3 + gazebo  + **nav2_smac_planner**(Hybrid a star)
use rviz send a goal

![Image](https://github.com/user-attachments/assets/023b3175-66a2-4afe-a129-be856dd1b3df)

```

#### Expected behavior
valid global path

#### Actual behavior
weird global path

#### Reproduction instructions

#### Additional information
amcl:
  ros__parameters:
    use_sim_time: False
    alpha1: 0.2
    alpha2: 0.2
    alpha3: 0.2
    alpha4: 0.2
    alpha5: 0.2
    base_frame_id: "base_link"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: false
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: 100.0
    laser_min_range: -1.0
    laser_model_type: "likelihood_field"
    max_beams: 60
    max_particles: 2000
    min_particles: 500
    odom_frame_id: "odom"
    pf_err: 0.05
    pf_z: 0.99
    recovery_alpha_fast: 0.0
    recovery_alpha_slow: 0.0
    resample_interval: 1
    robot_model_type: "nav2_amcl::DifferentialMotionModel"
    save_pose_rate: 0.5
    sigma_hit: 0.2
    tf_broadcast: true
    transform_tolerance: 1.0
    update_min_a: 0.2
    update_min_d: 0.25
    z_hit: 0.5
[... 321 lines cut ...]
      plugin: "nav2_behaviors::Wait"
    assisted_teleop:
      plugin: "nav2_behaviors::AssistedTeleop"
    local_frame: odom
    global_frame: map
    robot_base_frame: base_link
    transform_tolerance: 0.1
    simulate_ahead_time: 1.0
    max_rotational_vel: 1.0
    min_rotational_vel: 0.1
    rotational_acc_lim: 0.25

velocity_smoother:
  ros__parameters:
    smoothing_frequency: 30.0
    scale_velocities: False
    feedback: "CLOSED_LOOP"
    max_velocity: [0.5, 0.0, 1.0]
    min_velocity: [-0.5, 0.0, -1.0]
    max_accel: [0.8, 0.0, 2.0]
    max_decel: [-0.8, 0.0, -2.0]
    odom_topic: "odom"
    odom_duration: 0.1
    deadband_velocity: [0.0, 0.0, 0.0]
    velocity_timeout: 1.0

SteveMacenski, 2025-05-23:
Humble's pretty out of date at this point and doesn't receive much attention due to massive updates made in the last 3 years. Have you tried working with it on Jazzy, Rolling, or soon-to-be-kilted? There are new features and fixes. There's also a new `humble_main` branch that has the latest and greatest (up until last week) of `main` that is compatible with `humble`. You may want to test that out for ease of use. 

For your first image: What is the issue you have with it specifically?

For the second image: That is very strange, I agree. That appears to be an issue where the analytic expansion is happening from a given node that is not the leaf child in the path (either somehow the wrong bin is being associated or the analytic expansion refinement isn't correctly removing the traceback pointer chain of the bypassed nodes). Given its 5x and that analytic expansion refinement is 5x that seems to be a good guess. Have you seen this happen before and have any pictures you can share? If you test against another more modern branch like `humble_main` or `main`, do you see this as well?

wang-fire, 2025-05-24:
Thank you for your reply.
     1. The branch **I am currently working on is the latest version of humble_main**, my ROS distribution is Humble.
    
     2. The first image and the second image have similar issues; those thin red arrows correspond to the topic/expansions, which should be used to debug the smac Planner's expansions for analysis.
    
     3. This issue is relatively easy to reproduce, occurring about once in every 10 planning attempts.
     
     4. Does the analytic expansion refinement refer to the parameter analytic_expansion_ratio? I set it to 1.0, but the phenomenon still occurs.
     
     5. Additionally, there are multiple angles separated from a single point, as shown in the image below.

[image]

wang-fire, 2025-05-24:
I migrated from the humble branch, but I did not encounter this issue in the humble branch.

closed 2025-05-24 by wang-fire