ros-navigation/navigation2 #5192: nav2_smac_planner,  path is not continuous

wang-fire, 2025-05-26:
**Required Info:**

    Operating System: Ubuntu22.04

    Computer: X86_64

    ROS2 Version:ros-humble（**humble_main branch**）

    Version or commit hash: humble_main:latest

**Steps to reproduce issue**

Simulation Environment: Turtlebot3 + gazebo  +  nav2_smac_planner (Hybrid a star)

**Question**

I am using humble_main branch, nav2_smac_planner, I believe this route is divided into two segments: one segment is the forward extension, and the second segment is generated by the analytic_expansion. My questions are: 
         1)  The endpoint of the forward extension and the starting point of the analytic_expansion should be coincide, rather than overlap? 
         2) If question1) is right, I want to avoid any backward motion caused by 1）. Can this be achieved through a smoothing plugin?

[image]
[image]

#### Tuning / Configuration Goal

local_costmap:
  local_costmap:
    ros__parameters:
      update_frequency: 4.0
      publish_frequency: 2.0
      global_frame: odom
      robot_base_frame: base_link
      rolling_window: true
      width: 3
      height: 3
      resolution: 0.05
      # footprint: "[[-0.27, -0.32], [-0.27, 0.32], [0.27, 0.32], [0.27, -0.32]]"
      footprint: "[[-0.125, -0.125], [-0.125, 0.125], [0.125, 0.125], [0.125, -0.125]]"
      #robot_radius: 0.22
      plugins: ["voxel_layer", "inflation_layer"]
      # filters: ["keepout_filter"]
      # keepout_filter:
      #   plugin: "nav2_costmap_2d::KeepoutFilter"
      #   enabled: True
      #   filter_info_topic: "costmap_filter_info"
      inflation_layer:
        plugin: "nav2_costmap_2d::InflationLayer"
        cost_scaling_factor: 3.0
        inflation_radius: 0.40
      voxel_layer:
        plugin: "nav2_costmap_2d::VoxelLayer"
        enabled: True
        publish_voxel_map: True
        origin_z: 0.0
        z_resolution: 0.05
        z_voxels: 16
        max_obstacle_height: 2.0
        mark_threshold: 0
        observation_sources: scan
        scan:
          # A relative topic will be appended to the parent of the local_costmap namespace.
          # For example:
          #   * User chosen namespace is `tb4`.
          #   * User chosen topic is `scan`.
          #   * Topic will be remapped to `/tb4/scan` without `local_costmap`.
          #   * Use global topic `/scan` if you do not wish the node namespace to apply
          topic: scan
          max_obstacle_height: 2.0
          clearing: True
          marking: True
          data_type: "LaserScan"
          raytrace_max_range: 3.0
          raytrace_min_range: 0.0
          obstacle_max_range: 2.5
          obstacle_min_range: 0.0
      static_layer:
        plugin: "nav2_costmap_2d::StaticLayer"
        map_subscribe_transient_local: True
      always_send_full_costmap: True
      service_introspection_mode: "disabled"

global_costmap:
  global_costmap:
    ros__parameters:
      update_frequency: 1.0
      publish_frequency: 1.0
      global_frame: map
      robot_base_frame: base_link
      #robot_radius: 0.22
      # footprint: "[[-0.27, -0.32], [-0.27, 0.32], [0.27, 0.32], [0.27, -0.32]]"
      footprint: "[[-0.125, -0.125], [-0.125, 0.125], [0.125, 0.125], [0.125, -0.125]]"
      resolution: 0.05
      track_unknown_space: true
      plugins: ["static_layer", "obstacle_layer", "inflation_layer"]
      # filters: ["keepout_filter"]
      # keepout_filter:
      #   plugin: "nav2_costmap_2d::KeepoutFilter"
      #   enabled: True
      #   filter_info_topic: "costmap_filter_info"
      obstacle_layer:
        plugin: "nav2_costmap_2d::ObstacleLayer"
        enabled: True
        observation_sources: scan
        scan:
          # A relative topic will be appended to the parent of the global_costmap namespace.
          # For example:
          #   * User chosen namespace is `tb4`.
          #   * User chosen topic is `scan`.
          #   * Topic will be remapped to `/tb4/scan` without `global_costmap`.
          #   * Use global topic `/scan` if you do not wish the node namespace to apply
          topic: scan
          max_obstacle_height: 2.0
          clearing: True
          marking: True
          data_type: "LaserScan"
          raytrace_max_range: 3.0
          raytrace_min_range: 0.0
          obstacle_max_range: 2.5
          obstacle_min_range: 0.0
      static_layer:
        plugin: "nav2_costmap_2d::StaticLayer"
        map_subscribe_transient_local: True
      inflation_layer:
        plugin: "nav2_costmap_2d::InflationLayer"
        cost_scaling_factor: 3.0
        inflation_radius: 0.40
      always_send_full_costmap: True
      service_introspection_mode: "disabled"

map_server:
  ros__parameters:
    # yaml_filename: "map.yaml"
    service_introspection_mode: "disabled"

map_saver:
  ros__parameters:
    save_map_timeout: 5.0
    free_thresh_default: 0.25
    occupied_thresh_default: 0.65
    map_subscribe_transient_local: True
    service_introspection_mode: "disabled"

planner_server:
  ros__parameters:
    expected_planner_frequency: 20.0
    planner_plugins: ["GridBased"]
    costmap_update_timeout: 1.0
    service_introspection_mode: "disabled" 
    GridBased:
      plugin: "nav2_smac_planner::SmacPlannerHybrid" # In Iron and older versions, "/" was used instead of "::"
      downsample_costmap: False           # whether or not to downsample the map
      downsampling_factor: 2             
      tolerance: 0.10                     
      allow_unknown: true                 # allow traveling in unknown space
      max_iterations: 1000000            
      max_on_approach_iterations: 1000    # Maximum number of iterations after within tolerances to continue to try to find exact solution
      max_planning_time: 5.0              # max time in s for planner to plan, smooth
      motion_model_for_search: "REEDS_SHEPP"    # Hybrid-A* Dubin, Redds-Shepp
      angle_quantization_bins: 72         # Number of angle bins for search
      analytic_expansion_ratio: 1.0       # The ratio to attempt analytic expansions during search for final approach.
      analytic_expansion_max_length: 1.5  
      analytic_expansion_max_cost: 200.0  # The maximum single cost for any part of an analytic expansion to contain and be valid, except when necessary on approach to goal
      analytic_expansion_max_cost_override: False  #  Whether or not to override the maximum cost setting if within critical distance to goal (ie probably required)
      minimum_turning_radius: 0.40        # minimum turning radius in m of path / vehicle
      reverse_penalty: 100.0                # Penalty to apply if motion is reversing, must be => 1
      change_penalty: 0.3                 # Penalty to apply if motion is changing directions (L to R), must be >= 0
      non_straight_penalty: 1.5           # Penalty to apply if motion is non-straight, must be => 1
      cost_penalty: 5.0                  
      retrospective_penalty: 0.015
      lookup_table_size: 5.0             # Size of the dubin/reeds-sheep distance window to cache, in meters.
      cache_obstacle_heuristic: False    
      debug_visualizations: True         
      use_quadratic_cost_penalty: False
      downsample_obstacle_heuristic: False
      allow_primitive_interpolation: True
      coarse_search_resolution: 2         # Number of bins to skip when doing a coarse search for the path. Only used for all_direction goal heading mode.
      goal_heading_mode: "DEFAULT"       
      smooth_path: False                  
      smoother:
        max_iterations: 1000
        w_smooth: 0.3
        w_data: 0.2
        tolerance: 1.0e-10
        do_refinement: true
        refinement_num: 2

SteveMacenski, 2025-05-27:
Does this happen on Humble or Jazzy branch? I'm wondering if some recent work introduced a regression https://github.com/ros-navigation/navigation2/pull/4127. That work is not backported to Humble / Jazzy so if you don't see it there, that would point to an issue with this PR which touches the part of the code that could cause something like this. 

This should absolutely not be happening with the only exception: If your robot's turning radius is too large to make a turn, it might actually choose to go backwards at a turn, to go forward again from another angle in order to make the turn. Think of it as a 3-point turn maneuver. Please verify that this is not the case first before we go further (but from these 2 images it looks like its at the junction with the analytic expansions, so I don't think that's the case - it looks like a regression. You'd see this behavior elsewhere than just the switch to the analytic expansions if it were a turning radius related issue). 

FYI @stevedanomodolor

stevedanomodolor, 2025-05-27:
> Does this happen on Humble or Jazzy branch? I'm wondering if some recent work introduced a regression [#4127](https://github.com/ros-navigation/navigation2/pull/4127). That work is not backported to Humble / Jazzy so if you don't see it there, that would point to an issue with this PR which touches the part of the code that could cause something like this.
> 
> This should absolutely not be happening with the only exception: If your robot's turning radius is too large to make a turn, it might actually choose to go backwards at a turn, to go forward again from another angle in order to make the turn. Think of it as a 3-point turn maneuver. Please verify that this is not the case first before we go further (but from these 2 images it looks like its at the junction with the analytic expansions, so I don't think that's the case - it looks like a regression. You'd see this behavior elsewhere than just the switch to the analytic expansions if it were a turning radius related issue).
> 
> FYI [@stevedanomodolor](https://github.com/stevedanomodolor) I think this points out a potential issue with your PR

I will verify   if i get this behavior locally because the analytical expansion should have the same behavior as is. What was done was only do more copmutation when you have multiple goals.

wang-fire, 2025-05-27:
> Does this happen on Humble or Jazzy branch? I'm wondering if some recent work introduced a regression [#4127](https://github.com/ros-navigation/navigation2/pull/4127). That work is not backported to [...]

I have not encountered this issue on the Humble branch. 

I also found two other issues in analytic_expansion.cpp on this branch:

   1） On line 100 and line 119 (bool score = refineAnalyticPath(current_node, current_goal_node, getter, analytic_nodes);), the refineAnalyticPath function returns a float value, but it is assigned to a variable of type bool.

    2）If refineAnalyticPath executes successfully, the subsequent association of the current_node is incorrect because the parent of the path generated by the refine analytical expansion is no longer the current_node.

@SteveMacenski @stevedanomodolor

stevedanomodolor, 2025-05-27:
> I have not encountered this issue on the Humble branch.
>
> I also found two other issues in analytic_expansion.cpp on this branch:
>
>    1） On line 100 and line 119 (bool score = [...]

Can you test it and see if you still get the issue you are having? I also noticed the two issues you mentioned, but can not test until the evening.

wang-fire, 2025-05-27:
> > > Does this happen on Humble or Jazzy branch? I'm wondering if some recent work introduced a regression [#4127](https://github.com/ros-navigation/navigation2/pull/4127). That work is not backported to Humble / Jazzy so if you don't see it there, that would point to an issue with this PR which touches the part of the code that could cause something like this.
> > > This should absolutely not be happening with the only exception: If your robot's turning radius is too large to make a turn, it might actually choose to go backwards at a turn, to go forward again from another angle in order to make the turn. Think of it as a 3-point turn maneuver. Please verify that this is not the case first before we go further (but from these 2 images it looks like its at the junction with the analytic expansions, so I don't think that's the case - it looks like a regression. You'd see this behavior elsewhere than just the switch to the analytic expansions if it were a turning radius related issue).
> > > FYI [@stevedanomodolor](https://github.com/stevedanomodolor) I think this points out a potential issue with your PR
> > 
> > 
> > I have not encountered this issue on the Humble branch.
> > I also found two other issues in analytic_expansion.cpp on this branch:
> > 1） On line 100 and line 119 (bool score = refineAnalyticPath(current_node, current_goal_node, getter, analytic_nodes);), the refineAnalyticPath function returns a float value, but it is assigned to a variable of type bool.
> > ```
> > 2）If refineAnalyticPath executes successfully, the subsequent association of the current_node is incorrect because the parent of the path generated by the refine analytical expansion is no longer the current_node.
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > [@SteveMacenski](https://github.com/SteveMacenski) [@stevedanomodolor](https://github.com/stevedanomodolor)
> 
> Nice catch 😌, my mistake 😅. Can you test it and see if you still get the issue you are having? I also noticed the two issues you mentioned, but can not test until the evening.

The other two issues I mentioned earlier have been fixed.  Can you give me some ideas on how to fix the current issue of the analytic path and forward path being discontinuous? This problem occurs about once in every ten attempts in my environment and has been bothering me for 1-2 days.

stevedanomodolor, 2025-05-27:
@SteveMacenski  The issue still persist, i tried this git commit https://github.com/ros-navigation/navigation2/commit/b6a5387d6823de6eb8008da995a72086e8ace1e5.  One before my merge. See video below. Myabe it is configuration issue?. I am using the exact same configuration as @wang-fire 

https://github.com/user-attachments/assets/1a32c503-dd5d-4e3b-9346-5e963447356f

SteveMacenski, 2025-05-28:
@stevedanomodolor do you see it after your fix? From both of your experiences, it is always just '1' that is off? 

Can this be reproduced with the defaults in general?  Does adjusting `cost_penalty` `reverse_penalty` and `non_straight_penalty` to be more aligned with [our defaults ](https://docs.nav2.org/configuration/packages/smac/configuring-smac-hybrid.html)change anything (leaving the rest the same)? I'm curious if there's an actual "need" for this behavior in this tuning in order to maximize scoring. When you have the cost penalty so high, its possible that it incentivizes taking small maneuvers in the reeds-shepp path to set itself up to reduce the amount of cells it spends in non-0-costed space later. I have shown in the past if I blow up the penalties in extreme directions I can get somewhat odd behavior for correct reasons. The reeds-shepp analytic expansion will do a refinement stage and its possible in that from the cost score that it chooses to go backwards for a bit in order to get setup at a better angle for reducing integrated path cost.

stevedanomodolor, 2025-05-28:
> Hello!  
> I'm afraid I've discovered another issue related to this PR. Specifically, a planner (I use the lattice one, humble_main branch) can sometimes generate an incorrect x (and possibly y) value in the goal pose, resulting in an entirely incorrect path to the another, not target, position. I tested reverting commit f5543c39aba5d3302208bc6e9696e7cbd41de8c9, and that resolved the problem. But I have no idea how is it possible. 

How can i reproduce this?

slazarev8, 2025-05-28:
> How can i reproduce this, what goals are you sending?

I use diff platform with reverse_expansion, the footprint is rectangle. The goal is about 10m far away from robot. Not sure other details are important, because I don't use full nav2 stack. 
 

> Can you with this pr [#5198](https://github.com/ros-navigation/navigation2/pull/5198) to see if you get the x and y issue.

No, this fix does not treat my issue =(

stevedanomodolor, 2025-05-28:
> I use diff platform with reverse_expansion, the footprint is rectangle. The goal is about 10m far away from robot. Not sure other details are important, because I don't use full nav2 stack.
>
>
>
> No, [...]

To prevent from mixing issues and given that your stack is not full nav2, while i work on understanding the initial issue with the path including backward and forward movement, i recommend you look and log into the input that gets sent to the planners. You can find this here  https://github.com/ros-navigation/navigation2/blob/1278df713a4fa4e80cc6d5903046180ec3b3686b/nav2_smac_planner/src/smac_planner_hybrid.cpp#L449.

Maybe @SteveMacenski has more insight on the x y issue.

SteveMacenski, 2025-05-28:
@slazarev8 please be more descriptive. What is your ROS version, branch, CPU type, give us some visuals/images, reproduction examples on a map, etc.

stevedanomodolor, 2025-05-29:
@SteveMacenski  

Before

https://github.com/user-attachments/assets/ec40ba31-c14d-45e5-9e8b-e5f158dc36af

stevedanomodolor, 2025-05-29:
After

https://github.com/user-attachments/assets/942ef7ce-3c24-469c-a3c1-f0ae88276f4d

stevedanomodolor, 2025-05-30:
It setill happnes on the feature branch.  It happens mostly in corners, not sure if my costmap definition is too high
This is my config. 

```
amcl:
  ros__parameters:
    alpha1: 0.2
    alpha2: 0.2
    alpha3: 0.2
    alpha4: 0.2
    alpha5: 0.2
    base_frame_id: "base_footprint"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: false
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: 100.0
    laser_min_range: -1.0
    laser_model_type: "likelihood_field"
    max_beams: 60
    max_particles: 2000
    min_particles: 500
    odom_frame_id: "odom"
    pf_err: 0.05
    pf_z: 0.99
    recovery_alpha_fast: 0.0
    recovery_alpha_slow: 0.0
    resample_interval: 1
    robot_model_type: "nav2_amcl::DifferentialMotionModel"
    save_pose_rate: 0.5
    sigma_hit: 0.2
    tf_broadcast: true
    transform_tolerance: 1.0
    service_introspection_mode: "disabled"
    update_min_a: 0.2
    update_min_d: 0.25
    z_hit: 0.5
    z_max: 0.05
    z_rand: 0.5
    z_short: 0.05
    scan_topic: scan

bt_navigator:
  ros__parameters:
    global_frame: map
    robot_base_frame: base_link
    odom_topic: odom
    bt_loop_duration: 10
    filter_duration: 0.3
    default_server_timeout: 20
    wait_for_service_timeout: 1000
    service_introspection_mode: "disabled"
    navigators: ["navigate_to_pose", "navigate_through_poses"]
    navigate_to_pose:
      plugin: "nav2_bt_navigator::NavigateToPoseNavigator"
      enable_groot_monitoring: false
      groot_server_port: 1667
    navigate_through_poses:
      plugin: "nav2_bt_navigator::NavigateThroughPosesNavigator"
      enable_groot_monitoring: false
      groot_server_port: 1669
    # 'default_nav_through_poses_bt_xml' and 'default_nav_to_pose_bt_xml' are use defaults:
    # nav2_bt_navigator/navigate_to_pose_w_replanning_and_recovery.xml
    # nav2_bt_navigator/navigate_through_poses_w_replanning_and_recovery.xml
    # They can be set here or via a RewrittenYaml remap from a parent launch file to Nav2.

    # plugin_lib_names is used to add custom BT plugins to the executor (vector of strings).
[... 591 lines cut ...]
      use_collision_detection: true
      costmap_topic: "local_costmap/costmap_raw"
      footprint_topic: "local_costmap/published_footprint"
      transform_tolerance: 0.1
      projection_time: 5.0
      simulation_step: 0.1
      dock_collision_threshold: 0.3

loopback_simulator:
  ros__parameters:
    base_frame_id: "base_footprint"
    odom_frame_id: "odom"
    map_frame_id: "map"
    scan_frame_id: "base_scan" # tb4_loopback_simulator.launch.py remaps to 'rplidar_link'
    update_duration: 0.02
    scan_range_min: 0.05
    scan_range_max: 30.0
    scan_angle_min: -3.1415
    scan_angle_max: 3.1415
    scan_angle_increment: 0.02617
    scan_use_inf: true

```

https://github.com/user-attachments/assets/07b838e8-a9c7-4355-8a3e-659f41e2a354

SteveMacenski, 2025-06-01:
> It setill happnes on the feature branch. It happens mostly in corners, not sure if my costmap definition is too high

As I said in my explanation, it can still happen but its not a bug, that is the reeds-shepp path that is computed to create a feasible path.

closed 2025-06-01 by SteveMacenski

SteveMacenski, 2025-06-04:
An attempt at a fix: https://github.com/ros-navigation/navigation2/pull/5235 @slazarev8 can you test?

slazarev8, 2025-06-05:
> An attempt at a fix: [#5235](https://github.com/ros-navigation/navigation2/pull/5235) [@slazarev8](https://github.com/slazarev8) can you test?

As discussed [here](https://github.com/ros-navigation/navigation2/pull/5217#issuecomment-2940815364) I've tested it, and on [this](https://github.com/ros-navigation/navigation2/pull/5235) branch I can't reproduce my problem.

PR #5207 (2025-06-01): Prototype solving #5192 Issue 2: Reeds-Shepp reduce small reverse expansions
Resolves #5192
PR #5216 (2025-06-01): Prototype solving #5192 Issue 2: Reeds-Shepp reduce small reverse expansions (backport #5207)
Resolves #5192<hr>This is an automatic backport of pull request #5207 done by [Mergify](https://mergify.com).